openbmc/openbmc #830: [Barreleye v1.99.0-291-g7a6b0ba] The base /org listing pruned off ?

gkeishin, 2016-11-23:
rango@ubuntu:~/openbmc-test-automation$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  GET https://xx.xx.xx.xx/
{
 "data": [], 
 "message": "200 OK", 
 "status": "ok"
}

This is suppose to return /org in the data listing.. so the base seems to start with /org/openbmc

rango@ubuntu:~/openbmc-test-automation$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  GET https://xx.xx.xx.xx/org/
{
"data": [
    "/org/openbmc"
  ], 
  "message": "200 OK", 
  "status": "ok"

``` What changed which made to behave like this to prune off the base /org listing. ```

There are test case which validates this and are broken because of this..

bradbishop, 2016-11-23:
This has the same cause as https://github.com/openbmc/openbmc/issues/828

gkeishin, 2016-11-24:
We still continue to see this in latest build v1.99.0-293-g2feb3ec - ( https://openpower.xyz/job/openbmc-build/386/ )
- [x] Journal log - 
[barreleye_v1.99.0-293-g2feb3ec_Poweron_journal.txt](https://github.com/openbmc/openbmc/files/610526/barreleye_v1.99.0-293-g2feb3ec_Poweron_journal.txt)

rahulmah, 2016-12-06:
@bradbishop : The issue which you had mention earlier is already closed and fixed. But this issue is still seen with the latest build v1.99.0-309. Please try to look at this issue and provide update.

gkeishin, 2017-01-31:
This behavior is also seen with the new URI xyz path as well

bradbishop, 2017-07-14:
I think this was fixed here: openbmc/openbmc#1544

closed 2017-07-14 by bradbishop