openbmc/openbmc #831: [Barreleye v1.99.0-291-g7a6b0ba] REST settings view listing gone ?

gkeishin, 2016-11-23:
The host is booted however when I looked at get request to /org/openbmc/  we now no longer see settings service..  any explanation here ?

- [x] Journal log 
[barreleye_v1.99.0-291-g7a6b0ba_HOST_BOOTED.txt](https://github.com/openbmc/openbmc/files/608575/barreleye_v1.99.0-291-g7a6b0ba_HOST_BOOTED.txt)

rango@ubuntu:~/TEST_LOGS$curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  GET https://xx.xx.xx.xx/org/openbmc/
{
  "data": [
    "/org/openbmc/sensors", 
    "/org/openbmc/control", 
    "/org/openbmc/buttons", 
    "/org/openbmc/inventory", 
    "/org/openbmc/HostServices", 
    "/org/openbmc/watchdog"
  ], 
  "message": "200 OK", 
  "status": "ok"
}

gkeishin, 2016-11-23:
This is also seems then due to https://github.com/openbmc/openbmc/issues/828 .. which seems fixed.. awaiting build to verify..

gkeishin, 2016-11-24:
Still an issue in v1.99.0-293-g2feb3ec build  ( https://openpower.xyz/job/openbmc-build/386/ )

ratagupt, 2016-11-24:
Nov 21 06:09:47 barreleye settings_manager.py[786]: Traceback (most recent call last):
Nov 21 06:09:47 barreleye settings_manager.py[786]:   File "/usr/sbin/settings_manager.py", line 245, in <module>
Nov 21 06:09:47 barreleye settings_manager.py[786]:     allobjects = create_object(s.SETTINGS)
Nov 21 06:09:47 barreleye settings_manager.py[786]:   File "/usr/sbin/settings_manager.py", line 71, in create_object
Nov 21 06:09:47 barreleye settings_manager.py[786]:     paths = mapper.get_subtree_paths(setting['subtree'], 0)
Nov 21 06:09:47 barreleye settings_manager.py[786]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 62, in get_subtree_paths
Nov 21 06:09:47 barreleye settings_manager.py[786]:     retries)
Nov 21 06:09:47 barreleye settings_manager.py[786]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 42, in retry
Nov 21 06:09:47 barreleye settings_manager.py[786]:     return func()
Nov 21 06:09:47 barreleye settings_manager.py[786]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 61, in <lambda>
Nov 21 06:09:47 barreleye settings_manager.py[786]:     path, depth, interfaces, signature='sias'),
Nov 21 06:09:47 barreleye settings_manager.py[786]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Nov 21 06:09:47 barreleye settings_manager.py[786]:     **keywords)
Nov 21 06:09:47 barreleye settings_manager.py[786]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Nov 21 06:09:47 barreleye settings_manager.py[786]:     message, timeout)
Nov 21 06:09:47 barreleye settings_manager.py[786]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic
Nov 21 06:09:47 barreleye systemd[1]: org.openbmc.settings.Host.service: Main process exited, code=exited, status=1/FAILURE
Nov 21 06:09:47 barreleye systemd[1]: Failed to start Phosphor Settings Manager.
Nov 21 06:09:47 barreleye systemd[1]: org.openbmc.settings.Host.service: Unit entered failed state.
Nov 21 06:09:47 barreleye systemd[1]: org.openbmc.settings.Host.service: Failed with result 'exit-cod

geissonator, 2016-11-28:
I think both this one and #832 match up with #812 as well.

sivassrr, 2016-12-06:
Are we seeing this problem with the latest driver?

gkeishin, 2016-12-06:
Yes this still seen.,. probably it may intend to hide the settings perhaps.. only @bradbishop can shed some info on this..

williamspatrick, 2017-03-10:
Already fixed.

closed 2017-03-10 by williamspatrick