openbmc/openbmc #832: [ Barreleye v1.99.0-293-g2feb3ec  ] Poweron Failed to boot Host

gkeishin, 2016-11-24:
Nov 21 06:36:08 barreleye systemd[1]: Reached target Start Host0 (Pre).
``` Nov 21 06:36:12 barreleye power_control.exe[767]: ERROR PowerControl: Pgood poll timeout```

`Build v1.99.0-293-g2feb3ec - ( https://openpower.xyz/job/openbmc-build/386/ )`
`root@barreleye:~# uptime`
`06:43:05 up 35 min,  load average: 0.18, 0.46, 0.41`
`root@barreleye:~# obmcutil state`
` = HOST_POWERING_ON`
- [x] Journal log - 
[barreleye_v1.99.0-293-g2feb3ec_Poweron_journal.txt](https://github.com/openbmc/openbmc/files/610526/barreleye_v1.99.0-293-g2feb3ec_Poweron_journal.txt)

gkeishin, 2016-11-24:
This same error is seen across Barreleye systems.

ratagupt, 2016-11-24:
Nov 21 06:09:49 barreleye settings_manager.py[783]: Traceback (most recent call last):
Nov 21 06:09:49 barreleye settings_manager.py[783]:   File "/usr/sbin/settings_manager.py", line 245, in <module>
Nov 21 06:09:49 barreleye settings_manager.py[783]:     allobjects = create_object(s.SETTINGS)
Nov 21 06:09:49 barreleye settings_manager.py[783]:   File "/usr/sbin/settings_manager.py", line 71, in create_object
Nov 21 06:09:49 barreleye settings_manager.py[783]:     paths = mapper.get_subtree_paths(setting['subtree'], 0)
Nov 21 06:09:49 barreleye settings_manager.py[783]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 62, in get_subtree_paths
Nov 21 06:09:49 barreleye settings_manager.py[783]:     retries)
Nov 21 06:09:49 barreleye settings_manager.py[783]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 42, in retry
Nov 21 06:09:49 barreleye settings_manager.py[783]:     return func()
Nov 21 06:09:49 barreleye settings_manager.py[783]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 61, in <lambda>
Nov 21 06:09:49 barreleye settings_manager.py[783]:     path, depth, interfaces, signature='sias'),
Nov 21 06:09:49 barreleye settings_manager.py[783]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Nov 21 06:09:49 barreleye settings_manager.py[783]:     **keywords)
Nov 21 06:09:49 barreleye settings_manager.py[783]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Nov 21 06:09:49 barreleye settings_manager.py[783]:     message, timeout)
Nov 21 06:09:49 barreleye settings_manager.py[783]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic
Nov 21 06:09:49 barreleye systemd[1]: org.openbmc.settings.Host.service: Main process exited, code=exited, status=1/FAILURE
Nov 21 06:09:49 barreleye systemd[1]: Failed to start Phosphor Settings Manager.
Nov 21 06:09:49 barreleye systemd[1]: org.openbmc.settings.Host.service: Unit entered failed state.
Nov 21 06:09:49 barreleye systemd[1]: org.openbmc.settings.Host.service: Failed with result 'exit-code'.
Nov 21 06:09:51 barreleye system_manager.py[731]: SystemManager Goto System State: BMC_READY
Nov 21 06:09:51 barreleye system_manager.py[731]: Running System State: BMC_READY

ratagupt, 2016-11-24:
it is same as 831.
https://github.com/openbmc/openbmc/issues/831

amboar, 2016-11-24:
@ratagupt please use the triple backtick markdown when pasting logs into github comments. It formats the content into a scroll pane with fixed-width text which makes it much easier on the eye and stops github from rendering emojis in your log output.

```
It looks like this
```

gkeishin, 2016-11-30:
Poweron problem seems to fade away in newer drivers.  Did someone fixed it ?

gkeishin, 2016-12-01:
Not see in latest build
Latest build : v1.99.0-302-g0d1a644 - ( https://openpower.xyz/job/openbmc-build/393/ )

geissonator, 2016-12-07:
Hit this today on BarPVT when running HW CI.  I updated to a good build but still hit "ERROR PowerControl: Pgood poll timeout" during power on.  A PDU cycle fixed the issue.

sivassrr, 2017-02-09:
@gkeishin Lets leave it open, so that it will come under development radar

geissonator, 2017-02-19:
Had just powered on this witherspoon successfully.  I powered it of, and then tried to power on again and hit this issue:

```
Feb 02 17:45:25 witherspoon phosphor-host-state-manager[897]: Host State transaction request
Feb 02 17:45:25 witherspoon systemd[1]: Starting Inventory Upload...
Feb 02 17:45:25 witherspoon systemd[1]: Stopped target Power0 Off.
Feb 02 17:45:25 witherspoon systemd[1]: Starting Wait for Power0 to turn on...
Feb 02 17:45:25 witherspoon systemd[1]: Stopped target Chassis0 (Stop).
Feb 02 17:45:25 witherspoon systemd[1]: Stopped Stop Watchdog0.
Feb 02 17:45:25 witherspoon systemd[1]: Stopped target Chassis0 (Power Off).
Feb 02 17:45:25 witherspoon systemd[1]: Stopped Reload mboxd after power off.
Feb 02 17:45:25 witherspoon systemd[1]: Stopped Disable VCS on a power off.
Feb 02 17:45:25 witherspoon systemd[1]: Stopped Wait for Power0 to turn off.
Feb 02 17:45:25 witherspoon systemd[1]: Stopped Stop Power0.
Feb 02 17:45:25 witherspoon systemd[1]: Starting Disable UCD VCS rails...
Feb 02 17:45:26 witherspoon phosphor-host-state-manager[897]: Recieved signal that host is off
Feb 02 17:45:26 witherspoon phosphor-host-state-manager[897]: Change to Host State
Feb 02 17:45:26 witherspoon phosphor-chassis-state-manager[896]: Recieved signal that power OFF is complete
Feb 02 17:45:26 witherspoon phosphor-chassis-state-manager[896]: Change to Chassis Power State
ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 02 17:45:26 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 02 17:45:26 witherspoon kernel[754]: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 02 17:45:27 witherspoon systemd[1]: Started Disable UCD VCS rails.
Feb 02 17:45:27 witherspoon systemd[1]: Starting Perform AVS bus workaround on VRMs...
Feb 02 17:45:27 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Feb 02 17:45:27 witherspoon systemd[1]: Started Perform AVS bus workaround on VRMs.
Feb 02 17:45:28 witherspoon systemd[1]: Starting Disable the AVS bus on the VRMs...
Feb 02 17:45:28 witherspoon systemd[1]: Started Disable the AVS bus on the VRMs.
Feb 02 17:45:28 witherspoon systemd[1]: Starting Apply voltage overrides to VRMs...
Feb 02 17:45:29 witherspoon vrm-control.sh[1428]: rail        set        read        current
Feb 02 17:45:29 witherspoon vrm-control.sh[1428]: -------        ------- ------- -------
Feb 02 17:45:29 witherspoon vrm-control.sh[1428]: vdna        0.898V        0.004V        0.000A
Feb 02 17:45:30 witherspoon vrm-control.sh[1428]: vddra        1.352V        0.012V        0.000A
Feb 02 17:45:31 witherspoon vrm-control.sh[1428]: vdnb        0.898V        0.004V        0.000A
Feb 02 17:45:31 witherspoon vrm-control.sh[1428]: vddrb        1.352V        0.012V        0.000A
Feb 02 17:45:31 witherspoon systemd[1]: Started Apply voltage overrides to VRMs.
Feb 02 17:45:31 witherspoon systemd[1]: Starting Enable the AVS bus on VRMs...
Feb 02 17:45:32 witherspoon systemd[1]: Started Enable the AVS bus on VRMs.
Feb 02 17:45:32 witherspoon systemd[1]: Starting Disable VCS before Witherspoon power on...
Feb 02 17:45:32 witherspoon systemd[1]: Started Disable VCS before Witherspoon power on.
Feb 02 17:45:33 witherspoon systemd[1]: Reached target Power0 On (Pre).
Feb 02 17:45:33 witherspoon systemd[1]: Starting Start Power0...
Feb 02 17:45:33 witherspoon power_control.exe[901]: PowerControl: setting power up SOFTWARE_PGOOD to 1
Feb 02 17:45:33 witherspoon power_control.exe[901]: PowerControl: setting power up BMC_POWER_UP to 1
Feb 02 17:45:33 witherspoon systemd[1]: Started Start Power0.
Feb 02 17:45:33 witherspoon timemanager[934]:  PGOOD has changed..
Feb 02 17:45:33 witherspoon system_manager.py[878]: Running System State: HOST_POWERING_ON
Feb 02 17:45:36 witherspoon inventory_upload.py[1388]: ERROR (name_properties): "name" property in /bmc/inventory/misc is incorrect ("0000000000000000" instead of base node name)
Feb 02 17:45:36 witherspoon inventory_upload.py[1388]: ERROR (name_properties): "name" property in /bmc/inventory/cpu0 is incorrect ("PROCESSOR MODULE" instead of base node name)
Feb 02 17:45:36 witherspoon inventory_upload.py[1388]: ERROR: Input tree has errors, aborting (use -f to force output)
Feb 02 17:45:36 witherspoon inventory_upload.py[1388]: Uploading inventory to PNOR in dtb format...
Feb 02 17:45:36 witherspoon inventory_upload.py[1388]: About to erase 0x027a0000..0x027a9000 !
Feb 02 17:45:36 witherspoon inventory_upload.py[1388]: Erasing...
Feb 02 17:45:36 witherspoon inventory_upload.py[1388]: [570B blob data]
Feb 02 17:45:36 witherspoon inventory_upload.py[1388]: About to program "/tmp/tmpsp9HE9" at 0x027a0000..0x027a9000 !
Feb 02 17:45:36 witherspoon inventory_upload.py[1388]: Programming & Verifying...
Feb 02 17:45:36 witherspoon inventory_upload.py[1388]: [56B blob data]
Feb 02 17:45:36 witherspoon inventory_upload.py[1388]: Updating actual size in partition header...
Feb 02 17:45:36 witherspoon systemd[1]: Started Inventory Upload.
Feb 02 17:45:46 witherspoon power_control.exe[901]: ERROR PowerControl: Pgood poll timeout
Feb 02 17:45:46 witherspoon timemanager[934]:  PGOOD has changed..
```

gkeishin, 2017-02-20:
On Witherspoon also I see this @geissonator 
```
Feb 02 18:17:56 witherspoon inventory_upload.py[904]: [56B blob data]
Feb 02 18:17:56 witherspoon inventory_upload.py[904]: Updating actual size in partition header...
Feb 02 18:17:56 witherspoon systemd[1]: Started Inventory Upload.
Feb 02 18:18:05 witherspoon power_control.exe[855]: ERROR PowerControl: Pgood poll timeout
Feb 02 18:18:05 witherspoon timemanager[876]:  PGOOD has changed..

root@witherspoon:~# obmcutil bootprogress
units =
value = Off
error = 0
root@witherspoon:~# obmcutil power
pgood = 0
state = 1
pgood_timeout = 0
root@witherspoon:~#
```

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-71"
VERSION_ID="v1.99.2-71-gbc49f79-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-71"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170218231615
root@witherspoon:~#
```

geissonator, 2017-03-09:
Trying to get to the bottom of this as it's greatly impacting my ability to test my new 'bmc reset while chassis power on' code.  

I'm running latest code on a barreleye.  I'm seeing this on multiple barreleyes:
```
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-51"
VERSION_ID="v1.99.3-51-gfb43217"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-51"
BUILD_ID="v1.99.3"
```

I flash new code and it does not boot (hits pgood timeout issue)
I PDU cycle the system and pgood comes on, but PNOR no longer boots
I flash PNOR and the system will boot to the OS
I power down and try to power back on and we're back to pgood issue.

One thing I noticed is that we get the ethernet drop, even when pgood doesn't come on, which seems to indicate a portion of the power on sequence is happening.  Patrick noted that the partial power on could be what's causing the PNOR corruption.

Here's a working power on sequence:
```
Feb 01 02:54:39 barreleye power_control.exe[809]: PowerControl: setting power up POWER_PIN to 0
Feb 01 02:54:39 barreleye systemd[1]: Started Start Power0.
Feb 01 02:54:39 barreleye timemanager[879]:  PGOOD has changed..
Feb 01 02:54:39 barreleye system_manager.py[758]: Running System State: HOST_POWERING_ON
Feb 01 02:54:40 barreleye fan_control.exe[865]: Failed to GetObject: Traceback (most recent call last):
Feb 01 02:54:40 barreleye fan_control.exe[865]:   File "/usr/lib/python2.7/site-packages/dbus/service.py", line 707, in _message_cb
Feb 01 02:54:40 barreleye fan_control.exe[865]:     retval = candidate_method(self, *args, **keywords)
Feb 01 02:54:40 barreleye fan_control.exe[865]: TypeError: GetObject() takes exactly 3 arguments (2 given)
Feb 01 02:54:40 barreleye fan_control.exe[865]: fanctl: Failed to get bus name for /org/openbmc/sensors/speed/fan0
Feb 01 02:54:40 barreleye fan_control.exe[865]: fanctl: Failed to set fan[0] speed[255]
Feb 01 02:54:39 barreleye systemd[1]: Started Set Fans to Maximum.
ftgmac100 1e660000.ethernet eth0: NCSI interface down
Feb 01 02:54:41 barreleye kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Feb 01 02:54:41 barreleye kernel[697]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
Feb 01 02:54:44 barreleye kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Feb 01 02:54:44 barreleye kernel[697]: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Feb 01 02:54:44 barreleye power_control.exe[809]: PowerControl: pgood: 1, setting reset USB_RESET to 1
Feb 01 02:54:44 barreleye system_manager.py[758]: Running System State: HOST_POWERED_ON
Feb 01 02:54:44 barreleye systemd[1]: Started Wait for Power0 to turn on.
Feb 01 02:54:44 barreleye timemanager[879]:  PGOOD has changed..
Feb 01 02:54:44 barreleye timemanager[879]: Changing time settings is *deferred* now
Feb 01 02:54:44 barreleye timemanager[879]:  User Settings have changed..
Feb 01 02:54:44 barreleye systemd[1]: Reached target Power0 On.
```

Here's the failing scenario:
```
Feb 01 03:04:07 barreleye systemd[1]: Started Start Power0.
Feb 01 03:04:07 barreleye system_manager.py[758]: Running System State: HOST_POWERING_ON
Feb 01 03:04:07 barreleye timemanager[879]:  PGOOD has changed..
ftgmac100 1e660000.ethernet eth0: NCSI interface down
Feb 01 03:04:08 barreleye kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Feb 01 03:04:08 barreleye kernel[697]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
Feb 01 03:04:10 barreleye kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Feb 01 03:04:10 barreleye kernel[697]: ftgmac100 1e660000.ethernet eth0: NCSI interface up
```

GPIO info in fail state.
```
[barreleye]# obmcutil power
pgood = 0
state = 1
pgood_timeout = 0

[barreleye]# gpioutil  -n PGOOD
GPIO ID: 315; DIRECTION: in
0
[barreleye]# cat /sys/class/gpio/gpio315/active_low 
0
[barreleye]# cat /sys/class/gpio/gpio315/direction  
in
[barreleye]# cat /sys/class/gpio/gpio315/edge      
none
[barreleye]# cat /sys/class/gpio/gpio315/value 
0

```

geissonator, 2017-03-09:
Values when power on works:
```
[barreleye]# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10
[barreleye]# cat /sys/class/gpio/gpio315/active_low 
0
[barreleye]# cat /sys/class/gpio/gpio315/direction  
in
[barreleye]# cat /sys/class/gpio/gpio315/edge      
none
[barreleye]# cat /sys/class/gpio/gpio315/value 
1
```

geissonator, 2017-03-10:
I moved the system back to v1.99.2, got 3 boots in a row, but then started hitting this issue so not sure if that told me much.

williamspatrick, 2017-03-10:
At one point this issue was the 'bus busy" error.  It seems that the comments have gone down a different path.  Should this issue be left open?

yadron, 2017-03-10:
In "v1.99.3-dirty" after BMC booted I need to wait for about 5 mins, because DBUS services not loaded yet. So it looks like I already logged in by SSH to BMC after BMC booted, but it is still busy with starting different services or other DBUS services priority problems. I see this as of IPMI commands does't work at all during these first 5 mins (like "ipmitool lan print" and other that suppose working). After command "ipmitool lan print" succeed at last, everything begin working (and I can poweron host). Possibly the poweron doesn't work with 1.99.2 or 1.99.3 during first 5 mins after BMC booted too. Possibly this is similar to issue https://github.com/openbmc/openbmc/issues/1110 (that saying "_Instead of connecting to the system bus (sd_bus_open_system()) at startup of the daemon, it was delayed till the first command got executed. This fixed the problem._" and as result service ipmid-network needs manual restart after waiting for some mins and then it works if I got it right).

geissonator, 2017-03-13:
Opened #1298 to log some additional data.  We had the hardware team look at a witherspoon in the failed state and they found that certain voltage rails had invalid data, it was voltage rails that none of the software in openbmc should be touching so the theory is that something on the i2c bus is getting left in a bad state on the initial operation and then subsequent retries are programming invalid values.

amboar, 2017-03-14:
Maybe only tangentially related, but has anyone tested with my i2c-core patch? There may have been other instances of corruption that weren't detected by the stack protector.

http://patchwork.ozlabs.org/patch/737385/

geissonator, 2017-03-14:
Unfortunately (or fortunately?) the witherspoon @gkeishin tested on today did 10 boots without issue and the barreleye I grabbed today had no pgood issues either (using this mornings master).  It'll probably be back tomorrow.

amboar, 2017-03-24:
I have submitted some work-arounds that are now merged:

* Disable the ucd90160 in the kernel devicetree: openbmc/linux@f80ce42a332cd296d698fb5996dd18f785f63e79
* Fixes for the Witherspoon userspace Vcs work-around script: https://gerrit.openbmc-project.xyz/#/c/3249/

We're still unsure how the corruption actually occurs

gkeishin, 2017-03-25:
The new build https://openpower.xyz/job/openbmc-build/651/   which had @shenki https://gerrit.openbmc-project.xyz/#/c/3253/  PGOOD fix,  ran good for 50 Power on on / off back to back..

gkeishin, 2017-03-31:
Looks all good now with the fixes gone in.. Closing this one out.. Thanks everyone on this..

closed 2017-03-31 by gkeishin

shenki, 2017-03-31:
Thanks George! Good to hear it's worked out. Thank you to everyone who worked on this, in particular @amboar

johnhcwang, 2017-03-31:
If we actually need to support ucd90160 sensor to monitor voltage and temperature on it, and have to avoid this issue when power cycle, is there any way to do it since ucd90160 isn't controlled by hwmon in this wordaround?

amboar, 2017-03-31:
@johnhcwang yes, there is, as of yesterday. Brendan Higgins from Google posted v6 of his effort to upstream the Aspeed I2C bus driver:

https://lists.ozlabs.org/pipermail/openbmc/2017-March/007074.html

He independently discovered what sounds like a suspiciously similar problem, and included the fix in that series. I've tested the series last night, and could not reproduce this issue in 130 consecutive, successful, boots of the host. This provides some evidence that our issue was the same as his.

If I recall correctly, the problem appears to stem from not sending repeat start events from irq context, instead waiting for the switch to user context. On a loaded CPU and with our 10ms scheduler slice, the 35ms SMBus timeout period passes and the UCD begins misbehaving. The fix is to deal with such things from the irq handler.

shenki, 2017-03-31:
@johnhcwang please follow this ticket. We will re-enable the hwmon driver in the near future https://github.com/openbmc/openbmc/issues/1415

johnhcwang, 2017-03-31:
@amboar and @shenki , thanks and I'll follow the new ticket.