openbmc/openbmc #833: [ Barreleye v1.99.0-293-g2feb3ec ] TimeManager service is not running

rahulmah, 2016-11-24:
TimeManager service is not running in Barreleye system. We need to start it manually to get it running. It need to be auto started.

Process running with recently flash system
```root@barreleye:~# ps | grep time
  659 systemd-   0:01 /lib/systemd/systemd-timesyncd
  855 root       0:00 grep time
root@barreleye:~#
```
Start timemanager service
```
root@barreleye:~#
root@barreleye:~# Registering dbus methods
Last known time_mode: NTP
Last known time_owner: BMC
Last known use_dhcp_ntp: yes
Reading Power Control key: pgood
Changing settings *allowed* now
Getting System Settings: time_mode
Requested_Mode: NTP Current_Mode: NTP
Mode is already set to : NTP
Getting System Settings: time_owner
Owner is already set to : BMC
Getting System Settings: use_dhcp_ntp
use_dhcp_ntp = yes
Last known host_offset:0
Initial Uptime Usec: 1479708475876051
```
Time manager service running after starting manually
```
root@barreleye:~# ps | grep time
  659 systemd-   0:01 /lib/systemd/systemd-timesyncd
  856 root       0:00 /usr/sbin/timemanager
  858 root       0:00 grep time
root@barreleye:~#
```

**BMC fw info**
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-293"
VERSION_ID="v1.99.0-293-g2feb3ec-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-293"
BUILD_ID="v1.99.0"
root@barreleye:~#
root@barreleye:~# cat /etc/version
20161123215517

ratagupt, 2016-11-24:
@rahulmah :- Please attach the journal logs..so it will be easy to debug..

rahulmah, 2016-11-25:
@ratagupt : Journal will not capture any useful info in this case. Cross checked with Vishwa, no logs are required here. Can you please assign this issue to Vishwa?

vishwabmc, 2016-11-25:
@ratagupt The issue is that org.openbmc.TimeManager.service is not put into /lib/systemd/system on this HW ( on my QEMU, I saw it was ). So there will not be anything in journal about timemanager.

rahulmah, 2016-12-15:
@adamliyi : When we can expect fix for this issue? Due to this issue, all new clock related TCs in automation fails. This need to fix on priority.

vishwabmc, 2017-01-10:
https://gerrit.openbmc-project.xyz/#/c/1712/

closed 2017-01-11 by williamspatrick