openbmc/openbmc #835: [ Barreleye v1.99.0-295-g3e20a3d ] Host didn't Poweroff

gkeishin, 2016-11-25:
The poweroff didnt bring the BMC state to HOST_POWERED_OFF..  seems like it's stuck at  HOST_POWERING_OFF state forever

- [x] Journal Log   
[barreleye_v1.99.0-295-g3e20a3d_HOST_POWERING_OFF.txt](https://github.com/openbmc/openbmc/files/612551/barreleye_v1.99.0-295-g3e20a3d_HOST_POWERING_OFF.txt)

`root@barreleye:~# obmcutil state`
 = HOST_POWERING_ON
`root@barreleye:~# obmcutil poweroff`
 = None
`root@barreleye:~# uptime`
 06:12:48 up 4 min,  load average: 1.07, 2.00, 0.99
`root@barreleye:~# obmcutil state`
 = HOST_POWERING_OFF
`root@barreleye:~# uptime`
 06:22:10 up 14 min,  load average: 0.00, 0.40, 0.60
`root@barreleye:~# obmcutil state`
 = HOST_POWERING_OFF

`root@barreleye:~# obmcutil power`
pgood = 0
state = 0
pgood_timeout = 0
root@barreleye:~#

ratagupt, 2016-11-25:
ov 21 06:11:32 barreleye systemd[1]: Starting Set Fans to Maximum...
Nov 21 06:11:32 barreleye fan_control.exe[766]: Failed to GetObject: The name org.openbmc.ObjectMapper was not provided by any .service files
Nov 21 06:11:32 barreleye fan_control.exe[766]: fanctl: Failed to get bus name for /org/openbmc/sensors/speed/fan0
Nov 21 06:11:32 barreleye fan_control.exe[766]: fanctl: Failed to set fan[0] speed[255].

Seems need to make the change in https://github.com/openbmc/skeleton/blob/master/fanctl/fan_control.c..
>>>>>>>
Changing the bus name,interfacename and objectname  to "xyz.openbmc_project.ObjectMapper"..
>>>>>>>>>

gkeishin, 2016-12-01:
This is not seen on new build 
Latest build : v1.99.0-302-g0d1a644 - ( https://openpower.xyz/job/openbmc-build/393/ )

closed 2016-12-01 by gkeishin