openbmc/openbmc #836: [ Witherspoon v1.99.0-295-g3e20a3d ] RTC Unable to read the hardware clock

gkeishin, 2016-11-25:
```
Nov 21 06:08:17 witherspoon kernel[708]: rtc-rv8803 11-0032: Voltage low, data loss detected.
Nov 21 06:08:17 witherspoon kernel[708]: rtc-rv8803 11-0032: Voltage low, data is invalid.
Nov 21 06:08:17 witherspoon kernel[708]: rtc-rv8803 11-0032: rtc core: registered rx8900 as rtc0
Nov 21 06:08:17 witherspoon kernel[708]: rtc-rv8803 11-0032: Voltage low, data is invalid.
Nov 21 06:08:17 witherspoon kernel[708]: rtc-rv8803 11-0032: hctosys: unable to read the hardware clock
```

- [x] Journal log  
[witherspoon_v1.99.0-295-g3e20a3d_journal.txt](https://github.com/openbmc/openbmc/files/612562/witherspoon_v1.99.0-295-g3e20a3d_journal.txt)

ratagupt, 2016-11-25:
I ran the command on the same machine and didn't find any issue.

root@witherspoon:~# ssh root@9.3.185.29
root@witherspoon:~# hwclock --utc --systohc 
root@witherspoon:~# hwclock --utc --systohc 
root@witherspoon:~# hwclock --show          
Mon Nov 21 08:21:53 2016  0.000000 seconds
root@witherspoon:~# hwclock  --hctosys 
root@witherspoon:~# date
Mon Nov 21 08:27:27 UTC 2016
root@witherspoon:~# hwclock --show
Mon Nov 21 08:28:10 2016  0.000000 seconds

@gkeishin Can you tell me what is the use case?hoe you are running the test case?

ratagupt, 2016-11-25:
root@witherspoon:~# hwclock --utc --systohc 
root@witherspoon:~# hwclock --utc --systohc 
root@witherspoon:~# hwclock --show          
Mon Nov 21 08:21:53 2016  0.000000 seconds
root@witherspoon:~# hwclock  --hctosys 
root@witherspoon:~# date
Mon Nov 21 08:27:27 UTC 2016
root@witherspoon:~# hwclock --show
Mon Nov 21 08:28:10 2016  0.000000 seconds

@gkeishin Can you tell me what is the use case?hoe you are running the test case?

gkeishin, 2016-11-25:
@ratagupt Not a use case.. I just booted the system and I saw this error messages in the Journal.. and I keep seeing this every boot..

amboar, 2016-11-27:
@ratagupt please use the code-fencing (triple-backtick) markdown when pasting console/log output into github. This way your output will render correctly (i.e. without random strike-throughs or emojis). See @gkeishin's example in the original post.

shenki, 2016-11-28:
@gkeishin Please add a comment with the machine you were seeing this on, and which version of the firmware and/or kernel you are running.

I had a look at w8 and the rtc appears ok:

```
rtc-rv8803 11-0032: Voltage low, temperature compensation stopped.
rtc-rv8803 11-0032: rtc core: registered rx8900 as rtc0
rtc-rv8803 11-0032: setting system clock to 2016-11-23 23:52:01 UTC (1479945121)
```

w5 appears broken:

```
rtc-rv8803 11-0032: Voltage low, temperature compensation stopped.
rtc-rv8803 11-0032: Voltage low, data loss detected.
rtc-rv8803 11-0032: Voltage low, data is invalid.
rtc-rv8803 11-0032: rtc core: registered rx8900 as rtc0
rtc-rv8803 11-0032: Voltage low, data is invalid.
rtc-rv8803 11-0032: hctosys: unable to read the hardware clock
```

gkeishin, 2016-11-28:
@shenki  This is on w1  - v1.99.0-295-g3e20a3d

sivassrr, 2016-12-06:
Any update here?

shenki, 2016-12-06:
I can't reproduce on any other machine. Can you follow up with people in the lab to see if there's something special about w1?

gkeishin, 2016-12-06:
@shenki I have been tracking this over last few builds and i don't see it popping up except once instance of it.. If there is any I'll re-open this.. For now this stays closed..

closed 2016-12-06 by gkeishin