openbmc/openbmc #838: Can't ping network in AST2400

Youngzhong, 2016-11-25:
Dear All
I can't ping network in AST2400.But the MAC and IP are OK in net confing.
When the BMC boot,We find this info(ftgmac100 1e660000.ethernet eth0: NCSI interface down).
The NCSI interface is closed.Thanks!

HW Info:
BCM54612   NCSI

BMC Info:
==================================================
root@palmetto:~# ifconfig eth0 192.168.200.6
root@palmetto:~# ifconfig
eth0      Link encap:Ethernet  HWaddr E6:C5:F6:F5:DA:B7  
          inet addr:192.168.200.6  Bcast:192.168.200.255  Mask:255.255.255.0
          inet6 addr: fe80::e4c5:f6ff:fef5:dab7%984/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:152 errors:0 dropped:0 overruns:0 frame:0
          TX packets:239 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:7640 (7.4 KiB)  TX bytes:23995 (23.4 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1%984/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:720 errors:0 dropped:0 overruns:0 frame:0
          TX packets:720 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:54720 (53.4 KiB)  TX bytes:54720 (53.4 KiB)
root@palmetto:~# ifconfig eth0 up  
root@palmetto:~# ftgmac100 1e660000.ethernet eth0: NCSI interface down
root@palmetto:~# ping 192.168.200.128
PING 192.168.200.128 (192.168.200.128): 56 data bytes
--- 192.168.200.128 ping statistics ---
2 packets transmitted, 0 packets received, 100% packet loss
============================================

gwshan, 2016-11-25:
Please tell the best way to contact you: IRC or sametime etc.

ya-mouse, 2016-11-30:
Try to run tcpdump on the 192.168.200.128 side with:
> tcpdump -evni ethX

gwshan, 2016-11-30:
If I'm correct enough, "tcudump -evni ethx" on remote side won't capture any packets from the source (BMC). I still want to debug it if possible. Please let me know the best way to contact the tester and get the BMC's access information, thanks!

Youngzhong, 2016-12-16:
If we will add Lan NCSI code or driver.Please tell me where I can modify?Thanks!

gwshan, 2016-12-16:
@Youngzhong, NCSI driver is enabled by default. Please let me know the best way to catch you on line. I can be reached by gwshan@au1.ibm.com or shan.gavin@gmail.com

gwshan, 2017-02-20:
I wasn't able to connect with @Youngzhong Anyone knows the best way to contact him?

gwshan, 2017-02-27:
I don't expect any response from @Youngzhong, I think it can be closed.

closed 2017-02-27 by shenki