openbmc/openbmc #840: Witherspoon: BMC Self test result ipmi throwing error

sivassrr, 2016-11-28:
root@witherspoon:/tmp# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-295"
VERSION_ID="v1.99.0-295-g3e20a3d-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-295"
BUILD_ID="v1.99.0"

Tried to add automated test case for ipmi self test result  - Related to https://github.com/openbmc/openbmc/issues/436

ipmitool bmc selftest is the generic command, I believe. 

root@witherspoon:/tmp# ./ipmitool bmc selftest  
Could not open device at /dev/ipmi0 or /dev/ipmi/0 or /dev/ipmidev/0: No such file or directory

root@witherspoon:/tmp# ./ipmitool chassis selftest
Could not open device at /dev/ipmi0 or /dev/ipmi/0 or /dev/ipmidev/0: No such file or directory

sivassrr, 2016-11-28:
root@witherspoon:/tmp# ./ipmitool -I dbus bmc selftest
Selftest: not implemented

sivassrr, 2016-11-28:
Looks like it is not implemented and based on the discussion, It seems down the line , it will be get implemented seems.

sivassrr, 2016-11-28:
root@witherspoon:/tmp# ./ipmitool -I dbus chassis selftest
Self Test Results    : not implemented

closed 2016-11-28 by sivassrr

williamspatrick, 2016-11-28:
We implemented a response to the IPMI command of "not supported". We have no plan of actually doing a self-test. If the BMC is up far enough to respond to IPMI requests then any sort of self-test has passed.

sivassrr, 2016-11-28:
@williamspatrick - ok then it is not going to add any value to do automation. Will remove it from our automation schedule.

cvam0000, 2023-03-22:
hey @williamspatrick still mc-selftest is in the planning phase or do we have something for it now ?

williamspatrick, 2023-03-22:
> hey @williamspatrick still mc-selftest is in the planning phase or do we have something for it now ?

I don't know of any plans to implement an `mc selftest`.