openbmc/openbmc #843: Compile shared library in phosphor-dbus-interfaces and export required hpp and yaml

geissonator, 2016-11-28:


geissonator, 2016-11-28:
Some discussion on this up here - https://gerrit.openbmc-project.xyz/#/c/1192/

shenki, 2016-11-29:
Hey @geissonator, can you add some notes as to what this shared library will do, and why we need it? The commit message would be a good place, or the bug here.

amboar, 2016-11-29:
I agree whole-heartedly with @shenki. I briefly looked at this issue and the patch and couldn't make much of either.

williamspatrick, 2016-11-29:
That commit itself probably has little to do with the story but there was some discussion in there about what the commit should have been.

The repository, phosphor-dbus-interfaces, is a collection of YAML files in a format understood by 'sdbus++' (from sdbusplus).  The tool will generate C++ bindings from the YAML for server, client, and errors (C++ exceptions).  Applications may then implement the server interfaces or call the client interfaces.  Rather than each application using sdbus++ to generate the code, we will link all of the generated bindings into a shared library.  Generating that shared library (and exporting associated headers) is the main purpose of this issue.

shenki, 2016-11-29:
I recall a discussion about the complexity of the shared library situation on other management controllers you've worked on. Are we sure we want to go down that path here?

williamspatrick, 2016-11-29:
That issue was on cross-"component" shared library dependencies that did not have any clear hierarchy relationship.  This shared library has a clear hierarchy.

sdbusplus itself has (or will have soon) a shared library.  There is still a clear layering of libraries here and not a nested web.  libc <- systemd <- sdbusplus <- phosphor-dbus-interfaces <- application

There should be no application <-> application coupling outside of dbus interfaces.

williamspatrick, 2016-12-10:
Code is availabe in Gerrit:
```
remote:   https://gerrit.openbmc-project.xyz/1475 build: ensure sdbus++ exists  (plus a few more commits)```

There is one bug in sdbusplus that currently prevents this from compiling without applying this patch:
```
diff --git a/xyz/openbmc_project/Object/Delete.interface.yaml b/xyz/openbmc_project/Object/Delete.interface.yaml
index bcddbeb..da9833e 100644
--- a/xyz/openbmc_project/Object/Delete.interface.yaml
+++ b/xyz/openbmc_project/Object/Delete.interface.yaml
@@ -1,7 +1,7 @@
 description: >
     Implement to provide object removal.
 methods:
-    - name: Delete
+    - name: Delete_
       description: >
           Delete the object implementing Delete.
```

See openbmc/sdbusplus#10.

shenki, 2016-12-12:
On Tue, Nov 29, 2016 at 12:19 PM, Patrick Williams
<notifications@github.com> wrote:
> That issue was on cross-"component" shared library dependencies that did not have any clear hierarchy relationship. This shared library has a clear hierarchy.
>
> sdbusplus itself has (or will have [...]

I don't think this alleviates the concerns.

Is there a good reason we need to have this complex layering of dependencies?

What does your proposed implementation provide over compiling the
interface code in the application directly?

williamspatrick, 2016-12-12:
Smaller flash footprint.  Single point of update.  Standard reasons for using a shared library, no?

williamspatrick, 2016-12-12:
The 'phosphor-dbus-interfaces' repository contains the collection of well-documented dbus interface files in YAML format.  Without a shared library I have to either:

1. Export generated header and cpp files into someplace like /usr/shared. Each application would need to depend on these generated files and compile them into their own application.
2. Each application would need to depend on the original YAML files, have the make invocation to generate the appropriate header and cpp files locally, and then compile them into their application.

What 'concerns' are you referring to here?  You are referring to past comments I made about shared libraries.  These issues were that the FSP code base regularly uses shared libraries, which are just collections of all the code in that "component" and which happens to call whatever other code in other shared libraries it decided to in order to accomplish its function.  The result is no hierarchy, no obvious dependencies, and instead just a tangled ball of every shared library dependent on every other shared library.  This causes the memory footprint of each application to be well beyond what is necessary and the load times to be similarly bad.

I don't have any problem with shared libraries themselves.  There simply needs to be a clear dependency in the shared library relationships such that there are never cycles.

shenki, 2016-12-12:
On Mon, Dec 12, 2016 at 3:20 PM, Patrick Williams
<notifications@github.com> wrote:
>
> Each application would need to depend on the original YAML files, have the make invocation to generate the appropriate header and cpp files locally, and then compile them into their application.

This seems to be a sensible approach to me.

I think you will end up with a larger disk footprint with multiple
shared libraries, as each .so has it's own headers, as well as
representing a different file on disk (bytes to represent the
filename).

In addition your application startup time is increased as ld.so must
resolve the symbols from shared libraries.

On the other hand compiling the entire application in one executable
means all symbols are resolved at build time, and with link time
optimisation you can save code size.

williamspatrick, 2016-12-12:
Yocto uses prelink. 

The space savings is from not having the generated code in multiple applications. 

I can experiment with an archive instead of shared library if you think LTO is really going to save that much.

geissonator, 2017-01-19:
https://gerrit.openbmc-project.xyz/#/c/1472/

closed 2017-01-19 by geissonator