openbmc/openbmc #846: SDK does not support python yaml library

adamliyi, 2016-11-29:
I built SDK using latest openbmc (git: b51d9f10b1bd309a5ee34a6fcfba334a9ff7ebae). When building phosphos-inventory-manager, I got error:
```
make
  GEN      generated.cpp
Traceback (most recent call last):
  File "./pimgen.py", line 7, in <module>
    import yaml
ImportError: No module named yaml
make: *** [generated.cpp] Error 1
```
Have to do:
```
export PYTHONPATH=/opt/openbmc-phosphor/2.1/sysroots/armv5e-openbmc-linux-gnueabi/usr/lib/python2.7/site-packages/
```
to continue.

msbarth, 2016-12-08:
This sounds like the pythondir variable was not set as that should point to the site-packages dir within the SDK. Also, was the python set during configure used within the makefile?

williamspatrick, 2016-12-09:
I'm pretty sure everything was resolved before. At one point we were missing the yaml package. Closing.

closed 2016-12-09 by williamspatrick

anon99, 2017-08-03:
I tried building "bitbake obmc-phosphor-image" using a tree acquired using the zipfile and got this message, so I then git cloned the repo and have the same problem. [ImportError: No module found yaml]
in ...sdbusplus/1.0-r1/build/example]. I tied the workaround export PYTHONPATH but it didnt seem to help.
Does a "clean" build of HEAD really work?. Or is there some tweak/workaround (or something I'm missing.