openbmc/openbmc #847: phosphor-inventory-manager build failure

adamliyi, 2016-11-29:
I can understand there are still patches for PIM to be merged - this issue may disappear as more patches for PIM are merged.

I tried to build PIM (https://github.com/openbmc/phosphor-inventory-manager/commit/e30dd7788547c1e0762606bfa715934e60231e7e) using latest SDK, but failed. Attached the build failure I met. Maybe useful for debugging.
[PIM_build_fail.txt](https://github.com/openbmc/openbmc/files/619129/PIM_build_fail.txt)

spinler, 2016-12-01:
@bradbishop Have you seen this?

adamliyi, 2016-12-02:
Update: tried with latest PIM commit: https://github.com/openbmc/phosphor-inventory-manager/commit/451f8d931a7a10d34c03f4348f64c3057f8d34f3
```
make[2]: Entering directory `/home/adam/workdir/obmc_git_push/phosphor-inventory-manager/test'
  CXX      test.o
test.cpp: In function 'void runTests(phosphor::inventory::manager::Manager&)':
test.cpp:51:57: error: use of deleted function 'sdbusplus::message::message::message(const sdbusplus::message::message&)'
         auto cleanup = sdbusplus::message::message(reply);
                                                         ^
In file included from /opt/openbmc-phosphor/2.1/sysroots/armv5e-openbmc-linux-gnueabi/usr/include/sdbusplus/bus.hpp:6:0,
                 from /opt/openbmc-phosphor/2.1/sysroots/armv5e-openbmc-linux-gnueabi/usr/include/sdbusplus/server.hpp:3,
                 from ../manager.hpp:7,
                 from test.cpp:16:
/opt/openbmc-phosphor/2.1/sysroots/armv5e-openbmc-linux-gnueabi/usr/include/sdbusplus/message.hpp:51:5: note: declared here
     message(const message&) = delete;
     ^
In file included from /opt/openbmc-phosphor/2.1/sysroots/armv5e-openbmc-linux-gnueabi/usr/include/c++/5.3.0/cassert:43:0,
                 from test.cpp:18:
test.cpp:52:46: error: cannot convert 'sdbusplus::message::message' to 'sd_bus_message*' for argument '1' to 'int sd_bus_message_get_errno(sd_bus_message*)'
         assert(sd_bus_message_get_errno(reply) == 0);
                                              ^
make[2]: *** [test.o] Error 1
make[2]: Leaving directory `/home/adam/workdir/obmc_git_push/phosphor-inventory-manager/test'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/adam/workdir/obmc_git_push/phosphor-inventory-manager'
make: *** [all] Error 2
```

anoo1, 2016-12-14:
I incorrectly closed this issue with a commit intended for issue #874.

gtmills, 2018-05-31:
@adamliyi No update in over a year. Can this be closed?

closed 2018-06-14 by anoo1