openbmc/openbmc #848: extended/test_bios_update.robot fails with "Error -1 erasing chip"

micwalsh, 2016-11-29:
I ran extended/test_bios_update.robot against barp01 system.  It failed...

Nov 28 19:37:22 barreleye flash_bios.exe[735]: Starting flasher: bios,/tmp/barreleye.pnor,/org/openbmc/control/flash/bios,
Nov 28 19:37:22 barreleye flash_bios.exe[735]: flasher: /tmp/barreleye.pnor, BMC = 1, address = 0x0
Nov 28 19:37:22 barreleye flash_bios.exe[735]: Setting up flash
Nov 28 19:37:22 barreleye flash_bios.exe[735]: Erasing... (may take a while !) Error -1 erasing chip

[bare-01.test_bios_update.161129.152224.status.txt](https://github.com/openbmc/openbmc/files/620320/bare-01.test_bios_update.161129.152224.status.txt)

micwalsh, 2016-11-29:
Also, I should mention that doing the code update manually as follows worked fine:
pflash -E -f -i -p /tmp/barreleye.pnor

[log.txt](https://github.com/openbmc/openbmc/files/620372/log.txt)

shenki, 2016-11-30:
Your bug report must contain the OpenBMC commit that you reproduced the issue with. This can be found in the outout of `/etc/os-release`.

geissonator, 2016-11-30:
Mike was pinging me on this earlier so I had the info.  Fairly recent build.  Seems like our REST api path to update PNOR is broke but pflash still works fine. 

```
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-296"
VERSION_ID="v1.99.0-296-ge743b5a-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-296"
BUILD_ID="v1.99.0"
```

spinler, 2016-12-01:
@micwalsh   Your testcases don't collect BMC FFDC on fails?  Maybe I just don't see it?

spinler, 2016-12-01:
I'm captured the journal, but there isn't much else interesting other than what Mike first posted.

[journal_beye6.txt](https://github.com/openbmc/openbmc/files/625777/journal_beye6.txt)

spinler, 2016-12-02:
@shenki  How do we figure out why this call is failing?
```
	rc = arch_flash_erase_chip(bl);
	if(rc) {
		fprintf(stderr, "Error %d erasing chip\n", rc);
		return(rc);
	}
```
It recreates every time with Mike's testcase.  Is there any extra data I can capture?

geissonator, 2016-12-08:
@cyrilbur-ibm any thoughts on this one?  Seems like you've been the one recently in this area :)

shenki, 2016-12-08:
```
Nov 28 20:03:23 barreleye flash_bios.exe[735]: Starting flasher: bios,/tmp/barreleye.pnor,/org/openbmc/control/flash/bios,
Nov 28 20:03:23 barreleye flash_bios.exe[735]: flasher: /tmp/barreleye.pnor, BMC = 1, address = 0x0
Nov 28 20:03:23 barreleye flash_bios.exe[735]: Setting up flash
Nov 28 20:03:23 barreleye flash_bios.exe[735]: Erasing... (may take a while !) Error -1 erasing chip
```

We should add more debug output to flasher, such as what it passes to the `arch_flash_init` call, and what the return value of `arch_flash_init` is.

shenki, 2016-12-08:
```
uint8_t
flash(FlashControl* flash_control, enum flash_access chip, uint32_t address, char* write_file, char* obj_path) 
{
        int rc;
        printf("flasher: %s, BMC = %d, address = 0x%x\n", write_file, chip, address); 

        /* Prepare for access */
        rc = flash_access_setup(chip);
        if(rc) {
                return FLASH_SETUP_ERROR;
        }
```

@cyrilbur-ibm thinks that this is the wrong function to be calling, and we need to call `arch_flash_init`.

That does not explain why it ever worked though.

cyrilbur-ibm, 2016-12-09:
https://github.com/open-power/skiboot/blob/master/external/common/arch_flash_arm.c#L300

I'll send a patch to the skeleton if I can work it out how to gerrit.

Done.

cyrilbur-ibm, 2016-12-09:
https://gerrit.openbmc-project.xyz/#/c/1452/

shenki, 2016-12-15:
This is merged as of 1da3a353e58e6439f5b7c5a0634e1b382bb27bfc. Please re-test with `v1.99.0-322-g1da3a353e58e` or later.

closed 2017-10-26 by micwalsh