openbmc/openbmc #852: Zaius FSI debug support

GHF, 2016-11-30:
We need pdbg or OpenFSI support for Zaius to boot the host.

pdbg doesn't work out of the box because of open-power/pdbg#8 and because its GPIOs are hardcoded for some other system. Even with fixes for those, I haven't had any luck with it:

```
root@zaius:~# /tmp/pdbg -b fsi -p 0 getcfam 0x281c
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
getaddr: Error reading register
root@zaius:~# /tmp/pdbg -b fsi probe
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
Overall System Topology
=======================

BMC FSI Backend@0
    FSI2PIB@1000

CFAMs Present
=============

0: BMC FSI Backend@0

Processors/Chiplets Present
===========================
```

Using these pins:

```
#define FSI_CLK»»       208»    //GPIOAA0
#define FSI_DAT»»       32»     //GPIOE0
#define CRONUS_SEL»     126»    //GPIOP6
#define FSI_ENABLE»     24»     //GPIOD0
#define FSI_DAT_EN»     118»    //GPIOO6
```

GHF, 2016-11-30:
with a bit more patching (open-power/pdbg/pull/9), I think pdbg is working (?):

```
root@zaius:~# /tmp/pdbg -b fsi getcfam 0xc09
CRC error: 0x83ff00000
CRC error: 0x6
CRC error: 0x48
CRC error: 0x48
CRC error: 0x48
CRC error: 0x70
CRC error: 0x70
CRC error: 0x70
CRC error: 0x70
CRC error: 0x40
p0:0xc09: 0x00000000120d1049
```

Even so, I don't see how this putcfam command in this script works with pdbg: https://github.com/openbmc/openbmc/blob/0d6405c721af309ddceb004046d849287301ed42/meta-openbmc-machines/meta-openpower/common/recipes-phosphor/host/op-pdbg-host-control/start_host.sh

GHF, 2016-12-01:
I talked to Patrick about this. While pdbg is intended to be the userspace frontend for FSI communication, it does not yet have support for the OpenFSI kernel drivers as a backend. Thus, we have to stick with my Zaius-specific patches to SoftFSI pdbg.

Furthermore, those scripts were not tested and carry the ecmd syntax of "set range of bits to value," which pdbg does not support and also does not have similar read-modify-write functionality.

Finally, the scripts will not be run because they lack systemd services with appropriate dependency relationships that insert them into the boot sequence and the BitBake recipes to install the services and scripts into the BMC image.

apopple, 2016-12-01:
> Furthermore, those scripts were not tested and carry the ecmd syntax of "set range of bits to value," which pdbg does not support and also does not have similar read-modify-write functionality.

Right. Adding an optional mask for the put commands to allow read-modify-write is on my list of things to do still.

> Even so, I don't see how this putcfam command in this script works with pdbg: https://github.com/openbmc/openbmc/blob/0d6405c721af309ddceb004046d849287301ed42/meta-openbmc-machines/meta-openpower/common/recipes-phosphor/host/op-pdbg-host-control/start_host.sh

The sequence I've been using to start the SBE is:
root@w2:/tmp# pdbg-p9 -b fsi putcfam 0x900 0x1
p0:0x900: 0x0000000000000001
root@w2:/tmp# pdbg-p9 -b fsi putcfam 0x2801 0x80C00000
p0:0x2801: 0x0000000080c00000
root@w2:/tmp# pdbg-p9 -b fsi getcfam 0x2809           
p0:0x2809: 0x0000000080200000
The write to 0x900 puts the chip in async mode which is required for SoftFSI to be able to start the SBE although it shouldn't be causing the CRC errors you're seeing.

> p0:0xc09: 0x00000000120d1049
That looks like it's working as that's the expected value. You shouldn't be seeing any CRC errors though, my guess is it's an issue with the probing we do for secondary modules. Can you try the following:

./pdbg-p9 -b fsi putcfam 0x900 0x1
./pdbg-p9 -b fsi -a probe

Thanks!

GHF, 2016-12-01:
> That looks like it's working as that's the expected value. You shouldn't be seeing any CRC errors though, my guess is it's an issue with the probing we do for secondary modules.

That machine does have a second module plugged in, though neither module has working SBE so that putcfam always errors out ("putaddr: Error writing register"). I tested your commands on a different bringup machine with a single module Adrian provided and I still see CRC errors. In addition, it's very flaky:

```
root@zaius:~# pdbg -b fsi getcfam 0xc09
CRC error: 0x83ff00000
CRC error: 0x6
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0xffffffffa
p0:0xc09: 0x00000000120d1049
root@zaius:~# pdbg -b fsi putcfam 0x900 0x1
CRC error: 0x83ff00000
CRC error: 0x6
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0xffffffffa
CRC error: 0x6
putaddr: Error writing register
root@zaius:~# pdbg -b fsi putcfam 0x900 0x1
CRC error: 0x83ff00000
CRC error: 0x6
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0xffffffffa
CRC error: 0x6
putaddr: Error writing register
root@zaius:~# pdbg -b fsi putcfam 0x900 0x1
CRC error: 0x83ff00000
CRC error: 0x6
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0xffffffffa
p0:0x900: 0x0000000000000001
```

The probe command also shows errors:

```
root@zaius:~# pdbg -b fsi -a probe
CRC error: 0x83ff00000
CRC error: 0x6
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0xffffffffa
CRC error: 0x6
CRC error: 0x6
CRC error: 0x6
CRC error: 0x6
CRC error: 0x6
CRC error: 0x6
CRC error: 0x6
CRC error: 0x6
CRC error: 0x6
CRC error: 0x6
CRC error: 0x6
CRC error: 0x6
Overall System Topology
=======================

BMC FSI Backend@0
    FSI2PIB@1000

CFAMs Present
=============

0: BMC FSI Backend@0

Processors/Chiplets Present
===========================

Processor-ID 0: FSI2PIB@1000
```

Also without `-a` (I guess it's not really default like it says in "Run command on all possible processors/chips/threads (default)?"):

```
root@zaius:~# pdbg -b fsi probe
CRC error: 0x83ff00000
CRC error: 0x6
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0xffffffffa
Overall System Topology
=======================

BMC FSI Backend@0
    FSI2PIB@1000

CFAMs Present
=============

0: BMC FSI Backend@0

Processors/Chiplets Present
===========================

```

I didn't try kicking off SBE since I haven't ported the VCS workarounds part for Zaius yet.

apopple, 2016-12-02:
> Also without -a (I guess it's not really default like it says in "Run command on all possible processors/chips/threads (default)?")
Yeah, I need to update the documentation. Without any processors/chips/threads specified pdbg won't try and go find them as it's quite slow - it only probes for the specified targets.

It's likely the CRC errors are due to timing problems. I hadn't realised pdbg was being used more widely for P9 work so hadn't gotten around to pushing the P9 hacks out. I've cleaned them up a bit and pushed them to github - https://github.com/open-power/pdbg - hopefully I didn't break too much.

GHF, 2016-12-02:
Yep, we have a limited number of FSP2 boxes here and no OpenFSI yet, so your pdbg soft FSI work is critical to even powering on a P9. :)

```
root@zaius:~# pdbg -b fsi -d p9w putcfam 0x283a 0xdeadbeef
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
p0:0x283a: 0x00000000deadbeef
root@zaius:~# pdbg -b fsi -d p9w getcfam 0x283a
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
p0:0x283a: 0x00000000deadbeef
```

I patched in Zaius GPIOs on top of your new code again and it looks like the dataful read responses pass the CRC, but we still get some CRC errors for shorter responses (what is 0x7f?).

It looks like the only functional difference for us is `#define P9W_CLOCK_DELAY 20` from a value of 3 prior to https://github.com/open-power/pdbg/commit/23410a93cc70009094cbab22fda4063fa6879962#diff-4d0b1593d2b770d9b60c3c4af17be7e8R73

We had only tried stretching that delay to 6 previously, which didn't help. We noticed in the clocking code:
```
	for (j = 0; j < clock_delay; j++);
        for (i = 0; i < num_clks; i++) {
                write_gpio(pin, 0);
                write_gpio(pin, 1);
        }
	for (j = 0; j < clock_delay; j++);
```
that it blips the clock line low for a minimal amount of time, just to generate two edges. The delay seems to set the clock high time to 2 * CLOCK_DELAY, with no change to low time. Is that OK?

GHF, 2016-12-03:
A different problem I noticed is that I can't talk to the second socket with pdbg. This is what I'm seeing with `probe`:

```
root@zaius:~# pdbg -b fsi -d p9w -a probe
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
Overall System Topology
=======================

BMC FSI Backend@0
    FSI2PIB@1000

CFAMs Present
=============

0: BMC FSI Backend@0

Processors/Chiplets Present
===========================

Processor-ID 0: FSI2PIB@1000
```

If I add `-p 0` (let alone `-p 1`) or `-a`, getcfam fails and _not_ with a CRC error:

```
root@zaius:~# pdbg -b fsi -d p9w getcfam 0x283a
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
BMC FSI Backend@0
    FSI2PIB@1000
p0:0x283a: 0x0000000000000000
root@zaius:~# pdbg -b fsi -d p9w -p 0 getcfam 0x283a
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
BMC FSI Backend@0
    FSI2PIB@1000
getaddr: Error reading register
No valid targets found or specified. Try adding -p/-c/-t options to specify a target.
Alternatively run pdbg -a probe to get a list of all valid targets
root@zaius:~# pdbg -b fsi -d p9w -a getcfam 0x283a
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
CRC error: 0x7f
BMC FSI Backend@0
    FSI2PIB@1000
getaddr: Error reading register
No valid targets found or specified. Try adding -p/-c/-t options to specify a target.
Alternatively run pdbg -a probe to get a list of all valid targets
```
(I had added a `dump_targets(0, &targets[0]);` to get probe debugging information)

apopple, 2016-12-05:
> I patched in Zaius GPIOs on top of your new code again and it looks like the dataful read responses pass the CRC, but we still get some CRC errors for shorter responses (what is 0x7f?).
You shouldn't be getting any CRC errors, so something is wrong (I don't see any on either Witherspoon or P8). On CRC errors it prints the response received - a value of 0x7f indicates all ones which is typical of either a processor module that is powered down or a mux not working as '1' is the default state of the data pin due to the pull ups. I would double check your FSI_DAT_EN/FSI_ENABLE/CRONUS_SEL pin assignments in this case, although it is strange that it seems to be somewhat working so it may be some other problem.

> that it blips the clock line low for a minimal amount of time, just to generate two edges. The delay seems to set the clock high time to 2 * CLOCK_DELAY, with no change to low time. Is that OK?
I have not actually scoped the signal lines so I don't know what the clock looks like but it has been working fine on Witherspoon and P8. The delays were added for setup/hold times on the data lines rather than limiting the frequency. That said most of that was derived via trial-and-error rather than by scoping the lines so feel free to tweak timings, etc. For what it's worth I was seeing the same CRC errors on Witherspoon for CLOCK_DELAY < 20 so perhaps try bumping that up further.

> A different problem I noticed is that I can't talk to the second socket with pdbg. This is what I'm seeing with probe:
To get to the second module you need to set the cfam in async mode (set bit 0 of cfam register 0x900). That said you shouldn't be getting any CRC errors so I would focus on getting that fixed before attempting the second socket.

GHF, 2016-12-06:
> To get to the second module you need to set the cfam in async mode (set bit 0 of cfam register 0x900). That said you shouldn't be getting any CRC errors so I would focus on getting that fixed before attempting the second socket.

Yep, I have that async mode enabled (putcfam manually) and verified the GPIO assignments. This is the boot script I tested:

```
./fsi_mode.sh bmc                                                               
./p9_workaround.sh                                                              
./vcs.sh on                                                                     
$pdbg -b fsi -d p9w putcfam 0x283f 0x20000000                                   
$pdbg -b fsi -d p9w putcfam 0x900 0x1                                           
$pdbg -b fsi -d p9w putcfam 0x2801 0x80C00000
```

I tried tracing the calls and it looks like it's the higher target addresses that cause errors:

```
getcfam target=(null) addr=0x0000000000000800
getcfam target=BMC FSI Backend addr=0x0000000000000c09
getcfam target=BMC FSI Backend addr=0x0000000000080c09
getcfam target=BMC FSI Backend addr=0x0000000000100c09
getcfam target=BMC FSI Backend addr=0x0000000000180c09
getcfam target=BMC FSI Backend addr=0x0000000000200c09
CRC error: 0x7f
getcfam error. Response: 0xffffffff
getcfam target=BMC FSI Backend addr=0x0000000000280c09
CRC error: 0x7f
getcfam error. Response: 0xffffffff
getcfam target=BMC FSI Backend addr=0x0000000000300c09
CRC error: 0x7f
getcfam error. Response: 0xffffffff
getcfam target=BMC FSI Backend addr=0x0000000000380c09
CRC error: 0x7f
getcfam error. Response: 0xffffffff
getcfam target=BMC FSI Backend addr=0x0000000000400c09
CRC error: 0x7f
getcfam error. Response: 0xffffffff
```

Also, what arguments should I pass in order to talk only to p0? As I mentioned, I tried `-p 0` with and without async mode and it doesn't work either way.

apopple, 2016-12-07:
Ok, so that's some useful debug. It looks like it's getting errors probing secondary processor sockets. This could be because async mode is not enabled or some other issue which I haven't hit. I assume you continue to get the errors after running pdbg -b fsi -d p9w putcfam 0x900 0x1?

> Also, what arguments should I pass in order to talk only to p0? As I mentioned, I tried -p 0 with and without async mode and it doesn't work either way.

Actually I just noticed the bug/TODO here which I need to fix - processor selection is ignored for cfam operations when using the FSI backend. So adding -p0 wont do much at the moment, sorry for the confusion. Will try and get a fix out today or tomorrow, although I'm not sure it will fix the CRC issues as I haven't had enough hardware access to test pdbg on P9 extensively. If it's possible to get access to a box that would certainly help.

However if you're only interested in a single socket at this stage you can comment out these lines in fsi_backend_targets_init():
`	cfam_count += hmfsi_target_probe(&targets[0], &targets[1], MAX_TARGETS);
	for (i = 1; i < cfam_count; i++)
		target_class_add(&cfams, &targets[i], i);
`
As that will effectively prevent probing secondary masters which is where you are hitting the CRC failures above.

GHF, 2016-12-07:
> It looks like it's getting errors probing secondary processor sockets. This could be because async mode is not enabled or some other issue which I haven't hit. I assume you continue to get the errors after running pdbg -b fsi -d p9w putcfam 0x900 0x1?

Yes, it didn't seem to help even with the modules in async mode and clocked. At least we can rule out soft FSI signal integrity issues for the time being.

> If it's possible to get access to a box that would certainly help.

Patrick mentioned that Adrian Barrera can set up the Zaius machine we left with them at Austin for remote access and time sharing. Please do bug him about it as he is also keen on getting BMC pdbg working.

> However if you're only interested in a single socket at this stage you can comment out these lines...

That's helpful to know. For now I've just changed that error message to stderr so that scripts can ignore it (see pull request).

apopple, 2016-12-08:
> Patrick mentioned that Adrian Barrera can set up the Zaius machine we left with them at Austin for remote access and time sharing. Please do bug him about it as he is also keen on getting BMC pdbg working.

I bugged Adrian and got access to a single module system. Initially I wasn't seeing the issue but after power cycling the host it re-appeared. This might imply there is an init setting somewhere that we're missing as I have not encountered the issue on our Witherspoon systems.

That said the errors are only occurring during probe of secondary processor modules, so they can be safely ignored for the moment until we figure out what's going on.

GHF, 2016-12-15:
Adrian said yesterday he will be working with Alistair on secondary module support for pdbg.

I agree that the probe errors continue to be low priority.

apopple, 2017-01-06:
> That said the errors are only occurring during probe of secondary processor modules, so they can be safely ignored for the moment until we figure out what's going on.

So the CRC errors are due to an uninitialised setting on the hMFSI ports. The below script should establish good baseline settings for pdbg to access secondary processor modules:

```
#!/bin/sh
PDBG="./pdbg -b fsi -d p9w -p0"`

# Turn on async mode. We expect and can safely ignore CRC errors due
# to device probing until the hMFSI error recovery options are set
# below.
$PDBG putcfam 0x900 0x1 2>&1 | grep -v "CRC error"

# Set hMFSI error recovery
$PDBG putcfam 0x34b8 0x0000c000 2>&1 | grep -v "CRC error"

# Set hMFSI timings
$PDBG putcfam 0x3400 0xd0040110
$PDBG putcfam 0x3401 0xffff0000

# Enable hMFSI ports
$PDBG putcfam 0x3404 0xf0000000
```

spinler, 2017-01-06:
I'll add the above to vcs_workaround.sh

spinler, 2017-01-09:
I added 2 chip support to start_host.sh and vcs_workaround.sh with https://gerrit.openbmc-project.xyz/#/c/1704/  (Also pulls in a new pdbg)

gtmills, 2018-05-23:
@spinler @GHF Can this be closed?

closed 2019-04-27 by stale[bot]