openbmc/openbmc #853: [ Barreleye v1.99.0-301-g7734ca3 ] Barreleye system went into bad state after code update to #392

rahulmah, 2016-12-01:
Barreleye system went into unpingable state after code update to #392/v1.99.0-301-g7734ca3/20161130142626. 

Below is messages seen in serial console of the system before hitting the issue. Seem like there happen to be some NO space problem while writing.
 
```cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
Updating bmc...
Erasing block: 15/15 (100%)
Writing kb: 928/928 (100%)
Verifying kb: 928/928 (100%)
Updating u-boot-env...
Erasing block: 2/2 (100%)
Writing kb: 128/128 (100%)
Verifying kb: 128/128 (100%)
jffs2: notice: (1148) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
ls: /run/initramfs/image-*: No such file or directory
Flash update completed.
Remaining mounts:
tmpfs / tmpfs rw,nosuid,nodev,mode=755 0 0
dev /dev devtmpfs rw,relatime,size=248232k,nr_inodes=62058,mode=755 0 0
proc /proc proc rw,relatime 0 0
sys /sys sysfs rw,relatime 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
reboot: Restarting system```

Tried rebooting but still don't see system getting in pingable state.  Below is the data from serial console after pdu update.

```DRAM Init-DDR3
CBR0-1234567012345670123456701234567
CBR134Done
▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒```

Logs: 
[Serial Console logs during code update.txt](https://github.com/openbmc/openbmc/files/623739/Serial.Console.logs.during.code.update.txt)

[Code update execution.txt](https://github.com/openbmc/openbmc/files/623740/Code.update.execution.txt)

gkeishin, 2016-12-01:
**There are two visible problems I noticed here:**

`From Automation perspective:`

The warm reset closing the connection and the code update error out ( I'm looking into it)
However the code update will still continue to completion in general

`From Code update perspective:`

post activating the image and warm reset the system doesn't come up or hang in boot, that's something to be looked at. Probably @shenki  could help us here.

geissonator, 2016-12-01:
George got our systems recovered and reported our latest driver is ok.  May have been a network glitch here in our lab that caused the above issue.

gkeishin, 2016-12-01:
We are good here on  v1.99.0-302-g0d1a644 - ( https://openpower.xyz/job/openbmc-build/393/ )

amboar, 2016-12-02:
Several things strike me from your console output:

```
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
```

Presumably this is copying rwfs data to a tmpfs for restoration, but somehow we ran out of space. We can't really dig into that now but maybe we should look at adding better error handling to the update script.

```
ls: /run/initramfs/image-*: No such file or directory
```

As above, we should better handle this case.

rahulmah, 2016-12-08:
Closing this issue as we are not seeing this issue. We will reopen this issue if we see it again.

closed 2016-12-08 by rahulmah