openbmc/openbmc #855: Default Witherspoon fan speeds

spinler, 2016-12-02:
There is no fan control code yet, and the fans aren't defaulting to high enough speeds and things are melting.
As a workaround until the fan code is done, we can either
1. figure out the watchdog so it defaults to max (but that may take WDT inits), or
2. pull in the fan script that is floating around the lab

2 is quicker, and though we'll eventually have to figure out the watchdog we can do it later when there's more time

amboar, 2016-12-04:
@spinler What watchdog are you talking about here?

spinler, 2016-12-05:
@amboar The one that drives WDTRST1 (pin T22)

spinler, 2016-12-07:
https://gerrit.openbmc-project.xyz/#/c/1400/
https://gerrit.openbmc-project.xyz/#/c/1401/
https://gerrit.openbmc-project.xyz/#/c/1402/

closed 2017-01-05 by geissonator