openbmc/openbmc #859: /run 100% full after reboot due to journal logs

geissonator, 2016-12-02:
This one just suddenly popped up on BarE-01.  I'm not sure what was going on prior but the system had been ok up until today.  

When I got on, I noticed this when trying to do a code update:

```
update: --copy-files --clean-saved-files --no-restore-files
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00320000: 0x0a0a instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00320004: 0x2020 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00320008: 0x3c20 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x0032000c: 0x434f instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00320010: 0x4550 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00320014: 0x6d74 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00320018: 0x683c instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x0032001c: 0x3e6c instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00320020: 0x6165 instead
jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00320024: 0x6572 instead
jffs2: Further such events for this erase block will not be printed
jffs2: Cowardly refusing to erase blocks on filesystem with no valid JFFS2 nodes
jffs2: empty_blocks 63, bad_blocks 0, c->nr_blocks 64
mount: mounting /dev/mtdblock6 on /run/initramfs/rw failed: Input/output error
umount: can't unmount /run/initramfs/rw: Invalid argument
Skipping empty update of rwfs.
```

I then asked King to dediprog the chip.  I was also seeing the /run being full issue prior to the dediprog.

My theory is that maybe the bad mtd block is causing some sort of filesystem corruption when writing the logs out?

Here's the df showing /run 100% full
```
[barreleye]# df
Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     248232         0    248232   0% /dev
tmpfs                   249508    249256       252 100% /run
/dev/mtdblock5           14720     14720         0 100% /run/initramfs/ro
/dev/mtdblock6            4096       328      3768   8% /run/initramfs/rw
cow                       4096       328      3768   8% /
tmpfs                   249508         0    249508   0% /dev/shm
tmpfs                   249508         0    249508   0% /sys/fs/cgroup
tmpfs                   249508         0    249508   0% /tmp
tmpfs                   249508         4    249504   0% /var/volatile
```

I verified you can run a "journalctl --vacuum-size=1M" to clean up the issue but it comes back as soon as you reboot.

The issue are all these files out here:
```
[barreleye]# ls -al /run/log/journal/8ab46a14f30e4aea8c4772d38a182202/
total 243364
drwxr-s---    2 root     systemd-      1620 Nov 27 01:49 .
drwxr-sr-x    3 root     systemd-        60 Nov 27 01:49 ..
-rwxr-x---    1 root     systemd-       240 Nov 27 01:49 system.journal
-rwxr-x---    1 root     systemd-   3194880 Nov 27 01:49 system@08da04be9fee44cfadbd3eee4cae8400-0000000000000001-ffffffffffffffff.journal
-rwxr-x---    1 root     systemd-   3194880 Nov 27 01:49 system@08da04be9fee44cfadbd3eee4cae8400-0000000000000002-ffffffffffffffff.journal
-rwxr-x---    1 root     systemd-   3194880 Nov 27 01:49 system@08da04be9fee44cfadbd3eee4cae8400-0000000000000003-ffffffffffffffff.journal
-rwxr-x---    1 root     systemd-   3194880 Nov 27 01:49 system@08da04be9fee44cfadbd3eee4cae8400-0000000000000004-ffffffffffffffff.journal
...
```
Adriana was taking a look at a hexdump of the logs, didn't see anything obvious:

```
[barreleye]# hexdump -C /run/log/journal/8ab46a14f30e4aea8c4772d38a182202/system@08da04be9fee44cfadbd3eee4cae8400-0000000000000044-ffffffffffffffff.journal | more
00000000  4c 50 4b 53 48 48 52 48  00 00 00 00 01 00 00 00  |LPKSHHRH........|
00000010  02 00 00 00 00 00 00 00  20 30 bd af f1 8d 4c da  |........ 0....L.|
00000020  a0 97 09 81 e0 36 0f e6  8a b4 6a 14 f3 0e 4a ea  |.....6....j...J.|
00000030  8c 47 72 d3 8a 18 22 02  e1 60 f8 44 86 91 4b 3a  |.Gr..."..`.D..K:|
00000040  92 c6 f0 51 f6 c7 33 66  08 da 04 be 9f ee 44 cf  |...Q..3f......D.|
00000050  ad bd 3e ee 4c ae 84 00  f0 00 00 00 00 00 00 00  |..>.L...........|
00000060  10 bf 30 00 00 00 00 00  e0 15 00 00 00 00 00 00  |..0.............|
00000070  a0 5a 01 00 00 00 00 00  00 01 00 00 00 00 00 00  |.Z..............|
00000080  d0 14 00 00 00 00 00 00  b8 76 01 00 00 00 00 00  |.........v......|
...
```

FW Levels
```
[barreleye]# cat /etc/version 
20161201131642
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-302"
VERSION_ID="v1.99.0-302-g0d1a644-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-302"
BUILD_ID="v1.99.0"
```

geissonator, 2016-12-13:
Angel replaced our BMC flash chip and we still have this issue, so I'm at a loss on this one.

amboar, 2016-12-13:
@geissonator you've outlined several issues:

1. The rwfs cannot be mounted at update time due to bad magic
2. `/run` is fairly full
3. There are four 3MB of journal files lying around under `/run`

Do you also see 1. during normal system boot (as opposed to update)? Separately I imagine running `journalctl --vacuum-size=1M` will resolve 2. and 3 by removing ~11MB of unwanted data.

Have you tried other builds? If so, what builds? Were they generated from a clean tree? What happened with the image before it were applied to the system?

> Angel replaced our BMC flash chip and we still have this issue, so I'm at a loss on this one.

Which issue is this? This was to test your bad flash erase block hypothesis? If so what image did you apply?

shenki, 2016-12-14:
The output of dmesg will tell us the flash chips and also provide the kernel version

geissonator, 2016-12-16:
I think with the replacement of the flash chip, issue 1 is gone (or it was really just a side affect of this full filesystem issue).  

But the filling up of /run happens on every reboot with every recent build.  The builds are coming from openpower.xyz and are the same builds that work on other barreleye systems.  Even if we prune the journald files, they come back on the next reboot and fill up the filesystem again.  Please note that that I didn't list all of the files in  /run/log/journal/8ab46a14f30e4aea8c4772d38a182202/ above.  There were easily 30-40 of them after every reboot.  The systems is BarE01 if you guys want to jump on the serial console and look at anything.

amboar, 2016-12-19:
>  Even if we prune the journald files, they come back on the next reboot and fill up the filesystem again.

What does this mean? Are you saying that immediately after reboot `/run` is already full as if the files were there from the start? Or just that after some uptime `/run` becomes full?

amboar, 2016-12-19:
So I've looked at why `/run` is reporting as full. The problem is that we're spewing so much data into the journal at `info` level that we're filling up `/run` with logs during the boot process. To verify, I configured:

```
# grep notice /etc/systemd/journald.conf | grep -v '^#'
MaxLevelStore=notice
MaxLevelSyslog=notice
```
And on the following boot we have:

```
# df -h /run
Filesystem                Size      Used Available Use% Mounted on
tmpfs                   243.7M     30.2M    213.5M  12% /run
```

There are two approaches: 

1. Implement a policy either limiting `MaxLevelStore`, `RuntimeMaxUse` or `RuntimeKeepFree`. Unfortunately these are not configurable from the kernel commandline and must be done in `/etc/systemd/journald.conf` or under `/etc/systemd/journal.conf.d/`.
2. Find out why we log > 200MB of information to the journal before we've even reached a getty

vishwabmc, 2017-01-10:
I looked at 2 systems. One with /run sitting at 4% and other is this one with /run sitting 100%.
```
On a system that is sitting at 4%, I see the complete journal entries are inside "/run/log/journal/a391296ec4ec4912b9550b558f0d06e1/system.journal/```

```
On the system that is sitting at 100%, system.journal is EMPTY() and there are individual 3Meg files for each entry.```

So there seems to be some policy that is playing the trick here of not updating the correct file but creating multiple 3Meg files

vishwabmc, 2017-01-10:
I tried to look at the contents of each file and all they have in that 3 Meg file is just one readable line.

```journalctl --file=system@91c0c4a647ba448cbe36c56138485ff4-0000000000000035-ffffffffffffffff.journal
Jul 20 06:36:29 barreleye kernel: squashfs: version 4.0 (2009/01/31) Phillip Lougher
```

journalctl --file=system@91c0c4a647ba448cbe36c56138485ff4-000000000000003a-ffffffffffffffff.journal
```
Jul 20 06:36:29 barreleye kernel: Serial: 8250/16550 driver, 6 ports, IRQ sharing enabled
```

vishwabmc, 2017-01-10:
On 100% system,
```
root@barreleye:/run/log/journal/59727046d39a4baeb4e47a34efa38b4a# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-360"
VERSION_ID="v1.99.0-360-g96d8e17-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-360"
BUILD_ID="v1.99.0"
root@barreleye:/run/log/journal/59727046d39a4baeb4e47a34efa38b4a# 
```

On 4% system,
```
root@palmetto:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-362"
VERSION_ID="v1.99.0-362-g477d0db-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-362"
BUILD_ID="v1.99.0"
```

===========

On yet another system, I saw /run was 12% and that had some of the .journal files.

So it looks like its about how often do the journal files do get archived and on this 100% box, it seems immediate.

vishwabmc, 2017-01-10:
To add more, I tried to verify the journal logs and all of 'em FAILed

```root@barreleye:/run/log/journal/59727046d39a4baeb4e47a34efa38b4a# journalctl --verify
PASS: /run/log/journal/59727046d39a4baeb4e47a34efa38b4a/system.journal                                       
0177e8: Invalid entry realtime timestamp: 18446744073709551615                                               
0177e8: Invalid object contents: Bad message                                                                 
File corruption detected at /run/log/journal/59727046d39a4baeb4e47a34efa38b4a/system@91c0c4a647ba448cbe36c56138485ff4-000000000000004e-ffffffffffffffff.journal:0177e8 (of 3194880 bytes, 3%).
FAIL: /run/log/journal/59727046d39a4baeb4e47a34efa38b4a/system@91c0c4a647ba448cbe36c56138485ff4-000000000000004e-ffffffffffffffff.journal (Bad message)
0177d8: Invalid entry realtime timestamp: 18446744073709551615                                               
0177d8: Invalid object contents: Bad message                                                                 
File corruption detected at /run/log/journal/59727046d39a4baeb4e47a34efa38b4a/system@91c0c4a647ba448cbe36c56138485ff4-000000000000004d-ffffffffffffffff.journal:0177d8 (of 3194880 bytes, 3%).
FAIL: /run/log/journal/59727046d39a4baeb4e47a34efa38b4a/system@91c0c4a647ba448cbe36c56138485ff4-000000000000004d-ffffffffffffffff.journal (Bad message)
0177b8: Invalid entry realtime timestamp: 18446744073709551615                                               
0177b8: Invalid object contents: Bad message                                                                 
File corruption detected at /run/log/journal/59727046d39a4baeb4e47a34efa38b4a/system@91c0c4a647ba448cbe36c56138485ff4-000000000000004c-ffffffffffffffff.journal:0177b8 (of 3194880 bytes, 3%).
FAIL: /run/log/journal/59727046d39a4baeb4e47a34efa38b4a/system@91c0c4a647ba448cbe36c56138485ff4-000000000000004c-ffffffffffffffff.journal (Bad message)
0175e0: Invalid entry realtime timestamp: 18446744073709551615                                               
0175e0: Invalid object contents: Bad message                                                                 
File corruption detected at /run/log/journal/59727046d39a4baeb4e47a34efa38b4a/system@91c0c4a647ba448cbe36c56138485ff4-000000000000004b-ffffffffffffffff.journal:0175e0 (of 3194880 bytes, 2%).
FAIL: /run/log/journal/59727046d39a4baeb4e47a34efa38b4a/system@91c0c4a647ba448cbe36c56138485ff4-000000000000004b-ffffffffffffffff.journal (Bad message)
```

vishwabmc, 2017-01-10:
https://bugs.mageia.org/show_bug.cgi?id=9921
https://bugs.freedesktop.org/show_bug.cgi?id=64116
http://unix.stackexchange.com/questions/86206/journalctl-verify-reports-corruption
https://www.reddit.com/r/linux/comments/1y6q0l/systemds_binary_logs_and_corruption/

All centered around some workarounds until there is some *fsck* available.. They all kind of hint that playing with SystemMaxUse seem to alleviate the issue..

I however see that both systems use the default one ( commented out lines in both )

vishwabmc, 2017-01-10:
http://unix.stackexchange.com/questions/86206/journalctl-verify-reports-corruption

vishwabmc, 2017-01-10:
This one is interesting : https://bbs.archlinux.org/viewtopic.php?id=159090

vishwabmc, 2017-01-10:
I see all the corruptions are from 'system' and not from 'user logins'.

I assume https://access.redhat.com/solutions/2649461 may have some solutions but I see its open for customers..

vishwabmc, 2017-01-10:
Since /run is mounted 'tmpfs', replacing the flash would not solve the issue. So this looks like a classic journal issue that many others have already seen and reported.

I don't think so there is anything more here for me to debug further. May want to open an issue with systemd.

vishwabmc, 2017-01-11:
@williamspatrick I don't see anything more here for me to debug. Should I just open a systemd issue ?.

lgon, 2017-02-21:
The kernel was constantly sending out "Time has been changed" log messages.  Angel attempted to replace the battery, but was unable to.  The message stopped sending, so the battery was possibly loose.  The current version of systemd we're using sets the log level of this message as INFO. The latest version of systemd sets the log level as DEBUG.

geissonator, 2017-02-21:
Re-seating the battery stopped the message, which fixed this issue.  We'll eventually pull in the upstream fix to not log this type of fail so continuously.  Closing this out.

closed 2017-02-21 by geissonator