openbmc/openbmc #860: BMC not responding to IPMI command from Host

astallma, 2016-12-05:
System Type:   Barreleye
BMC version:
          ~# cat /etc/os-release
          ID=openbmc-phosphor
          NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
          VERSION=0.1.0 (jethro)
          VERSION_ID=v1.0.5-5-g6044148-dirty
          PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
          BUILD_ID=v1.0.5
Kernel level:
          # uname -a
          Linux barreleye 4.4.26-openbmc-4.4-20161021-1 #1 Fri Oct 21 10:32:53 UTC 2016 armv5tejl GNU/Linux

**Scenario:**  

Attempting to send "Add Partial eSEL" IPMI command from the Host to the BMC.   The BMC is ack'ing the request, but no response is received.    Suspicious entries in journal.    The "Reserve SEL" command which precedes the "Add Partial eSEL" command completes successfully.

**Expected Results:**

Even if the command data isn't correct/valid, I'm expecting some response from the BMC.

**Details:**

Data sent:
```
         Request(0A42): ReserveSel

          Response(0B42): ReserveSel
          RcSuccess
          03000000

         Request (32F0): PartialAddExtendedSelData
          03000000 
          00000100
          00DF0000
          00000000
          04000000
          00000011
          11111111
          11111111
          11111111
          11111111
          11111100
```

I turned on IPMI tracing (/usr/sbin/ipmid -d 0xff) and I can see the successful Reserve SEL command, but no additional output is seen for the Add Partial eSEL command:
```
          IPMI Incoming: Seq 0x04, NetFn 0x0a, CMD: 0x42 
          IPMI Handling RESERVE-SEL 0x0001
          IPMI Response:
          0x000000: 00 03 00                                        ...             
          Failed to call the method: Input/output error
          Dest: :1.1, Path: /org/openbmc/HostIpmi/1
```

Journal Log snippet:
          The two timestamps in this log match exactly with the times the host is attempting to send this IPMI command:
```
          +Oct 23 01:16:49 barreleye systemd[1]: btbridged.service: Main process exited, code=exited, status=244/n/a
          +Oct 23 01:16:49 barreleye systemd[1]: btbridged.service: Unit entered failed state.
          +Oct 23 01:16:49 barreleye systemd[1]: btbridged.service: Failed with result 'exit-code'.
          +Oct 23 01:16:49 barreleye systemd[1]: btbridged.service: Service hold-off time over, scheduling restart.
          +Oct 23 01:16:49 barreleye systemd[1]: systemd-hwdb-update.service: Cannot add dependency job, ignoring: Unit systemd-hwdb-update.service is masked.
          +Oct 23 01:16:49 barreleye systemd[1]: Stopped Phosphor OpenBMC BT to DBUS.
          +Oct 23 01:16:49 barreleye systemd[1]: Started Phosphor OpenBMC BT to DBUS.
          +Oct 23 01:17:11 barreleye systemd[1]: btbridged.service: Main process exited, code=exited, status=244/n/a
          +Oct 23 01:17:11 barreleye systemd[1]: btbridged.service: Unit entered failed state.
          +Oct 23 01:17:11 barreleye systemd[1]: btbridged.service: Failed with result 'exit-code'.
          +Oct 23 01:17:11 barreleye systemd[1]: btbridged.service: Service hold-off time over, scheduling restart.
          +Oct 23 01:17:11 barreleye systemd[1]: systemd-hwdb-update.service: Cannot add dependency job, ignoring: Unit systemd-hwdb-update.service is masked.
          +Oct 23 01:17:11 barreleye systemd[1]: Stopped Phosphor OpenBMC BT to DBUS.
+Oct 23 01:17:11 barreleye systemd[1]: Started Phosphor OpenBMC BT to DBUS.
```

spinler, 2016-12-06:
@cyrilbur-ibm Cryil, I think this is a btbridged fail... Can you tell what went wrong?

astallma, 2016-12-07:
I made some progress on this today by changing the host code.  I can now see the OpenBMC IPMI trace showing my Add Partial eSEL commands.   I corrected some length issues and am only sending the 16 byte SEL with the first Add partial eSEL command (I was previously sending the SEL + the first chunk of data).    So, I'm past this specific problem, but perhaps additional verification can be done so a malformed IPMI message doesn't cause these failures and give no response to the host.   I'm leaving this issue open to address that concern.

closed 2018-06-11 by astallma