openbmc/openbmc #861: [ Palmetto v1.99.0-309-ged181d5] Palmetto system not booting to HOST_BOOTED state

rahulmah, 2016-12-06:
Palmetto system is not booting to HOST_BOOTED after power on. Waited for almost 30 mins after power on.

This issue is not seen with freshly reboot system using PDU. But after some usage of the system we start seeing this issue again. Once this issue is hit, we are not able to take the system to HOST_BOOTED state unless we do system reboot using PDU

Journal Logs: 
[Journal_log_poweron_failure.txt](https://github.com/openbmc/openbmc/files/634730/Journal_log_poweron_failure.txt)

spinler, 2016-12-06:
@rahulmah Was there any hostboot console output?

spinler, 2016-12-07:
I tried ~30 boots on palm309 with no problems.  Can you ping me the next time you hit it?

rahulmah, 2016-12-08:
Are you looking for serial console log? I tried also to run power on and off 50 times and I don't see the issue. But when we ran our entire test bucket(where our power on and off test case runs after some 9 test cases) and we found the problem again stated hitting. So seem like due to some operation of previous TCs this problem started coming. Not able to make out - Which operation has lead the system to result in such failure.

spinler, 2016-12-08:
It was recreated.  There was no host console output. 
Here are the 9 tests that ran before the fail:
Test SSL Connection
Test non-SSL Connection to port 80
Create error log on single FRU
Create error log on two FRU
Create multiple error logs
Delete error log
Association with invalid FRU
Assocition with no FRU error event
Association unchanged after reboot

rahulmah, 2016-12-09:
@spinler : I spent couple of hours to figure out minimum step to hit this issue. And here are the steps. I tried them couple of times and it is consistent in hitting the issue.

**1. Powercycle system using PDU.** (This is needed, if you want to start on the system which is not booting to HOST_BOOTED state)
root@palmetto:~# obmcutil state
 = BMC_READY				<---- BMC comes at BMC ready state after power cycle
 
**2. Power on and off the system**
root@palmetto:~# obmcutil poweron   <------ Power on system
 = None
root@palmetto:~# obmcutil state 
 = HOST_BOOTED         <------ System reaches HOST_BOOTED state after power on
root@palmetto:~# obmcutil poweroff  <------ Power off system
 = None
root@palmetto:~# obmcutil state
 = HOST_POWERED_OFF             <------ System will reach HOST_POWERED_OFF state

**3. Now reboot system using /sbin/reboot**
root@palmetto:~# /sbin/reboot
root@palmetto:~# obmcutil state
 = BMC_READY           <--------- System will reach BMC ready state after reboot
  
**4. Power on the system** 
root@palmetto:~# obmcutil poweron
 = None 
root@palmetto:~# obmcutil state
 = HOST_BOOTING       <------- BMC will come to HOST_BOOTING and does not boot to HOST_BOOTED

spinler, 2016-12-09:
OK, so it looks like reboots are what is breaking things.  That probably has to do with GPIOs.

spinler, 2016-12-09:
I grabbed the GPIOs from /sys/class/gpio after a cold reset, and again after a reboot (after power on/off), diffed them, and the differences were:

fail case:
gpio324 FSI_CLK: 0
gpio333 PCIE_RESET: 1
gpio334 USB_RESET : 1
gpio344 FSI_ENABLE: 0

So I set those to their opposites, to match the AC cycle case, but no luck.

spinler, 2016-12-09:
@rahulmah Do you know when this last worked?

@amboar @shenki Can you think of anything in the kernel that wouldn't get cleaned up on a reboot that could affect the host, besides the GPIOs I mentioned above?

rahulmah, 2016-12-12:
The last working build which I know is v1.99.0-275-g63248b4.

amboar, 2016-12-13:
I will try to reproduce/bisect this when I get some time.

@spinler These GPIO problems are pretty nebulous without deeper knowledge of the hardware. I don't immediately know whether anything else would affect the power-on sequence. However, there may be a timing dependency between power-on and changing the values of those GPIOs. It's unclear to me at what point in the boot process you tried to change their state.

williamspatrick, 2016-12-13:
Based on the problem description, I suspect that this commit from skeleton would have been most likely.
```
commit 0475f65f5feb9381ef2710dca96b545546ccd944
Author: Yi Li <adamliyi@msn.com>
Date:   Tue Oct 25 13:19:59 2016 +0800

    op-pwrctl: Hold PCI resets until BootProgress passes BASE_INITIALIZATION state
```

Looks like that commit is in the "known working" level though.

Another potential culprit is this:
```
commit 5d4a54e8949995f3e49366f575fcb47f119beb52
Author: Xo Wang <xow@google.com>
Date:   Fri Dec 2 10:26:15 2016 -0800

    pytools/gpioutil: Don't set direction if already set
```

adamliyi, 2016-12-15:
@mine260309 also observed similar abnormal Power-on behavior on Palmetto - I will double check my PCIE reset patch.

legoater, 2016-12-15:
same symptoms on our Palmetto also.

adamliyi, 2016-12-16:
As pointed out by @gwshan and @amboar , the reset pin setting based on polarity is not correct. I committed a fix here: https://gerrit.openbmc-project.xyz/#/c/1531/ . The reset pin state is as expected now:
```
Dec 16 09:04:22 palmetto power_control.exe[744]: PowerControl: pgood: 0, setting pci reset PCIE_RESET to 0
Dec 16 09:04:26 palmetto power_control.exe[744]: PowerControl: pgood: 1, setting reset USB_RESET to 1
Dec 16 09:04:26 palmetto power_control.exe[744]: PowerControl: pgood: 1, setting pci reset PCIE_RESET to 1
Dec 16 09:04:32 palmetto power_control.exe[744]: PowerControl: pgood: 0, setting reset USB_RESET to 0
Dec 16 09:04:32 palmetto power_control.exe[744]: PowerControl: pgood: 0, setting pci reset PCIE_RESET to 0
Dec 16 09:05:00 palmetto power_control.exe[744]: PowerControl: pgood: 1, setting reset USB_RESET to 1
Dec 16 09:05:00 palmetto power_control.exe[744]: PowerControl: pgood: 1, setting pci reset PCIE_RESET to 1
Dec 16 09:05:34 palmetto power_control.exe[744]: PowerControl: pgood: 0, setting reset USB_RESET to 0
Dec 16 09:05:34 palmetto power_control.exe[744]: PowerControl: pgood: 0, setting pci reset PCIE_RESET to 0
Dec 16 09:05:37 palmetto power_control.exe[744]: PowerControl: pgood: 1, setting reset USB_RESET to 1
Dec 16 09:05:37 palmetto power_control.exe[744]: PowerControl: pgood: 1, setting pci reset PCIE_RESET to 1
Dec 16 09:06:08 palmetto power_control.exe[744]: PowerControl: pgood: 0, setting reset USB_RESET to 0
Dec 16 09:06:08 palmetto power_control.exe[744]: PowerControl: pgood: 0, setting pci reset PCIE_RESET to 0
```
But even this fix, palmetto still cannot boot to host: the way I reproduce this issue is:

```
1. AC re-cycle, BMC root. run 'obmcutil poweron'. Host can boot normally to petiboot
2. on bmc: 'obmcuitl poweroff'
3. on bmc: 'reboot'
4. after bmc reboot: 'obmcutil pweron' - host will fail to boot.
```

amboar, 2016-12-19:
@adamliyi Is the step of rebooting the BMC required to reproduce the issue?

adamliyi, 2016-12-19:
@amboar, on my side - yes. If I do not reboot BMC. 'obmcutil poweron' works well.
```
1. AC re-cycle, BMC root. run 'obmcutil poweron'. Host can boot normally to petiboot
2. on bmc: 'obmcuitl poweroff'
3. on bmc: 'obmcutil poweron'   - The host can boot correctly.
```

rahulmah, 2016-12-19:
@amboar : Yes, reboot command is required to hit this issue.

amboar, 2016-12-19:
@adamliyi @rahulmah Okay, that's an interesting data point. Thanks.

SrideviRamesh, 2016-12-20:
@amboar : With latest palmetto (#419/20161219141046), I am seeing problem with first power on.  System is not booting to HOST_BOOTED state.  It continuously toggling around following states after "obmcutil poweron"(see attached bmc system info).
HOST_BOOTING, 
HOST_POWERING_ON, 
HOST_POWERED_ON, 
HOST_POWERING_OFF 

[20161220014255018974_MyFFDCLogs.zip](https://github.com/openbmc/openbmc/files/663337/20161220014255018974_MyFFDCLogs.zip)

Please find attached journal logs. Let me know if this is a same issue or I need to open a new issue.

root@palmetto:~# watch -n 30 obmcutil state
Every 30s: obmcutil state                                                                                                                                                        2016-11-21 08:16:18

 = HOST_BOOTING
Every 30s: obmcutil state                                                                                                                                                        2016-11-21 08:16:50

 = HOST_BOOTING
Every 30s: obmcutil state                                                                                                                                                        2016-11-21 08:17:21

 = HOST_BOOTING
Every 30s: obmcutil state                                                                                                                                                        2016-11-21 08:17:55

 = HOST_POWERING_ON
Every 30s: obmcutil state                                                                                                                                                        2016-11-21 08:18:29

 = HOST_BOOTING
Every 30s: obmcutil state                                                                                                                                                        2016-11-21 08:19:00

 = HOST_BOOTING
Every 30s: obmcutil state                                                                                                                                                        2016-11-21 08:19:32

 = HOST_BOOTING
Every 30s: obmcutil state                                                                                                                                                        2016-11-21 08:20:04

 = HOST_BOOTING
Every 30s: obmcutil state                                                                                                                                                        2016-11-21 08:20:36

 = HOST_POWERED_ON
Every 30s: obmcutil state                                                                                                                                                        2016-11-21 08:21:10

 = HOST_BOOTING
Every 30s: obmcutil state                                                                                                                                                        2016-11-21 08:21:42

 = HOST_BOOTING
Every 30s: obmcutil state                                                                                                                                                        2016-11-21 08:22:13

 = HOST_BOOTING

SrideviRamesh, 2017-01-20:
Palmetto system not booting to HOST_BOOTED state same issue is seen with latest BMC "v1.99.0-421"

please find FFDC, journal, serial console logs in the attachment

[journal_palm309.txt](https://github.com/openbmc/openbmc/files/718886/journal_palm309.txt)
[20170120005733026761_powerontest.zip](https://github.com/openbmc/openbmc/files/718890/20170120005733026761_powerontest.zip)
[palmetto_serial_console.txt](https://github.com/openbmc/openbmc/files/718893/palmetto_serial_console.txt)

rahulmah, 2017-01-30:
Please provide update on this issue. When are we expected to see fix for this defect?

SrideviRamesh, 2017-02-06:
Please provide update on this issue...still observing the issue with latest feb release drivers.
let u know if any required

rahulmah, 2017-02-15:
I think we can close this. Now with new xyz interface, we have HOST_RUNNING state instead of HOST_BOOTED state. I tried to wait for this new HOST_RUNNING state(instead of HOST_BOOTED) in step to recreate and I dont see the problem..

closed 2017-04-25 by rahulmah