openbmc/openbmc #862: Intermittently hitting org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic on way to BMC standby

geissonator, 2016-12-06:
There's been a variety of issues that list this as a side affect but no clear issue tracking it.  See #831, #832.  I figured it would be best to get one specifically for this problem.

It hits most often after a BMC reboot on the way to BMC_READY.  It seems to hit the org.openbmc.managers.System.service target the most. 

```
[barreleye]# systemctl status org.openbmc.managers.System.service | cat
\ufffd\ufffdorg.openbmc.managers.System.service - Phoshpor System Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.managers.System.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2016-11-21 06:09:40 UTC; 1h 39min ago
 Main PID: 725 (python)
   CGroup: /system.slice/org.openbmc.managers.System.service
           \ufffd\ufffd\ufffd\ufffd\ufffd\ufffd725 python /usr/sbin/system_manager.py

Nov 21 06:10:04 barreleye system_manager.py[725]: Traceback (most recent call last):
Nov 21 06:10:04 barreleye system_manager.py[725]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
Nov 21 06:10:04 barreleye system_manager.py[725]:     *message.get_args_list()))
Nov 21 06:10:04 barreleye system_manager.py[725]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 93, in <lambda>
Nov 21 06:10:04 barreleye system_manager.py[725]:     path, retry, x))
Nov 21 06:10:04 barreleye system_manager.py[725]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 105, in get_object_error
Nov 21 06:10:04 barreleye system_manager.py[725]:     self.error_callback(e)
Nov 21 06:10:04 barreleye system_manager.py[725]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 56, in default_error
Nov 21 06:10:04 barreleye system_manager.py[725]:     raise e
Nov 21 06:10:04 barreleye system_manager.py[725]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic
```

This fixes the issue when we hit it: systemctl restart org.openbmc.managers.System.service

The earliest occurrence of this seems to be in [Barreleye v1.99.0-291-g7a6b0ba] but due to it's intermittent nature, it's difficult to know exactly when it came in.  I'm still seeing it in code today:

```
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-305"
VERSION_ID="v1.99.0-305-g625ab37-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-305"
BUILD_ID="v1.99.0"
```

For debug, we need to look at the dbus during the failure time to see if any particular object is abusing the bus. If not, then we may need to look into some retry logic on this error.  

This is causing intermittent HW CI failures, false fails in CI are the worst kind of fails.

geissonator, 2016-12-07:
Hit today on a HW CI job with a different service:

```
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-309"
VERSION_ID="v1.99.0-309-ged181d5-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-309"
BUILD_ID="v1.99.0"
```

```
[barreleye]# systemctl status org.openbmc.managers.System.service | cat
\ufffd\ufffdorg.openbmc.managers.System.service - Phoshpor System Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.managers.System.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2016-11-21 06:09:29 UTC; 30min ago
 Main PID: 788 (python)
   CGroup: /system.slice/org.openbmc.managers.System.service
           \ufffd\ufffd\ufffd\ufffd\ufffd\ufffd788 python /usr/sbin/system_manager.py

Nov 21 06:10:12 barreleye system_manager.py[788]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
Nov 21 06:10:12 barreleye system_manager.py[788]:     *message.get_args_list()))
Nov 21 06:10:12 barreleye system_manager.py[788]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 93, in <lambda>
Nov 21 06:10:12 barreleye system_manager.py[788]:     path, retry, x))
Nov 21 06:10:12 barreleye system_manager.py[788]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 105, in get_object_error
Nov 21 06:10:12 barreleye system_manager.py[788]:     self.error_callback(e)
Nov 21 06:10:12 barreleye system_manager.py[788]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 56, in default_error
Nov 21 06:10:12 barreleye system_manager.py[788]:     raise e
Nov 21 06:10:12 barreleye system_manager.py[788]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic
Nov 21 06:10:15 barreleye system_manager.py[788]: New object: /org/openbmc/sensors/speed/fan5 (:1.14)
```

geissonator, 2016-12-08:
A full trace of it for settings and system manager

```
Nov 21 06:09:57 barreleye settings_manager.py[865]: Traceback (most recent call last):
Nov 21 06:09:57 barreleye settings_manager.py[865]:   File "/usr/sbin/settings_manager.py", line 245, in <module>
Nov 21 06:09:57 barreleye settings_manager.py[865]:     allobjects = create_object(s.SETTINGS)
Nov 21 06:09:57 barreleye settings_manager.py[865]:   File "/usr/sbin/settings_manager.py", line 71, in create_object
Nov 21 06:09:57 barreleye settings_manager.py[865]:     paths = mapper.get_subtree_paths(setting['subtree'], 0)
Nov 21 06:09:57 barreleye settings_manager.py[865]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 62, in get_subtree_paths
Nov 21 06:09:57 barreleye settings_manager.py[865]:     retries)
Nov 21 06:09:57 barreleye settings_manager.py[865]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 42, in retry
Nov 21 06:09:57 barreleye settings_manager.py[865]:     return func()
Nov 21 06:09:57 barreleye settings_manager.py[865]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 61, in <lambda>
Nov 21 06:09:57 barreleye settings_manager.py[865]:     path, depth, interfaces, signature='sias'),
Nov 21 06:09:57 barreleye settings_manager.py[865]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Nov 21 06:09:57 barreleye settings_manager.py[865]:     **keywords)
Nov 21 06:09:57 barreleye settings_manager.py[865]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Nov 21 06:09:57 barreleye settings_manager.py[865]:     message, timeout)
Nov 21 06:09:57 barreleye settings_manager.py[865]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic
```

```
Nov 21 06:10:07 barreleye system_manager.py[774]: Traceback (most recent call last):
Nov 21 06:10:07 barreleye system_manager.py[774]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
Nov 21 06:10:07 barreleye system_manager.py[774]:     *message.get_args_list()))
Nov 21 06:10:07 barreleye system_manager.py[774]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 93, in <lambda>
Nov 21 06:10:07 barreleye system_manager.py[774]:     path, retry, x))
Nov 21 06:10:07 barreleye system_manager.py[774]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 105, in get_object_error
Nov 21 06:10:07 barreleye system_manager.py[774]:     self.error_callback(e)
Nov 21 06:10:07 barreleye system_manager.py[774]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 56, in default_error
Nov 21 06:10:07 barreleye system_manager.py[774]:     raise e
Nov 21 06:10:07 barreleye system_manager.py[774]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic
```

mine260309, 2016-12-15:
I tried a couple of reboots with or without host booted.
The exception in `settings_manager.py` is reproduced twice, each occurs when:
1. BMC ready, and host is powered off;
2. BMC power on host until host is booted;
3. Reboot BMC;
4. During the first reboot, exception in `settings_manager.py` occurs.

But this seems not a problem because `org.openbmc.settings.Host.service` is configured as `Restart=always`, so it will be restarted and worked OK.

----
I did not reproduce the exception in `system_manager.py` yet.
But from the code `obmc/mapper/utils.py`, such exception will happen and it will retry:
```
        elif e.get_dbus_name() == 'org.freedesktop.DBus.Error.ObjectPathInUse':
            if retry > self.busy_retries:
                self.force_done()
                self.error_callback(e)
            else:
                gobject.timeout_add(
                    self.busy_retry_delay_milliseconds,
                    self.get_object_async,
                    path,
                    retry + 1)
```
The exception will not be raised until it has retried for `busy_retries` times.
So I guess such exception may not be avoided, we may investigate deeper, or increase the retires?

geissonator, 2016-12-15:
Thanks for digging into this one Lei!  I know one thing Patrick was interested in was a full log of the "dbus-monitor —system" during the initial BMC boot.  He'd like to look for any obvious issues with process's abusing the dbus.  If there's nothing obvious there, then I agree, we should probably just bump up the retries, and maybe increase our delay to give the bus more time to settle down.

mine260309, 2016-12-16:
With the steps, I managed to reproduce the issue once for both `system_manager.py` and `settings_manager.py`, and captured the `dbus-monitor --system` log.
[issue_862.tar.gz](https://github.com/openbmc/openbmc/files/656545/issue_862.tar.gz)

But in my case `system_manager.py` just gets an exception and keeps working, it seems to do no harm.

geissonator, 2016-12-16:
cat dbus-monitor.log | grep ObjectPathInUse | wc
     708  

So, we get that error 708 times here...that's a lot.

I don't see anything too obvious around the errors, it's all over the place.  The sender always seems to be 1.13, be interesting to know which object that is:

It also seems that this 1.13 is calling introspect on every object, which would generate a lot of dbus traffic for sure.

```
method call time=1476993510.857930 sender=:1.13 -> destination=:1.38 serial=274 path=/; interface=org.freedesktop.DBus.Introspectable; member=Introspect
error time=1476993510.875912 sender=:1.13 -> destination=:1.35 error_name=org.freedesktop.DBus.Error.ObjectPathInUse reply_serial=19
   string "busy processing bus traffic"
```

geissonator, 2016-12-16:
@bradbishop my guess is the object mapper is calling introspect on all objects as they appear on the bus?  Any thoughts on how to reduce it's impacts on the dbus?  Or do we just need more retries built in around this "busy bus" issue?

bradbishop, 2016-12-17:
The mapper falls back on introspection when something doesn't implement freedesktop.DBus.ObjectManager.  Implementing that would help ( I know settingsd doesn't ).

The dbus-monitor makes it clear:

```
signal time=1476993560.828886 sender=org.freedesktop.DBus -> destination=(null destination) serial=146 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameOwnerChanged
   string "org.openbmc.settings.Host"
   string ""
   string ":1.57"
method call time=1476993560.830047 sender=:1.13 -> destination=:1.57 serial=1184 path=/; interface=org.freedesktop.DBus.Introspectable; member=Introspect
method call time=1476993560.836385 sender=:1.58 -> destination=org.openbmc.ObjectMapper serial=4 path=/org/openbmc/ObjectMapper; interface=org.openbmc.ObjectMapper; member=GetObject
   string "/org/openbmc/settings/host0"
```

We are just deadlocked - NameOwnerChanged signal causes the mapper to introspect settingsd, and settingsd calls the mapper before starting its main loop (thus preventing itself from responding to the introspection call from the mapper).

To fix this, settingsd should not claim a busname until it is done creating objects.  A quick look at the code and it seems like that is what it does so I'm not sure what is going on there.

geissonator, 2016-12-19:
@mine260309 this is causing a _lot_ of instability issues in automation.  Could you try adding some more retries for now to org.openbmc.managers.System.service?

bradbishop, 2016-12-19:
I don't think retries are going to help.  I have some ideas to improve the situation...will work on that tomorrow.

mine260309, 2016-12-20:
@bradbishop Could you share your detailed idea so that I can work parallelly?

mine260309, 2016-12-20:
Based on the behavior that the exception occurs in setings_manager and system_manager, I tried the below changes as experiment:
1. In `settings_manager.py`, add a while loop to try-catch `mapper.get_subtree_paths()`;
2. In `obmc/mapper/utils.py`, decrease the retry timer to 500ms and increase the retry time to 20;
3. Add the logs of the exceptions.

With the changes, I rebooted Barreleye for a couple of times and all the reboots are OK and the obmc state goes into BMC_READY successfully.
And from journal log we can see that such exception does occur for a couple of times, but with the try-catch (or more retries) the exception is handled and eventually the code works OK.

The diff:
```
diff --git a/obmc/mapper/utils.py b/obmc/mapper/utils.py
index 6dd3750..3307420 100644
--- a/obmc/mapper/utils.py
+++ b/obmc/mapper/utils.py
@@ -34,10 +34,11 @@ class Wait(object):
         self.done = False
         self.callback = kw.pop('callback', None)
         self.error_callback = kw.pop('error_callback', self.default_error)
-        self.busy_retries = kw.pop('busy_retries', 5)
+        self.busy_retries = kw.pop('busy_retries', 20)
         self.busy_retry_delay_milliseconds = kw.pop(
-            'busy_retry_delay_milliseconds', 1000)
+            'busy_retry_delay_milliseconds', 500)
         self.waitlist_keyword = kw.pop('waitlist_keyword', None)
+        print 'MINEDBG: retries ' + str(self.busy_retries) + ", delay " + str(self.busy_retry_delay_milliseconds)

         self.bus.add_signal_receiver(
             self.name_owner_changed_handler,
@@ -104,6 +105,7 @@ class Wait(object):
                 self.force_done()
                 self.error_callback(e)
             else:
+                print "MINEDBG: delay and retry " + str(retry + 1)
                 gobject.timeout_add(
                     self.busy_retry_delay_milliseconds,
                     self.get_object_async,
```
```
diff --git a/settings_manager.py b/settings_manager.py
index 0205ea5..02e0e46 100644
--- a/settings_manager.py
+++ b/settings_manager.py
@@ -7,7 +7,8 @@ import dbus.mainloop.glib
 import os
 import os.path as path
 import sys
-from obmc.dbuslib.bindings import DbusProperties, get_dbus
+#from obmc.dbuslib.bindings import DbusProperties, get_dbus
+from obmc.dbuslib.bindings import DbusProperties, DbusObjectManager, get_dbus
 from IPy import IP

 settings_file_path = os.path.join(
@@ -68,7 +69,13 @@ def create_object(settings):
         for setting in settings.itervalues():
             if setting['type'] is not 'instance_query':
                 continue
-            paths = mapper.get_subtree_paths(setting['subtree'], 0)
+            while True:
+                try:
+                    paths = mapper.get_subtree_paths(setting['subtree'], 0)
+                except Exception as e:
+                    print 'MINEDBG: ' + str(e)
+                    continue
+                break

             if setting['keyregex'] == 'host':
                 # Always create at least one host object.
@@ -82,7 +89,7 @@ def create_object(settings):
                     "/org/openbmc/settings/" + m.group(1), settings)
     return allobjects

-class HostSettingsObject(DbusProperties):
+class HostSettingsObject(DbusProperties, DbusObjectManager):
     def __init__(self, bus, name, settings, path):
         super(HostSettingsObject, self).__init__(
             conn=bus,
```

geissonator, 2016-12-20:
Thanks Lei, lets get it up in gerrit for review.  Looks like a good workaround to me for now (except for the MINEDBG part of course :)

mine260309, 2016-12-21:
Submitted for review:
https://gerrit.openbmc-project.xyz/#/c/1571/
https://gerrit.openbmc-project.xyz/#/c/1572/

sivassrr, 2016-12-30:
Ran HWCI test on BarD-4 with Lei Yu patches for 15 Iterations and found all the time Basic Power On test fails. I believe that may not be related to Lei YU (@mine260309). 
[BarreleyeHWCIcombined15IterationLeiyupatch.pdf](https://github.com/openbmc/openbmc/files/678404/BarreleyeHWCIcombined15IterationLeiyupatch.pdf)

During the HWCI test, Run following command multiple times and didnt see any exception. 

root@barreleye:~# systemctl status org.openbmc.managers.System.service | cat; obmcutil state
\u25cf org.openbmc.managers.System.service - Phoshpor System Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.managers.System.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2016-11-21 06:09:24 UTC; 10min ago
 Main PID: 729 (python)
   CGroup: /system.slice/org.openbmc.managers.System.service
           \u2514\u2500729 python /usr/sbin/system_manager.py

Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:59 barreleye system_manager.py[729]: SystemManager Goto System State: BMC_READY
Nov 21 06:09:59 barreleye system_manager.py[729]: Running System State: BMC_READY
Nov 21 06:14:09 barreleye system_manager.py[729]: Running System State: HOST_POWERING_ON
Nov 21 06:19:17 barreleye system_manager.py[729]: Running System State: HOST_POWERING_OFF
 = HOST_POWERING_OFF
root@barreleye:~# systemctl status org.openbmc.managers.System.service | cat; obmcutil state
\u25cf org.openbmc.managers.System.service - Phoshpor System Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.managers.System.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2016-11-21 06:09:24 UTC; 10min ago
 Main PID: 729 (python)
   CGroup: /system.slice/org.openbmc.managers.System.service
           \u2514\u2500729 python /usr/sbin/system_manager.py

Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:59 barreleye system_manager.py[729]: SystemManager Goto System State: BMC_READY
Nov 21 06:09:59 barreleye system_manager.py[729]: Running System State: BMC_READY
Nov 21 06:14:09 barreleye system_manager.py[729]: Running System State: HOST_POWERING_ON
Nov 21 06:19:17 barreleye system_manager.py[729]: Running System State: HOST_POWERING_OFF
 = HOST_POWERING_OFF
root@barreleye:~# systemctl status org.openbmc.managers.System.service | cat; obmcutil state
\u25cf org.openbmc.managers.System.service - Phoshpor System Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.managers.System.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2016-11-21 06:09:24 UTC; 14min ago
 Main PID: 729 (python)
   CGroup: /system.slice/org.openbmc.managers.System.service
           \u2514\u2500729 python /usr/sbin/system_manager.py

Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:59 barreleye system_manager.py[729]: SystemManager Goto System State: BMC_READY
Nov 21 06:09:59 barreleye system_manager.py[729]: Running System State: BMC_READY
Nov 21 06:14:09 barreleye system_manager.py[729]: Running System State: HOST_POWERING_ON
Nov 21 06:19:17 barreleye system_manager.py[729]: Running System State: HOST_POWERING_OFF
 = HOST_POWERING_OFF
root@barreleye:~# systemctl status org.openbmc.managers.System.service | cat; obmcutil state
\u25cf org.openbmc.managers.System.service - Phoshpor System Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.managers.System.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2016-11-21 06:09:24 UTC; 14min ago
 Main PID: 729 (python)
   CGroup: /system.slice/org.openbmc.managers.System.service
           \u2514\u2500729 python /usr/sbin/system_manager.py

[... 101 lines cut ...]
Nov 21 06:09:59 barreleye system_manager.py[729]: SystemManager Goto System State: BMC_READY
Nov 21 06:09:59 barreleye system_manager.py[729]: Running System State: BMC_READY
Nov 21 06:14:09 barreleye system_manager.py[729]: Running System State: HOST_POWERING_ON
Nov 21 06:19:17 barreleye system_manager.py[729]: Running System State: HOST_POWERING_OFF
 = HOST_POWERING_OFF
root@barreleye:~# systemctl status org.openbmc.managers.System.service | cat; obmcutil state
\u25cf org.openbmc.managers.System.service - Phoshpor System Manager
[... cut ...]
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:58 barreleye system_manager.py[729]: MINEDBG: delay and retry 3
Nov 21 06:09:59 barreleye system_manager.py[729]: SystemManager Goto System State: BMC_READY
Nov 21 06:09:59 barreleye system_manager.py[729]: Running System State: BMC_READY
Nov 21 06:14:09 barreleye system_manager.py[729]: Running System State: HOST_POWERING_ON
Nov 21 06:19:17 barreleye system_manager.py[729]: Running System State: HOST_POWERING_OFF
 = HOST_POWERING_OFF
root@barreleye:~#

sivassrr, 2016-12-30:
root@barreleye:~# obmcutil state
Traceback (most recent call last):
 File "/usr/sbin/obmcutil", line 189, in <module>
   obj = bus.get_object(bus_name,obj_path)
 File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 241, in get_object
   follow_name_owner_changes=follow_name_owner_changes)
 File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 248, in __init__
   self._named_service = conn.activate_name_owner(bus_name)
 File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
   self.start_service_by_name(bus_name)
 File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
   'su', (bus_name, flags)))
 File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
   message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.managers.System was not provided by any .service files
root@barreleye:~#

[11:09]  
Just rebooted and then logged in and tried obmcutil state command.

[11:09]  
hit above error -- Checked with Lei YU and he recommended to wait till `org.openbmc.managers.System` service starts

mine260309, 2016-12-30:
@sivassrr Please remove all "MINEDBG" related lines in the log, they're just put on the test machine and is not part of the commit.

williamspatrick, 2016-12-30:
@sivassrr - Please edit your comments above and add 3 back-ticks around your log dumps.  It is really hard to read these when they are not a fix-spaced font and Github interprets various pieces as markdown.

sivassrr, 2017-01-02:
Thank you @williamspatrick  and learnt it.

sivassrr, 2017-01-04:
@mine260309 - As I tested using your sandbox, will let it be there. When we test officially those wouldnt come right. Hope it is fine.

sivassrr, 2017-01-04:
```

With driver 423, Could hit this on one of the witherspoon system.
{ret} = RequestsLibrary.RequestsKeywords . Get Request openbmc, ${base_uri}, &{kwargs}, timeout=${timeout}
Documentation: 	
[witherspoonDr423Iterations5TimeTCfails.pdf](https://github.com/openbmc/openbmc/files/685059/witherspoonDr423Iterations5TimeTCfails.pdf)

Send a GET request on the session object found using the
Start / End / Elapsed: 	20170104 16:26:42.973 / 20170104 16:26:43.340 / 00:00:00.367
16:26:43.340 	INFO 	Get Request using : alias=openbmc, uri=/org/openbmc/settings/host0/attr/time_mode, headers=None 	
16:26:43.340 	INFO 	${ret} = <Response [503]> 	
00:00:00.018 KEYWORD BuiltIn . Run Keyword If '${quiet}' == '${0}', Log Response, ${ret}
Documentation: 	

Runs the given keyword with the given arguments, if condition is true.
Start / End / Elapsed: 	20170104 16:26:43.341 / 20170104 16:26:43.359 / 00:00:00.018
00:00:00.007 KEYWORD rest_client . Log Response ${ret}
16:26:43.359 	INFO 	${resp} = <Response [503]> 	
00:00:00.001 KEYWORD ${content} = RequestsLibrary.RequestsKeywords . To Json ${resp.content}
16:26:43.361 	INFO 	${boot} = {u'description': u'org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic'} 	
00:00:00.001 KEYWORD BuiltIn . Should Be Equal ${boot}, Manual
Documentation: 	

Fails if the given objects are unequal.
Start / End / Elapsed: 	20170104 16:26:43.362 / 20170104 16:26:43.363 / 00:00:00.001
16:26:43.363 	INFO 	Argument types are:
<type 'dict'>
<type 'unicode'> 	
16:26:43.363 	FAIL 	{u'description': u'org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic'} != Manual

sivassrr, 2017-01-04:
[witherspoonDr423Iterations5TimeTCfails.pdf](https://github.com/openbmc/openbmc/files/685062/witherspoonDr423Iterations5TimeTCfails.pdf)

geissonator, 2017-01-10:
Same symptom, different error.  Restart of the service fixed the issue here as well:

```
Jan 10 23:41:43 barreleye system_manager.py[800]: Traceback (most recent call last):
Jan 10 23:41:43 barreleye system_manager.py[800]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
Jan 10 23:41:43 barreleye system_manager.py[800]:     *message.get_args_list()))
Jan 10 23:41:43 barreleye system_manager.py[800]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 93, in <lambda>
Jan 10 23:41:43 barreleye system_manager.py[800]:     path, retry, x))
Jan 10 23:41:43 barreleye system_manager.py[800]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 114, in get_object_error
Jan 10 23:41:43 barreleye system_manager.py[800]:     self.error_callback(e)
Jan 10 23:41:43 barreleye system_manager.py[800]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/utils.py", line 56, in default_error
Jan 10 23:41:43 barreleye system_manager.py[800]:     raise e
Jan 10 23:41:43 barreleye system_manager.py[800]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.LimitsExceeded: The maximum number of pending replies per connection has been reached
```

mine260309, 2017-01-11:
@geissonator Does this occur with or without the workaround fix?

geissonator, 2017-01-11:
Without your fixes.

mine260309, 2017-01-16:
Please be noted that this issue depends on https://gerrit.openbmc-project.xyz/#/c/1751/ to be merged as well.
Otherwise the changes in `phosphor-settingsd` will not run correctly.

sivassrr, 2017-01-27:
[obmcutilstatedbuserr.txt](https://github.com/openbmc/openbmc/files/734538/obmcutilstatedbuserr.txt)

With today's 483 driver, Could see this error. So Reopening it accordingly.

sivassrr, 2017-01-27:
Basically, I tried to run obmcutil state after I login.. .I see the dbus exception error.

geissonator, 2017-02-23:
I think we're all good here now, I haven't seen this in a while.  @sivassrr if you log in too quickly and run obmcutil state, you will get a dbus exception because the services are not running, but eventually they will come up and you will now always get to BMC_READY.

closed 2017-02-23 by geissonator