openbmc/openbmc #863: [Barreleye v1.0-stable] Connection timed out when power state change

johnhcwang, 2016-12-07:
Reproduce step:
1. Create 6 sessions to polling bmc version
```
$ while true; do time curl -b cjar -k https://192.168.2.199/org/openbmc/inventory/system/chassis/motherboard/bmc/attr/version; sleep 1s; done
```
2. Create one session to polling system state
```
$ while true; do time curl -b cjar -k https://192.168.2.199/org/openbmc/settings/host0/attr/system_state; sleep 1s; done
```
3. call /org/openbmc/control/chassis0/action/powerOn or powerOff

Sometimes we'll get the **_connection timed out_** and back to normal after 2~3 minutes.
That's means the user or RM can't know the real-time power state.

[image]

```
root@barreleye:~# dmesg
net_ratelimit: 2 callbacks suppressed
ftgmac100 1e660000.ethernet eth0: rx crc err
ftgmac100 1e660000.ethernet eth0: rx frame too long
ftgmac100 1e660000.ethernet eth0: rx crc err
ftgmac100 1e660000.ethernet eth0: rx frame too long
ftgmac100 1e660000.ethernet eth0: rx crc err
ftgmac100 1e660000.ethernet eth0: rx frame too long
ftgmac100 1e660000.ethernet eth0: rx crc err
ftgmac100 1e660000.ethernet eth0: rx frame too long
ftgmac100 1e660000.ethernet eth0: rx crc err
ftgmac100 1e660000.ethernet eth0: rx runt
ftgmac100 1e660000.ethernet eth0: rx crc err
ftgmac100 1e660000.ethernet eth0: rx frame too long
ftgmac100 1e660000.ethernet eth0: rx crc err
net_ratelimit: 2 callbacks suppressed
ftgmac100 1e660000.ethernet eth0: rx crc err
ftgmac100 1e660000.ethernet eth0: rx frame too long
```

anoo1, 2016-12-13:
Hi @gwshan , Is this something you may be able to help take a look? Kenneth clarified that the system was in a power on/off loop (for testing a different issue), and since powering on drops the network maybe it exacerbates this scenario.

gwshan, 2016-12-13:
I don't see those errors before. One thing is just as expected: the NCSI interface is temperarily unavailable when powering off and on the host. The NCSI stack running on the BMC side should received AEN packets to bring itself down and up again. However, 2 to 3 minutes consumed by that is more than I expected.

I will try to reproduce this on my local BMC first of all.

gwshan, 2017-02-06:
I tried to recreate this on palmetto, but not succeed. First of all, I think it's reasonable to operate the GPIO to do the power-on or power-off as we're not sure if the application (utility for power-on or off) is affecting other GPIO PINs.

What I did on palmetto is as below. Basically, network traffic (especially TCP) are sent to BMC while it's powering host on/off.

laptop# while true; do cat /tmp/data | nc palm8-bmc 1234; done
bmc# devmem 0x1e780020 32 0xDF48F7FD; \
          sleep 5;                                                   \
          devmem 0x1e780020 32 0xDF48F7FF; \
          nc -l -p 1234

johnhcwang, 2017-02-09:
@gwshan , did you create multi-sessions to REST service? I think that the REST service doesn't have too much loading in your test.

gwshan, 2017-02-16:
@johnhcwang, I didn't try REST. When debugging kernel issues, I usually build kernel image and load it via network (from uboot prompt). So I need one better way to recreate this issue. I think I need write up TCP connection program and integrate it to my buildroot based rootfs image. Ask the TCP server/client to communicate while issuing poweron and poweroff. @johnhcwang, do you think it's a doable way to recreate the issue?

johnhcwang, 2017-02-16:
I'm afraid that this is REST specific issue actually so that you cannot recreate.

gwshan, 2017-02-27:
@johnhcwang, I intend to agree with you and more precisely, it's likely to be something wrong with the GPIO configurations. Lets catch up to talk a bit about the issue.

I wrote some code (TCP server/client). With them, I started 100 TCP sessions during the NCSI is temporarily down and up. I didn't see any connection was lost.

gwshan, 2017-03-02:
I started 7 curl sessions as John said in comment#1 on Palmetto, in the meanwhile, I run below command and never hit the issue:

gwshan, 2017-03-02:
I started 7 curl sessions as John said in comment#1 on Palmetto, in the meanwhile, I run below command and never hit the issue:

root@palmetto:~# i=0; \
                             while [ $i -le 100 ] ; do echo ++++++++++$i; \
                             echo "   ON"; obmcutil poweron; sleep 60;   \
                             echo "   OFF"; obmcutil poweroff; sleep 10; \
                             let i++; done
            ...
++++++++++100
   ON
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
   OFF

gtmills, 2018-05-23:
@johnhcwang @anoo1 @gwshan Still a problem or can we close?

closed 2018-06-13 by anoo1