openbmc/openbmc #865: Factory Reset

geissonator, 2016-12-08:


geissonator, 2017-01-10:
Can we just wipe the write partitions so we go back to the defaults?

rfrandse, 2017-04-07:
make epic and create stories.

geissonator, 2017-05-08:
Design meeting notes:
- Talk with hostboot team, probably just going to clear all partitions that are r/w.  Maybe some need a reset instead of a clear?
- BMC code update + persistency design will drive what we do to BMC
- We may need to differentiate between a field reset and a mfg reset
-- For example, if mfg has reset system into "field" mode, we want a way to prevent resetting out of that
- 3 dbus interfaces
-- Reset Host (PNOR clear) (can be defined and implemented with pnor skeleton interface - @anoo1 )
-- Reset BMC network
-- Reset BMC (settings) - defer this one until BMC code update done

Break this into 4 stories.  1 to define generic interface, 3 to implement that interface for Host, BMC, and network.

whowutwut, 2017-08-29:
@geissonator More of a question, when running the FactoryReset function, do we expect that a default DHCP network config file will be generated in `/etc/systemd/network`?  

Doing this: 
```
[root@briggs01 ~]# rspconfig node-70e28414280d ip=dhcp
node-70e28414280d: [openbmc_debug] POST https: //172.11.140.5/login -d { "data":  [ "root", "0penBmc" ] }
node-70e28414280d: [openbmc_debug] login_response 200 OK
node-70e28414280d: [openbmc_debug] POST https: //172.11.140.5/xyz/openbmc_project/network/action/Reset -d {"data":[]}
node-70e28414280d: [openbmc_debug] rspconfig_dhcp_response 200 OK
node-70e28414280d BMC: Setting IP to DHCP...
node-70e28414280d: [openbmc_debug] PUT https: //172.11.140.5/xyz/openbmc_project/state/bmc0/attr/RequestedBMCTransition -d {"data":"xyz.openbmc_project.State.BMC.Transition.Reboot"}
node-70e28414280d: [openbmc_debug] rpower_reset_response 200 OK
node-70e28414280d BMC: reboot
```

This DOES work and the network config is reset and it's in DHCP mode, but no default  reset all network properties, but there's no default `00-bmc-eth0.network` file created.   Normal the file would contain something like: 
```
[Match]
Name=eth0
[Network]
DHCP=true
LinkLocalAddressing=yes
[DHCP]
ClientIdentifier=mac
```

mdmillerii, 2017-08-29:
I expect it to work because of `/usr/lib/systemd/network/default.network` added via `meta-phosphor` `systemd` recipe append.

Although its content is a underspecified from what you listed:
```
[Match]
Name=eth*
[Network]
DHCP=ipv4
```

whowutwut, 2017-08-29:
@mdmillerii  Ok, yeah, the content is probably something we copied from a while back.  Whatever it should be is OK..currently I see the following: 
* No file is generated in `/etc/systemd/network` that directory contains nothing 
* If we set a hostname via the REST API, that is also not cleared out, so all network settings are cleared, but hostname stays around after the reboot.  

Should I open issues for this?

closed 2017-11-03 by rfrandse