openbmc/openbmc #866: Use openFSI interfaces to start SBE on P8 and P9

geissonator, 2016-12-08:
Initially we're using the pdbg tool to initiate the power on sequence on witherspoon.  Eventually we'll want to use the openFSI interfaces directly in the obmc application code (for better error handling and reporting), and port that function to both P8 and P9 (the sequence will be different for P8 vs. P9 but the logic should be the same)

geissonator, 2016-12-08:
Stage 2 for now since that's where the most issues with chips will probably be found so better error handling on FSI issues would be expected.

geissonator, 2017-02-10:
op-hostctl in skeleton is the current P8 code.

spinler, 2017-02-14:
* [ ] Create new repository to hold the code
* [ ] Explore using FAPI2 framework
* [ ] Create new executable to start SBE (and maybe do a few other inits) via the OpenFSI equivalent of the 'putcfam' interfaces.   No dbus interfaces required.  When started will just execute the commands and exit.
* [ ] Create new executable to perform the VCS workaround on both processors.   If possible it will do an EC level check to know when to do it.  Can also perform the P9 clock mux workaround here, and some FSI inits for 2nd socket
* [ ] Update/create the appropriate service files/recipes so these get called during power ons

geissonator, 2017-03-16:
https://gerrit.openbmc-project.xyz/#/c/2985

closed 2017-03-16 by geissonator