openbmc/openbmc #867: Create shell script within debug tar to collect FFDC

geissonator, 2016-12-08:


mtritz, 2017-01-12:
- [ ] Determine requirements - which FFDC details should be collected and how?
- [ ] Identify preferred output format
- [ ] Assemble the script
- [ ] Integrate the completed script into the debug tar

mtritz, 2017-01-13:
In reference link above, the following items are collected.

DRIVER INFO:
Build Info
FW Level
BMC OS
BMC Uptime
BMC File System Disk Space Usage

APPLICATION DATA:
BMC state

BMC FILES:
BMC_proc_list
BMC_journalctl
BMC_dmesg
BMC_procinfo
BMC_meminfo

OS FILES:
OS_msglog
OS_cpufrequency
OS_dmesg
OS_boot
OS_procinfo
OS_meminfo
OS_netstat

OS FILES (VARIED BASED ON DISTRO):
OS_isusb
OS_kern
OS_authlog
OS_syslog
OS_info
OS_rsct
OS_secure

Is there anything missing from this list that would be useful for this script? Is there anything superfluous? My thinking is that for the files, especially the longer ones (tails of syslog, for instance), a file location might be more useful than direct output.

With regards to format, are there any requirements beyond a simple "Description: Info" list in a timestamped output file?

geissonator, 2017-01-13:
- Since this is focused only on BMC info, I think you can take off all OS_ items.
- I believe there's a systemctl command to get all process's in a bad state, that plus a "systemctl status <service> | cat" would be great info to have (i.e. get the status info for all process's in a bad state)
- Please also grab the console log, /var/log/obmc-console.log
- All the files in /var/lib/obmc/
- Error logs are coming soon, we'll want to grab those from the filesystem when ready

We'll want everything tar'd and zipped up in the end with the tar file named appropriately (commit-timestamp).tar.gz probably.  I don't think you need to do much other then add all the above data to individual files in the .tar.gz.

closed 2017-02-02 by geissonator