openbmc/openbmc #874: Incorrect response in Get BT Interface Capabilities Command - Input buffer size supported

astallma, 2016-12-09:
System Type: Barreleye
BMC version:
~# cat /etc/os-release
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v1.0.5-5-g6044148-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v1.0.5
Kernel level:  # uname -a
Linux barreleye 4.4.26-openbmc-4.4-20161021-1 #1 Fri Oct 21 10:32:53 UTC 2016 armv5tejl GNU/Linux

Scenario:
     Get BT Interface Capabilities Command is returning 0x40 as the max input buffer message size.   However, sending a message with a message buffer of this size is failing with unexpected results.   Based on the note below table 22, section 22.10 in the IPMI spec (v2.0) describing the Get BT Interface Capabilities Command, we believe the value being returned should be 0x3F to account for the size byte.

The note in the spec states: _"For a send, this means if the Get BT Interface Capabilities returns 255 in byte 3 (input buffer size) the driver can actually write 256 bytes to the input buffer (adding one for the length byte (byte 1)) that is sent in with the request.)"_

anoo1, 2016-12-09:
Fix for the stable branch: https://gerrit.openbmc-project.xyz/#/c/1473/

anoo1, 2016-12-14:
Merged in stable and master branch. Closing.

closed 2016-12-14 by anoo1