openbmc/openbmc #875: IPL: ConnectionError: HTTPSConnectionPool(host='beye6', port=443) on bard-04 with 920.1650.20171207a

swanman, 2016-12-09:
We hit this failure while testing 920.1650.20161207a on bard-04, a barreleye system.  There were 38 prior successful power on tests, so this is not a pervasive issue.  This is a shared system, so it is NOT sitting in the failed state.

I am noticing this error message in the journal output - not sure if related
Dec 08 20:48:52 barreleye phosphor-gevent[858]: error: [Errno 24] Too many open files

#(CST) 2016/12/08 14:41:51.107080 - 0.010289 - Checking every 3 seconds for up to 7 min for the state of the machine to match the state shown below.
match_state:
match_state[power]: ^1$
match_state[bmc]: ^HOST_BOOTED$
match_state[boot_progress]: ^FW Progress, Starting OS$
######################################################################################################Keyword 'Check State' failed after retrying for 7 minutes. The last error was: ConnectionError: HTTPSConnectionPool(host='beye6', port=443): Max retries exceeded with url: /login (Caused by ProtocolError('Connection aborted.', error(104, 'Connection reset by peer')))
#(CST) 2016/12/08 14:48:52.420493 - 373.847541 - BOOT_FAILED: BMC Power On failed. 

==========================================================================================
Copy this data to the defect:

openbmc_nickname: bard-04
openbmc_host: beye6
openbmc_username: root
openbmc_password: 0penBmc
os_host:
os_username: root
os_password: ********
pdu_host: rmtpdu17
pdu_username: admin
pdu_password: admin
pdu_slot_no: 1
openbmc_serial_host: Ran1
openbmc_serial_port: 2003
boot_stack:
boot_list: BMC Power On:BMC Power Off
max_num_tests: 600
plug_in_dir_paths: /afs/rch/projects/esw/dvt/autoipl/apollodev/bin/plug_ins/DB_Logging
status_file_path: /gsa/ausgsa/projects/a/autoipl/status/bard-04.161207.122046.006.status
openbmc_model: barreleye
boot_pass: 39
boot_fail: 1
test_mode: 0
quiet: 0
debug: 0
------------------------------------------------------------------------------------------
Last 10 boots:

#(CST) 2016/12/08 14:41:37.466243 - 0.065195 - Doing "BMC Power On".
------------------------------------------------------------------------------------------

FFDC data files:
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208144902194877_ObmcBootTest/20161208144902194877_RandomizedBootTesting/20161208144902194877_BMC_dmesg.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208144902194877_ObmcBootTest/20161208144902194877_RandomizedBootTesting/20161208144902194877_BMC_general.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208144902194877_ObmcBootTest/20161208144902194877_RandomizedBootTesting/20161208144902194877_BMC_journalctl.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208144902194877_ObmcBootTest/20161208144902194877_RandomizedBootTesting/20161208144902194877_BMC_meminfo.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208144902194877_ObmcBootTest/20161208144902194877_RandomizedBootTesting/20161208144902194877_BMC_procinfo.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208144902194877_ObmcBootTest/20161208144902194877_RandomizedBootTesting/20161208144902194877_BMC_proc_list.txt

========================================================================================== 

[20161208144902194877_BMC_dmesg.txt](https://github.com/openbmc/openbmc/files/642839/20161208144902194877_BMC_dmesg.txt)
[20161208144902194877_BMC_general.txt](https://github.com/openbmc/openbmc/files/642840/20161208144902194877_BMC_general.txt)
[20161208144902194877_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/642842/20161208144902194877_BMC_journalctl.txt)
[20161208144902194877_BMC_meminfo.txt](https://github.com/openbmc/openbmc/files/642841/20161208144902194877_BMC_meminfo.txt)
[20161208144902194877_BMC_proc_list.txt](https://github.com/openbmc/openbmc/files/642844/20161208144902194877_BMC_proc_list.txt)
[20161208144902194877_BMC_procinfo.txt](https://github.com/openbmc/openbmc/files/642843/20161208144902194877_BMC_procinfo.txt)

:

spinler, 2016-12-12:
The journal shows a lot of:
```
Dec 08 20:49:12 barreleye phosphor-gevent[858]: <WSGIServer at 0xb63d5370L fileno=5 address=('::20a4:f546:442c:a8b6%3058435856', 443, 0, 3058435856L)> failed with error
Dec 08 20:49:12 barreleye phosphor-gevent[858]: Traceback (most recent call last):
Dec 08 20:49:12 barreleye phosphor-gevent[858]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Dec 08 20:49:12 barreleye phosphor-gevent[858]:     result = self._run(*self.args, **self.kwargs)
Dec 08 20:49:12 barreleye phosphor-gevent[858]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Dec 08 20:49:12 barreleye phosphor-gevent[858]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Dec 08 20:49:12 barreleye phosphor-gevent[858]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Dec 08 20:49:12 barreleye phosphor-gevent[858]:     ciphers=ciphers)
Dec 08 20:49:12 barreleye phosphor-gevent[858]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 106, in __init__
Dec 08 20:49:12 barreleye phosphor-gevent[858]:     cert_reqs, ssl_version, ca_certs)
Dec 08 20:49:12 barreleye phosphor-gevent[858]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 31, in new_sslwrap
Dec 08 20:49:12 barreleye phosphor-gevent[858]:     context.load_cert_chain(certfile, keyfile)
Dec 08 20:49:12 barreleye phosphor-gevent[858]: IOError: [Errno 24] Too many open files
Dec 08 20:49:12 barreleye phosphor-gevent[858]: <Greenlet at 0xb4cbd5d0L: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0xb63d5370L fileno=5 address=('::20a4:f546:442c:a8b6%3058435856', 443, 0, 3058435856L)>>(<socket at 0xb4ccbad0L fileno=1023>, ('::c0b3:a6b6:8e91:f402%1190431048', 53252, 0, 119)> failed with IOError
Dec 08 20:49:13 barreleye phosphor-gevent[858]: Traceback (most recent call last):
Dec 08 20:49:13 barreleye phosphor-gevent[858]:   File "/usr/lib/python2.7/site-packages/gevent/baseserver.py", line 140, in _do_read
Dec 08 20:49:13 barreleye phosphor-gevent[858]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 93, in do_read
Dec 08 20:49:13 barreleye phosphor-gevent[858]:     client_socket, address = self.socket.accept()
Dec 08 20:49:13 barreleye phosphor-gevent[858]: error: [Errno 24] Too many open files
Dec 08 20:49:13 barreleye phosphor-gevent[858]: <WSGIServer at 0xb63d5370L fileno=5 address=('::20a4:f546:442c:a8b6%3058435856', 443, 0, 3058435856L)> failed with error
Dec 08 20:49:13 barreleye phosphor-gevent[858]: Traceback (most recent call last):
Dec 08 20:49:13 barreleye phosphor-gevent[858]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Dec 08 20:49:13 barreleye phosphor-gevent[858]:     result = self._run(*self.args, **self.kwargs)
Dec 08 20:49:13 barreleye phosphor-gevent[858]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Dec 08 20:49:13 barreleye phosphor-gevent[858]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Dec 08 20:49:13 barreleye phosphor-gevent[858]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Dec 08 20:49:13 barreleye phosphor-gevent[858]:     ciphers=ciphers)
Dec 08 20:49:13 barreleye phosphor-gevent[858]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 106, in __init__
Dec 08 20:49:13 barreleye phosphor-gevent[858]:     cert_reqs, ssl_version, ca_certs)
Dec 08 20:49:13 barreleye phosphor-gevent[858]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 31, in new_sslwrap
Dec 08 20:49:13 barreleye phosphor-gevent[858]:     context.load_cert_chain(certfile, keyfile)
Dec 08 20:49:13 barreleye phosphor-gevent[858]: IOError: [Errno 24] Too many open files```

swanman, 2016-12-14:
```
The following information was manually gathered by Andrew on a recreate today:

"ls -al /proc/*/fd/"  
see 3 process's with quite a few fd's, but nothing that looks overboard  

24:59 PM: root@barreleye:~# cat /proc/sys/fs/file-nr1259	0	49622 

for a rough count:
ls -al /proc/*/fd/ | wcls: /proc/2437/fd/3: cannot read link: No such file or directoryls: /proc/self/fd/3: cannot read link: No such file or directoryls: /proc/thread-self/fd/3: cannot read link: No such file or directory      850      6397     44604 
that only implies 850 open files, + 1259 from the kernel, so not seeing how we're even close to the 49622 max  
so hopefully that info plus your ffdc will help figure this one out

BMC FFDC automatically gathered from this fail:
/gsa/ausgsa/projects/a/autoipl/FFDC/20161214120041588419_ObmcBootTest/20161214120041588419_RandomizedBootTesting/20161214120041588419_BMC_dmesg.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161214120041588419_ObmcBootTest/20161214120041588419_RandomizedBootTesting/20161214120041588419_BMC_general.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161214120041588419_ObmcBootTest/20161214120041588419_RandomizedBootTesting/20161214120041588419_BMC_journalctl.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161214120041588419_ObmcBootTest/20161214120041588419_RandomizedBootTesting/20161214120041588419_BMC_meminfo.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161214120041588419_ObmcBootTest/20161214120041588419_RandomizedBootTesting/20161214120041588419_BMC_procinfo.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161214120041588419_ObmcBootTest/20161214120041588419_RandomizedBootTesting/20161214120041588419_BMC_proc_list.txt  
```

vishwabmc, 2017-02-14:
@swanman Please help get a zip of /proc when this happens ?. A connection reset by peer is a generic message but I am thinking having many open file descriptors is coming in the way of socket.

swanman, 2017-03-02:
We continue to hit this, so here is some fresh FFDC with what I could get from the tar command - kept erroring out before all the data was gathered:
'''
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.BMC_journalctl.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.BMC_dmesg.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.BMC_general.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.BMC_journalctl.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.BMC_meminfo.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.BMC_proc_list.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.BMC_procinfo.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.OS_authlog.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.OS_boot.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.OS_cpufrequency.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.OS_dmesg.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.OS_info.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.OS_isusb.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.OS_kern.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.OS_meminfo.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.OS_msglog.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.OS_netstat.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.OS_procinfo.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.OS_syslog.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/w15.170301.165721.SOL.txt

/gsa/ausgsa/projects/a/autoipl/FFDC/proc.tar.gz
'''

swanman, 2017-03-02:
Vishwa is taking a look at the live system.

geissonator, 2017-03-29:
Saw a system hit this today, looks like ipmid may be our culprit.

```
[witherspoon]# state
Traceback (most recent call last):
  File "/usr/sbin/obmcutil", line 174, in <module>
    bus = dbus.SystemBus()
  File "/usr/lib/python2.7/site-packages/dbus/_dbus.py", line 194, in __new__
    private=private)
  File "/usr/lib/python2.7/site-packages/dbus/_dbus.py", line 100, in __new__
    bus = BusConnection.__new__(subclass, bus_type, mainloop=mainloop)
  File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 122, in __new__
    bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.LimitsExceeded: The maximum number of active connections for UID 0 has been reached

[witherspoon]# ps -ef | grep -e dbus-daemon -e ipmid
  736 messageb   0:23 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation
 1837 root       0:02 /usr/sbin/ipmid
 2569 root       0:00 grep -e dbus-daemon -e ipmid
[witherspoon]# ls /proc/736/fd/ | wc
      267       267       958
[witherspoon]# ls /proc/1837/fd/ | wc
      227       227       798
```

```
[witherspoon]# systemctl restart org.openbmc.HostServices.service
[witherspoon]# 
[witherspoon]# 
[witherspoon]# ls /proc/736/fd/ | wc
       46        46       130

[witherspoon]# state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
```

closed 2017-03-31 by williamspatrick

mine260309, 2017-04-11:
I do not think we have fixed the root cause.
Below code snippet will cause fd leak every second:
```
    using namespace std::string_literals;
    static const auto intf = "xyz.openbmc_project.Inventory.Manager"s;
    static const auto path = "/xyz/openbmc_project/inventory"s;
    std::string service;
    while (true)
    {
        printf("Created a bus...\n");
        auto bus = sdbusplus::bus::new_default();
        auto mapperCall = bus.new_method_call("xyz.openbmc_project.ObjectMapper",
                                              "/xyz/openbmc_project/object_mapper",
                                              "xyz.openbmc_project.ObjectMapper",
                                              "GetObject");
        mapperCall.append(path);
        mapperCall.append(std::vector<std::string>({intf}));
        auto mapperResponseMsg = bus.call(mapperCall);
        std::this_thread::sleep_for(1s);
    }
```
Ideally, when `bus` is destructed, `sd_bus_unref()` is called and the bus connection shall be closed.

But due to https://github.com/systemd/systemd/issues/327 and https://github.com/poettering/systemd/commit/b898f9bd7eb803521ee8e5d51fe2174967541f86, systemd's library does not close the bus connection at all when we call a method.

Thus every loop consumes a fd in dbus-daemon and it's not released until the process exits.

> Queued but unwritten/unread messages also keep a reference
> to their bus connection object. For this reason, even if an
> application dropped all references to a bus connection it might
> not get destroyed right-away. Until all incoming queued
> messages are read, and until all outgoing unwritten messages are
> written, the bus object will stay
> alive. <function>sd_bus_flush()</function> may be used to write
> all outgoing queued messages so they drop their references. To
> flush the unread incoming messages use
> <function>sd_bus_close()</function>, which will also close the bus
> connection. When using the default bus logic it is a good idea to
> first invoke <function>sd_bus_flush()</function> followed by
> <function>sd_bus_close()</function> when a thread or process
> terminates, and thus its bus connection object should be
> freed.

* If `sdbusplus::bus::bus` is designed to use as freely as creating a dbus, call it, and close it, we should probably call `sd_bus_flush()` and `sd_bus_close()` to make sure the dbus connection is closed;
* If instead, it's designed only to be used for a service's event loop, a.k.a, only one bus for one service, then we have several miss uses of it which causes fd leak, e.g. the fixes in `phosphor-host-ipmid` and `ipmi-fru-parser`, and I can find others:
   1. **openpower-vpd-parser**
   2. **openpower-occ-control**
   3. **phosphor-logging**

They are probably the root cause of https://github.com/openbmc/openbmc/issues/1432

I would prefer fixing the issue by calling `sd_bus_flush()` and `sd_bus_close()` in `BusDeleter`, but it would be tricky and worth careful consideration since it may cause queued message to be dropped.

@williamspatrick please have a look at it.

williamspatrick, 2017-04-11:
Right. I am aware of this intrinsic problem. If you look at the history in sdbusplus I originally did a sd_bus_flush_close_unref but realized that doesn't actually work either for the reasons you mentioned. The sdbusplus code has no way currently to know if this is the one-and-only bus or if it is just an additional ref. 

The code change I made here was an attempt to get rid of all cases where we are creating a new bus on a per-transaction basis. We should not be doing this anyhow.

mine260309, 2017-04-11:
Then at least I know there are three places we use sdbusplus this way:
* openpower-vpd-parser
   Every call to `callPIM()` leaks 2 fds;
* openpower-occ-control
   Every call to `open_power::occ::finder::get()` leaks, although current it only calls this function once.
* phosphor-logging
   Every call to `logging::commit()` leaks.

Any particular reason why `sdbusplus::bus::bus` shall not be used as creating a new bus on a per-transaction basis?
From the interface, it can NOT be default constructed or copied. (ignoring the usage of `explicit bus(busp_t b);` and `busp_t release();`)
So the it will always be used as a standalone bus, and thus a standalone connection. If it's destructed, it's perfectly OK since no one is using the bus connection, and thus the messages queued can be ignored.
That's why I think `sd_bus_flush_close_unref()` should work and fix this issue?

williamspatrick, 2017-04-11:
> ignoring the usage of `explicit bus(busp_t b)`

That is a pretty big deal though.  There is a decent amount of code that gets a sd_bus* and temporarily creates a ::bus wrapper around it so they can use the nicer interfaces of sdbusplus.  In those cases we certainly cannot call `sd_bus_flush_close_unref`.  So that means I have to some how keep track of if the which mechanism this bus came from (new_* functions or direct constructor) and use an alternative BusDeleter.  Since BusDeleter is a static type I'd have to think about how to even make that possible.

There are two other issues.  

First, according to the docs, new_default may optionally return the same sd_bus object.  Unfortunately there is no sd_bus function like 'flush_close_unref_on_last'.  This means that a library function like phosphor-logging has no idea if the new_default it got was actually a unique new_default or the same new_default being used by the main event loop of the program.

Second, I'm not on the surface convinced it is always safe to call flush; I would need to investigate more deeply.  I've ran into other problems where certain bus methods cannot be called when we are already in a slot callback for that bus.  This is the reason why the constructor for bus gets the unique_value early rather than deferring that until needed.  I do not know if flush simply dumps all the messages or it attempts to process them all.  If it processes them all there is an issue.

Second,
> openpower-vpd-parser

This is a one-shot application so the leaks do not matter.

> openpower-occ-control

This is new code since I did the earlier review.  I don't expect this to be called multiple times.

> phosphor-logging

Agree this one could potentially be a problem.   I expect in most cases we are using phosphor-logging from within an application that also has a dbus event loop.  This means the leak should only be possible in applications which do not already have their own dbus event loop and are simply using phosphor-logging to get errors created.  I do not know of any case of this today that would be contributing to leaks.

mine260309, 2017-04-12:
I carefully checked #1432 log and ipmid code, it turns out that the root cause is indeed in `phosphor-logging`.

In `storageaddsel.cpp`, `send_esel_to_dbus()`
```
  1     log<level::INFO>("Received Host Event", entry("ESEL=%s", selData.get()));
  2
  3     try
  4     {
  5         elog<org::open_power::Error::Host::Event>(
  6             prev_entry<org::open_power::Error::Host::Event::ESEL>());
  7     }
  8     catch (org::open_power::Error::Host::Event& e)
  9     {
 10         commit(org::open_power::Error::Host::Event::err_code);
 11     }
```
We can see log `Received Host Event` in each power on/off cycle, it means the `try-catch` will be called on each cycle;
And in `commit()` (`phosphor-logging/elog.cpp`), it calls `sdbusplus::bus::new_default()` and cause the fd leak.

So after about 220 reboots, the issue is reproduced.
It **is** an issue.

mine260309, 2017-04-12:
To fix this issue, I'm proposing to have two types of constructors for `sdbusplus::bus::bus`
* If it's constructed from `new_default()` (or `new_user()`, `new_system()`), when destructed, unref, flush and close shall be called;
* If it's constructed from `explicit bus(busp_t b)` by user application, only unref shall be called when destructed.

Of course this means internally, `new_default()` will call a different version of `explicit bus(busp_t b)`.
Probably something like how `message` does:
```
explicit message(msgp_t m) : _msg(sd_bus_message_ref(m)) {}
message(msgp_t m, std::false_type) : _msg(m) {}
```

We can change bus's constructor to:
```
explicit bus(busp_t b);  # Use current BusDeleter
bus(busp_t b, std::false_type);  # Use a new BusDeleter, to unref, flush, and close
```

And this reminds me that the current usage of the explicit constructor requires the user application to add the reference counter:
```
sdbusplus::bus::bus bus{sd_bus_ref(sd_bus_pointer_from_elsewhere)};
```
I would prefer it's sdbusplus' responsibility to add the reference counter, which is consistent to `sdbusplus::message::message`.

@williamspatrick What do you think?

williamspatrick, 2017-04-12:
I realized that bus::new_default always creates a new bus and not reuse the per-thread default bus, so we can always safely close.  I still need to do the investigation on if flush is always allowed.

@mine260309 - I think your proposal makes sense.  The most difficult part is modifying the "Deleter" for the unique_ptr.  We can turn the deleter into a function pointer redirect that calls 'sd_bus_flush_close_unref' by default but calls 'sd_bus_unref' in the pointer-from-elsewhere case.  It adds effectively the same cost as a virtual function call.