openbmc/openbmc #876: IPL: OS is not pinging with the latest OpenBMC code on bard-04

swanman, 2016-12-09:
We routinely have not been able to do our Power On to OS tests on our barreleye system, bard-04, as the OS is not pinging.  When gsi checks the system, they have been able to correct the problem by putting on the GA level of openbmc code from rackspace (10 50 256)  .  This will obviously not work for our driver verification testing, as we need to be running with the latest daily builds.

Here is the FFDC we were able to gather from a run with 920.1650.20161207a yesterday.  This is a shared system, so it is NOT sitting at the failure.

#(CST) 2016/12/08 16:12:38.801956 - 0.009550 - Checking every 3 seconds for up to 14 min for the state of the machine to match the state shown below.
match_state:
match_state[power]: ^1$
match_state[bmc]: ^HOST_BOOTED$
match_state[boot_progress]: ^FW Progress, Starting OS$
match_state[os_ping]: ^1$
match_state[os_login]: ^1$
match_state[os_run_cmd]: ^1$
#############################Keyword 'Check State' failed after retrying for 14 minutes. The last error was:
#(CST) 2016/12/08 16:26:49.258254 - 37.894768 - **ERROR** The current state of the machine does NOT match the match state:
state:
state[power]: 1
state[bmc]: HOST_BOOTED
state[boot_progress]: FW Progress, Starting OS
state[os_ping]: 0
state[os_login]: 0
state[os_run_cmd]: 0

#(CST) 2016/12/08 16:26:49.283626 - 0.025372 - BOOT_FAILED: BMC Power On failed. 

==========================================================================================
Copy this data to the defect:

openbmc_nickname: bard-04
openbmc_host: beye6
openbmc_username: root
openbmc_password: 0penBmc
os_host: beye7
os_username: root
os_password: ********
pdu_host: rmtpdu17
pdu_username: admin
pdu_password: admin
pdu_slot_no: 1
openbmc_serial_host: Ran1
openbmc_serial_port: 2003
boot_stack:
boot_list: BMC Power On:BMC Power Off
max_num_tests: 600
plug_in_dir_paths: /afs/rch/projects/esw/dvt/autoipl/apollodev/bin/plug_ins/DB_Logging
status_file_path: /gsa/ausgsa/projects/a/autoipl/status/bard-04.161208.161031.status
openbmc_model: barreleye
boot_pass: 1
boot_fail: 1
test_mode: 0
quiet: 0
debug: 0
------------------------------------------------------------------------------------------
Last 10 boots:

#(CST) 2016/12/08 16:12:27.072175 - 0.068665 - Doing "BMC Power On".
------------------------------------------------------------------------------------------

FFDC data files:
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208162655043723_ObmcBootTest/20161208162655043723_RandomizedBootTesting/20161208162655043723_BMC_dmesg.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208162655043723_ObmcBootTest/20161208162655043723_RandomizedBootTesting/20161208162655043723_BMC_general.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208162655043723_ObmcBootTest/20161208162655043723_RandomizedBootTesting/20161208162655043723_BMC_inventory.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208162655043723_ObmcBootTest/20161208162655043723_RandomizedBootTesting/20161208162655043723_BMC_journalctl.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208162655043723_ObmcBootTest/20161208162655043723_RandomizedBootTesting/20161208162655043723_BMC_led.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208162655043723_ObmcBootTest/20161208162655043723_RandomizedBootTesting/20161208162655043723_BMC_meminfo.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208162655043723_ObmcBootTest/20161208162655043723_RandomizedBootTesting/20161208162655043723_BMC_procinfo.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208162655043723_ObmcBootTest/20161208162655043723_RandomizedBootTesting/20161208162655043723_BMC_proc_list.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208162655043723_ObmcBootTest/20161208162655043723_RandomizedBootTesting/20161208162655043723_BMC_record_log.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/20161208162655043723_ObmcBootTest/20161208162655043723_RandomizedBootTesting/20161208162655043723_BMC_sensor_list.txt

==========================================================================================

The following info was manually grabbed from the Host console by George, and it did not show any boot devices:

On BarD-04 -

root@barreleye:~# obmcutil state
= HOST_BOOTED
root@barreleye:~#

[9:13]  
Petitboot (v1.3.2-bcc819d)                   0000000000000000 0000000000000000
qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq

 System information
 System configuration
 Language
 Rescan devices
 Retrieve config from URL
*Exit to shell

qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq
Enter=accept, e=edit, n=new, x=exit, l=language, h=help
Info: Connected to pb-discover!

[Uploading 20161208162655043723_BMC_dmesg.txt…]()
[20161208162655043723_BMC_general.txt](https://github.com/openbmc/openbmc/files/642898/20161208162655043723_BMC_general.txt)
[20161208162655043723_BMC_inventory.txt](https://github.com/openbmc/openbmc/files/642897/20161208162655043723_BMC_inventory.txt)
[20161208162655043723_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/642899/20161208162655043723_BMC_journalctl.txt)
[20161208162655043723_BMC_led.txt](https://github.com/openbmc/openbmc/files/642900/20161208162655043723_BMC_led.txt)
[20161208162655043723_BMC_meminfo.txt](https://github.com/openbmc/openbmc/files/642901/20161208162655043723_BMC_meminfo.txt)
[20161208162655043723_BMC_proc_list.txt](https://github.com/openbmc/openbmc/files/642903/20161208162655043723_BMC_proc_list.txt)
[20161208162655043723_BMC_procinfo.txt](https://github.com/openbmc/openbmc/files/642902/20161208162655043723_BMC_procinfo.txt)
[20161208162655043723_BMC_record_log.txt](https://github.com/openbmc/openbmc/files/642904/20161208162655043723_BMC_record_log.txt)
[20161208162655043723_BMC_sensor_list.txt](https://github.com/openbmc/openbmc/files/642905/20161208162655043723_BMC_sensor_list.txt)

swanman, 2016-12-09:
Just noticed that the SOL output is not present in our FFDC.  I spoke with the test owners and they will work to get this included today.

amboar, 2016-12-12:
@swanman Please use the triple-backtick markdown when pasting logs into Github issues. This presents the log text in an easy to read manner and stops Github from mangling the content by trying to parse it as markdown for rendering.

```
It looks like this. You can have _underscores_ and **stars** without injecting unexpected formatting
```

spinler, 2016-12-12:
Stephanie said she's still waiting on having the testcase collect the host console output, and she'll recreate when it's there.

swanman, 2016-12-12:
I have uploaded the Host sol output:
[obmc-console.txt](https://github.com/openbmc/openbmc/files/647362/obmc-console.txt)

gwshan, 2016-12-12:
It seems a skiboot issue. No network card has been probed successfully in
skiboot.

[   22.023586474,5] PCI: Probing slots...
[   33.024048623,3] PHB#0001:02:04.0 Timeout waiting for downstream link
[   33.026576291,5] PHB#0000:00:00.0 [ROOT] 1014 03dc R:00 C:060400
B:01..ff SLOT=Slot1
[   33.026700815,5] PHB#0001:00:00.0 [ROOT] 1014 03dc R:00 C:060400
B:01..16 SLOT=PLX Switch
[   33.026827032,5] PHB#0001:01:00.0 [SWUP] 10b5 8718 R:ab C:060400
B:02..16 LOC_CODE=PLX Switch
[   33.026967739,5] PHB#0001:02:01.0 [SWDN] 10b5 8718 R:ab C:060400
B:03..07 SLOT=IO Board SATA
[   33.027103317,5] PHB#0001:02:02.0 [SWDN] 10b5 8718 R:ab C:060400
B:08..0c SLOT=IO Board USB
[   33.027237669,5] PHB#0001:02:03.0 [SWDN] 10b5 8718 R:ab C:060400
B:0d..11 SLOT=IO Board BMC
[   33.027367109,5] PHB#0001:02:04.0 [SWDN] 10b5 8718 R:ab C:060400
B:12..16 SLOT=IO Board NIC
[   33.027494374,5] PHB#0002:00:00.0 [ROOT] 1014 03dc R:00 C:060400
B:01..ff SLOT=Network Mezz
[   33.027618288,5] PHB#0020:00:00.0 [ROOT] 1014 03dc R:00 C:060400
B:01..ff SLOT=Storage Mezz
[   33.027740858,5] PHB#0021:00:00.0 [ROOT] 1014 03dc R:00 C:060400
B:01..ff SLOT=Slot3
[   33.027857230,5] PHB#0022:00:00.0 [ROOT] 1014 03dc R:00 C:060400
B:01..ff SLOT=Slot2

swanman, 2016-12-14:
Since we know this happens with newer openbmc code (with the same PNOR level), the openbmc folks need help figuring out what it is that they are not doing correctly to make this "skiboot problem" show up.  Any words of wisdom would be appreciated.

shenki, 2016-12-15:
I think there were PCIe GPIO related changes in openbmc recently. I suggest reverting those to see if the network card comes back.

shenki, 2016-12-15:
Please test older versions of openbmc master to work out when the regression happened. I suggest this one: https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=barreleye/374/artifact/images/barreleye/

swanman, 2016-12-15:
That level did not work either.

swanman, 2016-12-15:
```
# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-275"
VERSION_ID="v1.99.0-275-g63248b4-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-275"
BUILD_ID="v1.99.0"

 38.62086|ISTEP 21. 1
 52.49446|htmgt|OCCs are now running in ACTIVE state
 60.86782|ISTEP 21. 2
 60.86945|ISTEP 21. 3
[   61.146186381,5] SkiBoot skiboot-5.4.2 starting...
[   61.146191026,5] initial console log level: memory 7, driver 5
[   61.146194495,6] CPU: P8 generation processor(max 8 threads/core)
[   61.146197659,7] CPU: Boot CPU PIR is 0x0028 PVR is 0x004d0200
[   61.146201090,7] CPU: Initial max PIR set to 0x1fff
[   61.146727042,5] OPAL table: 0x300c0c40 .. 0x300c1110, branch table: 0x30002000
[   61.146733240,5] FDT: Parsing fdt @0xff00000
[   61.153498034,5] XSCOM: chip 0x0 at 0x3fc0000000000 [P8 DD2.0]
[   61.153510217,5] XSCOM: chip 0x8 at 0x3fc4000000000 [P8 DD2.0]
[   61.153519232,6] XSTOP: XSCOM addr = 0x2010c82, FIR bit = 31
[   61.153523127,6] MFSI 0:0: Initialized
[   61.153525754,6] MFSI 0:2: Initialized
[   61.153528293,6] MFSI 0:1: Initialized
[   61.153531187,6] MFSI 8:0: Initialized
[   61.153533676,6] MFSI 8:2: Initialized
[   61.153536179,6] MFSI 8:1: Initialized
[   61.153990493,5] LPC: LPC[000]: Initialized, access via XSCOM @0xb0020
[   61.154154881,5] LPC: LPC: Default bus on chip 0x0
[   61.154359420,6] MEM: parsing reserved memory from node /ibm,hostboot/reserved-memory
[   61.154372877,7] HOMER: Init chip 0
[   61.154376064,7]   PBA BAR0 : 0x0000003ffd800000
[   61.154379236,7]   PBA MASK0: 0x0000000000300000
[   61.154382293,7]   HOMER Image at 0x3ffd800000 size 4MB
[   61.154386477,7]   PBA BAR2 : 0x4000003ffda00000
[   61.154389635,7]   PBA MASK2: 0x0000000000000000
[   61.154392600,7]   SLW Image at 0x3ffda00000 size 1MB
[   61.154396624,7]   PBA BAR3 : 0x0000003fff800000
[   61.154399585,7]   PBA MASK3: 0x0000000000700000
[   61.154402609,7]   OCC Common Area at 0x3fff800000 size 8MB
[   61.154405997,7] HOMER: Init chip 8
[   61.154409053,7]   PBA BAR0 : 0x0000003ffdc00000
[   61.154412140,7]   PBA MASK0: 0x0000000000300000
[   61.154415153,7]   HOMER Image at 0x3ffdc00000 size 4MB
[   61.154419272,7]   PBA BAR2 : 0x4000003ffde00000
[   61.154422515,7]   PBA MASK2: 0x0000000000000000
[   61.154425464,7]   SLW Image at 0x3ffde00000 size 1MB
[   61.154429432,7]   PBA BAR3 : 0x0000003fff800000
[   61.154432532,7]   PBA MASK3: 0x0000000000700000
[   61.154435562,7]   OCC Common Area at 0x3fff800000 size 8MB
[   61.154449323,7] CPU idle state device tree init
[   61.154452829,4] SLW: HB-provided idle states property found
[   61.154689798,7] AST: PNOR LPC offset: 0x0c000000
[   61.154769440,5] PLAT: Using virtual UART
[   61.155138977,7] UART: Using LPC IRQ 4
[   61.158448811,5] PLAT: Detected Barreleye platform
[   61.163107444,5] CENTAUR: Found centaur for chip 0x0 channel 0
[   61.163204016,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 3
[   61.163304892,5] CENTAUR: Found centaur for chip 0x0 channel 1
[   61.163402469,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 2
[   61.163503695,5] CENTAUR: Found centaur for chip 0x0 channel 4
[   61.163600262,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 7
[   61.163704332,5] CENTAUR: Found centaur for chip 0x0 channel 5
[   61.163806285,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 6
[   61.163904164,5] CENTAUR: Found centaur for chip 0x8 channel 0
[   61.163994298,5] CENTAUR:   FSI host: 0x8 cMFSI0 port 3
[   61.164094782,5] CENTAUR: Found centaur for chip 0x8 channel 1
[   61.164189646,5] CENTAUR:   FSI host: 0x8 cMFSI0 port 2
[... 65 lines cut ...]
[   27.507485757,5]   0x003ffd6c0000..003ffd6fffff : ibm,hbrt-target-image@3ffd6c0000
[   27.507592434,5]   0x003ffd440000..003ffd6bffff : ibm,hbrt-code-image@3ffd440000
[   27.507693469,5]   0x000031c00000..000032ddffff : ibm,firmware-stacks
[   27.507781459,5]   0x000031000000..000031bfffff : ibm,firmware-data
[   27.507874671,5]   0x000030300000..000030ffffff : ibm,firmware-heap
[   27.507960431,5]   0x000030000000..0000302fffff : ibm,firmware-code
[   27.508057915,5]   0x002000000000..002000c401ff : ibm,firmware-allocs-memory@2000000000
[   27.508173087,5]   0x000039c00000..00003a8401ff : ibm,firmware-allocs-memory@0
[   27.508284490,5] IPMI: Resetting boot count on successful boot
[   27.508364610,5] INIT: Waiting for kernel...

 Petitboot (v1.3.2-db19513)                   0000000000000000 0000000000000000
 ??????????????????????????????????????????????????????????????????????????????

  System information
  System configuration
  Language
  Rescan devices
  Retrieve config from URL
 *Exit to shell           

 ??????????????????????????????????????????????????????????????????????????????
 Enter=accept, e=edit, n=new, x=exit, l=language, h=help
 Info: Connected to pb-discover!  
```

gwshan, 2016-12-15:
@swanman, if possible, you can catch me at sametime (gwshan@au1.ibm.com) so that I can take a close look on this, thanks! It's strange that all PCI devices are missed in skiboot. I didn't figure out how BMC GPIO can affect PCI domain in this large scale.

amboar, 2016-12-15:
@gwshan as @shenki mentioned the PCI reset mechanism was reworked in OpenBMC recently, which  could affect the PCI domain at this large a scale. I expect the image @shenki asked @swanman to test was built from a commit prior to the rework, but given that build apparently doesn't work either maybe we've broken something else. The timeout does seem suspiciously like a not-coming-out-of-reset problem to me though.

gwshan, 2016-12-15:
@ambor, yeah, there is one GPIO PIN (at least) used to issue PERST to one PCI slot, not all PCI slots in the system. Let me talk to @swanman when both of us are on line. To me, it looks like a skiboot bug.

gwshan, 2016-12-16:
@shenki reproduced this issue on "palm5" in ozlabs. The all PCIe links are down initially. Then I found the PCIE_RESET_PIN has been put into low-level, which represents resetting the PCIe slot. I forced it to high-level-of-voltage and the PCIe links (the one behind 0001:02:03.0) are up. I guess it's still related to the PCIe reset patch ("https://gerrit.openbmc-project.xyz/#/c/891/").

All PCIe links are down
==================
/ # lspci
0000:00:00.0 PCI bridge: IBM Device 03dc
0001:00:00.0 PCI bridge: IBM Device 03dc
0001:01:00.0 PCI bridge: PLX Technology, Inc. PEX 8718 16-Lane, 5-Port PCI Express Gen 3 (8.0 GT/s) Switch (rev aa)
0001:02:01.0 PCI bridge: PLX Technology, Inc. PEX 8718 16-Lane, 5-Port PCI Express Gen 3 (8.0 GT/s) Switch (rev aa)
0001:02:02.0 PCI bridge: PLX Technology, Inc. PEX 8718 16-Lane, 5-Port PCI Express Gen 3 (8.0 GT/s) Switch (rev aa)
0001:02:03.0 PCI bridge: PLX Technology, Inc. PEX 8718 16-Lane, 5-Port PCI Express Gen 3 (8.0 GT/s) Switch (rev aa)
0001:02:04.0 PCI bridge: PLX Technology, Inc. PEX 8718 16-Lane, 5-Port PCI Express Gen 3 (8.0 GT/s) Switch (rev aa)
0002:00:00.0 PCI bridge: IBM Device 03dc

/ # lspci -vvs 0001:02:03.0
0001:02:03.0 PCI bridge: PLX Technology, Inc. PEX 8718 16-Lane, 5-Port PCI Express Gen 3 (8.0 GT/s) Switch (rev aa) (prog-if 00 [Normal decode])
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 498
	Bus: primary=02, secondary=0d, subordinate=11, sec-latency=0
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [48] MSI: Enable- Count=1/8 Maskable+ 64bit+
		Address: 0000000000000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [68] Express (v2) Downstream Port (Slot+), MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	Report errors: Correctable- Non-Fatal+ Fatal+ Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 256 bytes, MaxReadReq 128 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #3, Speed 8GT/s, Width x2, ASPM L1, Exit Latency L0s <4us, L1 <4us
			ClockPM- Surprise+ LLActRep+ BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x0, TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
		SltCap:	AttnBtn- PwrCtrl- MRL- AttnInd- PwrInd- HotPlug- Surprise-
			Slot #3, PowerLimit 25.000W; Interlock- NoCompl-
		SltCtl:	Enable: AttnBtn- PwrFlt- MRL- PresDet- CmdCplt- HPIrq- LinkChg-
			Control: AttnInd Unknown, PwrInd Unknown, Power- Interlock-
		SltSta:	Status: AttnBtn- PowerFlt- MRL- CmdCplt- PresDet+ Interlock-
			Changed: MRL- PresDet+ LinkState-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis-, LTR+, OBFF Via message ARIFwd+
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled ARIFwd-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-, Selectable De-emphasis: -6dB
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [a4] Subsystem: PLX Technology, Inc. PEX 8718 16-Lane, 5-Port PCI Express Gen 3 (8.0 GT/s) Switch
	Capabilities: [100 v1] Device Serial Number aa-87-00-10-b5-df-0e-00
	Capabilities: [fb4 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP- SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 1f, GenCap+ CGenEn+ ChkCap+ ChkEn-
	Capabilities: [138 v1] Power Budgeting <?>
	Capabilities: [10c v1] #19
	Capabilities: [148 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
[... 62 lines cut ...]
	Capabilities: [a4] Subsystem: PLX Technology, Inc. PEX 8718 16-Lane, 5-Port PCI Express Gen 3 (8.0 GT/s) Switch
	Capabilities: [100 v1] Device Serial Number aa-87-00-10-b5-df-0e-00
	Capabilities: [fb4 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP- SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 1f, GenCap+ CGenEn+ ChkCap+ ChkEn-
	Capabilities: [138 v1] Power Budgeting <?>
[... cut ...]
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed+ WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [e00 v1] #12
	Capabilities: [f24 v1] Access Control Services
		ACSCap:	SrcValid+ TransBlk+ ReqRedir+ CmpltRedir+ UpstreamFwd+ EgressCtrl+ DirectTrans+
		ACSCtl:	SrcValid- TransBlk- ReqRedir- CmpltRedir- UpstreamFwd- EgressCtrl- DirectTrans-
	Capabilities: [b70 v1] Vendor Specific Information: ID=0001 Rev=0 Len=010 <?>

gwshan, 2016-12-16:
Below sequence worked for me. So it's related to PCIE_RESET PIN definitely:

bmc# obmcutil poweron
<wait for 3 seconds>
bmc# devmem 0x1e780000
<get the value in register, called 'reg_val'>
bmc# devmem 0x1e780000 (reg_val | (1 << 13))

The network card is seen in petiboot finally.

adamliyi, 2016-12-19:
Committed a fix here: https://gerrit.openbmc-project.xyz/#/c/1531/, to fix the Reset pin issue. Please have a look.

rahulmah, 2017-01-04:
@williamspatrick : I see that this fix has been integrated in skeleton already. When can we expect to see this fix integrated in openbmc master image.

williamspatrick, 2017-01-04:
Should be fixed now.

closed 2017-01-04 by williamspatrick