openbmc/openbmc #877: Jenkins real hardware test failures do not provide sufficient information to debug the issue

kc8apf, 2016-12-09:
https://gerrit.openbmc-project.xyz/#/c/1445/ triggered a real hardware test run which failed.  The link provided for logs (https://jenkins-openbmc.github.io/openbmc-logs/change-1445/patchset-1/) provides a view of what commands were sent to the REST API but not the console logs or anything showing machine-side errors.  I know that it failed to do the code update but I have no idea why.

anoo1, 2017-10-16:
This is done, Charlie will post an example of data collected and close it.

charleshofer, 2017-10-16:
Here's an example: https://jenkins-openbmc.github.io/openbmc-logs/change-7000/patchset-1/. You can click FFDC logs and see things from the BMC and OS side.

closed 2017-10-16 by anoo1

amboar, 2017-10-16:
This looks quite useful. Are these URLs provided in the comments on patches in Gerrit?

geissonator, 2017-10-16:
Yep, you'll get the link directly in your gerrit commit, but only if the job fails.