openbmc/openbmc #878: Jenkins real hardware tests failing on what appears to be a syntax error

kc8apf, 2016-12-09:
Rerunning tests for https://gerrit.openbmc-project.xyz/#/c/1445/2, the code update succeeded and tests ran.  The results (https://jenkins-openbmc.github.io/openbmc-logs/change-1445/patchset-2/) show many of the tests failing due to:
> Setup failed:
> TypeError: _custom_start_client() got an unexpected keyword argument 'timeout'

geissonator, 2016-12-12:
@gkeishin please take a look?

gkeishin, 2016-12-12:
This seems bit odd to failed like this on an SSH connection to begin with.. The tests should also have failed in that case..  looking into it..

gkeishin, 2016-12-12:
Not sure what has escalated the issue with SSHLibrary however, I noticed that 
python -m robot      execute works good but tox execution of the same seems appending extra "timeout" in the session establishment which seems to make it fail here

gkeishin, 2016-12-12:
I think i know the problem.. the tox doesn't seems to resolve the python package dependencies..

```
rango@ubuntu:~/openbmc-test-automation$ sudo pip show robotframework-sshlibrary
Name: robotframework-sshlibrary
Version: 2.1.2
Summary: Robot Framework test library for SSH and SFTP
Home-page: https://github.com/robotframework/SSHLibrary
Author: Robot Framework Developers
Author-email: robotframework@gmail.com
License: Apache License 2.0
Location: /usr/local/lib/python2.7/dist-packages
Requires: robotframework, paramiko
```
SSH library from robot framework does need explicitly paramiko 

So the deps arg in the tox needs to be updated and seems to work with paramiko version 2.0.2
paramiko==2.0.2

This seems to have fixed the issue for now..

closed 2016-12-12 by williamspatrick