openbmc/openbmc #880: [Witherspoon - v1.99.0-321-gb4f5019 ] PNOR update Failed

gkeishin, 2016-12-13:
`root@witherspoon:~# cat /etc/version`
`20161212180427`

```
root@witherspoon:~# journalctl --no-pager | grep flash_bios.exe
Dec 12 18:06:00 witherspoon flash_bios.exe[774]: Starting flasher: bios,,/org/openbmc/control/flash/bios,
Dec 12 18:06:00 witherspoon flash_bios.exe[774]: flasher: , BMC = 1, address = 0x0
Dec 12 18:06:00 witherspoon flash_bios.exe[774]: Setting up flash
Dec 12 18:06:00 witherspoon flash_bios.exe[774]: Starting flasher: bios,,/org/openbmc/control/flash/bios,
Dec 12 18:06:01 witherspoon flash_bios.exe[774]: flasher: , BMC = 1, address = 0x0
Dec 12 18:06:01 witherspoon flash_bios.exe[774]: Setting up flash
Dec 12 18:10:19 witherspoon flash_bios.exe[774]: Starting flasher: bios,/tmp/witherspoon.pnor,/org/openbmc/control/flash/bios,
Dec 12 18:10:19 witherspoon flash_bios.exe[774]: flasher: /tmp/witherspoon.pnor, BMC = 1, address = 0x0
Dec 12 18:10:19 witherspoon flash_bios.exe[774]: Setting up flash
Dec 12 18:10:19 witherspoon flash_bios.exe[774]: Erasing... (may take a while !) Error -1 erasing chip
```

- [x] Journal log here : 
[witherspoon_pnor_update.txt](https://github.com/openbmc/openbmc/files/649414/witherspoon_pnor_update.txt)

spinler, 2016-12-13:
Same signature as https://github.com/openbmc/openbmc/issues/848

shenki, 2016-12-15:
@gkeishin please put the output of `/etc/os-version` in your report instead.

Closing as a duplicate of #848

closed 2016-12-15 by shenki

gkeishin, 2016-12-15:
@shenki  In general,  isn't the v1.99.0-321-gb4f5019  mentioned in the title the only needed for the version information if we are looking at /etc/os-version ?

shenki, 2016-12-15:
Yes. You're right, I didn't notice the title. Thanks for pointing that out :)