openbmc/openbmc #881: [Barreleye v1.99.0-322-g1da3a35 ] DIMM23 properties not loaded

gkeishin, 2016-12-13:
`root@barreleye:~# uptime; obmcutil state ; cat /etc/version`
` 06:38:39 up 30 min,  load average: 0.39, 0.76, 0.94`
` = HOST_BOOTED`
`20161212213842`
`root@barreleye:~#`

Nov 21 06:08:59 barreleye inventory_items.py[694]: Loading from cache: /var/cache/obmc/org.openbmc.inventory.system.chassis.motherboard.dimm22@org.openbmc.InventoryItem.props
Nov 21 06:08:59 barreleye inventory_items.py[694]: Loading from cache: /var/cache
```/obmc/org.openbmc.inventory.system.chassis.motherboard.dimm23@org.openbmc.InventoryItem.props
Nov 21 06:08:59 barreleye inventory_items.py[694]: ERROR: Loading cache file:```
Nov 21 06:08:59 barreleye inventory_items.py[694]: Loading from cache: /var/cache/obmc/org.openbmc.inventory.system.chassis.motherboard.dimm20@org.openbmc.InventoryItem.props

Enumerating inventory, we can see dimm23 properties are not populated

"/org/openbmc/inventory/system/chassis/motherboard/dimm23": {},
"/org/openbmc/inventory/system/chassis/motherboard/dimm22": {
     "fault": "False", 
     "fru_type": "DIMM", 
     "is_fru": 1, 
     "present": "False", 
     "version": ""
   },

- [x] Journal Log - 
[20161213094743401408_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/649684/20161213094743401408_BMC_journalctl.txt)

adamliyi, 2016-12-15:
I will take look.

gkeishin, 2017-03-31:
The inventory are now on xyz path.. do we still need to track this one now ? @adamliyi

gkeishin, 2017-03-31:
Closing this one out.. Let me know if needed to re-open.

closed 2017-03-31 by gkeishin