openbmc/openbmc #882: Build failure: [barreleye 1da3a353] ERROR: Function failed: do_uboot_mkcimage with ob8.1650.20161213a

sdturne, 2016-12-13:
Barreleye build ob8.1650.20161213a at git commit 1da3a353e58e6439f5b7c5a0634e1b382bb27bfc
failed with the following output:

NOTE: Running task 1756 of 2900 (ID: 70, /esw/fspbld-local/obmcbar/Builds/extract/1da3a353e58e6439f5b7c5a0634e1b382bb27bfc/meta-phosphor/common/recipes-kernel/linux/linux-obmc_4.7.bb, do_compile_kernelmodules)
Log data follows:
| DEBUG: Executing shell function do_uboot_mkcimage
| none
| cat: linux.bin: No such file or directory
| WARNING: /esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/barreleye-openbmc-linux-gnueabi/linux-obmc/4.7+gitAUTOINC+6fcd70b7e3-r0/temp/run.do_uboot_mkcimage.19434:1 exit 1 from 'cat linux.bin $dt > linux-dts.bin'
| ERROR: Function failed: do_uboot_mkcimage (log file is located at /esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/barreleye-openbmc-linux-gnueabi/linux-obmc/4.7+gitAUTOINC+6fcd70b7e3-r0/temp/log.do_uboot_mkcimage.19434)
NOTE: recipe linux-obmc-4.7+gitAUTOINC+6fcd70b7e3-r0: task do_uboot_mkcimage: Failed
NOTE: recipe linux-obmc-4.7+gitAUTOINC+6fcd70b7e3-r0: task do_uboot_mkimage: Succeeded
NOTE: recipe linux-obmc-4.7+gitAUTOINC+6fcd70b7e3-r0: task do_assemble_fitimage: Succeeded
NOTE: recipe linux-obmc-4.7+gitAUTOINC+6fcd70b7e3-r0: task do_compile_kernelmodules: Started
NOTE: recipe linux-obmc-4.7+gitAUTOINC+6fcd70b7e3-r0: task do_compile_kernelmodules: Succeeded
NOTE: recipe gcc-runtime-5.3.0-r0: task do_configure: Succeeded
NOTE: recipe glibc-locale-2.23-r0: task do_package: Succeeded
NOTE: Tasks Summary: Attempted 1756 tasks of which 24 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /esw/fspbld-local/obmcbar/Builds/extract/1da3a353e58e6439f5b7c5a0634e1b382bb27bfc/meta-phosphor/common/recipes-kernel/linux/linux-obmc_4.7.bb, do_uboot_mkcimage
Summary: There were 14 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
Problem running the following command:
openbmc-build -customrc /afs/austin/projects/esw/obmc910/obmc910_configs/customrc-openbmc -extract-dir /esw/fspbld-local/obmcbar/Builds/extract/1da3a353e58e6439f5b7c5a0634e1b382bb27bfc -level 920.1652.20161213b -machine barreleye -output-dir /esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output -python-paths /afs/austin/projects/esw/python_versions/RH6/python_2.7.10/python/git/bin:/afs/austin/projects/esw/python_versions/RH6/python_2.7.10/python/python/bin -templateconf meta-openbmc-machines/meta-openpower/meta-rackspace/meta-barreleye/conf
ERROR: linux-obmc-4.7+gitAUTOINC+6fcd70b7e3-r0 do_uboot_mkcimage: Function failed: do_uboot_mkcimage (log file is located at /esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/barreleye-openbmc-linux-gnueabi/linux-obmc/4.7+gitAUTOINC+6fcd70b7e3-r0/temp/log.do_uboot_mkcimage.19434)
ERROR: Logfile of failure stored in: /esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/barreleye-openbmc-linux-gnueabi/linux-obmc/4.7+gitAUTOINC+6fcd70b7e3-r0/temp/log.do_uboot_mkcimage.19434
ERROR: Task 60 (/esw/fspbld-local/obmcbar/Builds/extract/1da3a353e58e6439f5b7c5a0634e1b382bb27bfc/meta-phosphor/common/recipes-kernel/linux/linux-obmc_4.7.bb, do_uboot_mkcimage) failed with exit code '1'
Build failed

amboar, 2016-12-13:
@sdturne when pasting logs into github can you please use the triple-backtick markdown to format the text. The output is easier to read and it stops github from interpreting log character sequences as markdown.

Regardless, the output mentions that

1. `linux.bin` wasn't found, and
2. There's a log file capturing the details of the failure

```
Log data follows:
| DEBUG: Executing shell function do_uboot_mkcimage
| none
| cat: linux.bin: No such file or directory
| WARNING: /esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/barreleye-openbmc-linux-gnueabi/linux-obmc/4.7+gitAUTOINC+6fcd70b7e3-r0/temp/run.do_uboot_mkcimage.19434:1 exit 1 from 'cat linux.bin $dt > linux-dts.bin'
| ERROR: Function failed: do_uboot_mkcimage (log file is located at /esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/barreleye-openbmc-linux-gnueabi/linux-obmc/4.7+gitAUTOINC+6fcd70b7e3-r0/temp/log.do_uboot_mkcimage.19434)
```

If you still have `log.do_uboot_mkcimage.19434` can you please attach it?

sdturne, 2016-12-13:
Its content is identical to the "Log data follows:" output that I pasted and you re-quoted in item 2.

amboar, 2016-12-13:
Okay, that's unhelpful. Your output also mentions:

```
Summary: There were 14 WARNING messages shown.
```

Can you provide the warning output?

sdturne, 2016-12-14:
```
WARNING: pigz-native-2.3.3-r0 do_fetch: Failed to fetch URL http://zlib.net/pigz/pigz-2.3.3.tar.gz, attempting MIRRORS if available
WARNING: expat-2.1.0-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/expat/expat-2.1.0.tar.gz, attempting MIRRORS if available
WARNING: flex-native-2.6.0-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/flex/flex-2.6.0.tar.bz2, attempting MIRRORS if available
WARNING: python-cython-native-0.24-r0 do_fetch: Failed to fetch URL http://www.cython.org/release/Cython-0.24.tar.gz, attempting MIRRORS if available
WARNING: libcheck-0.10.0-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/check/check-0.10.0.tar.gz, attempting MIRRORS if available
WARNING: libevent-2.0.22-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/levent/libevent-2.0.22-stable.tar.gz, attempting MIRRORS if available
WARNING: libcgroup-0.41-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/project/libcg/libcgroup/v0.41/libcgroup-0.41.tar.bz2, attempting MIRRORS if available
WARNING: systemd-systemctl-native-1.0-r6 do_populate_lic: Could not copy license file /esw/fspbld-local/obmcbar/Builds/extract/1da3a353e58e6439f5b7c5a0634e1b382bb27bfc/LICENSE to /esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/x86_64-linux/systemd-systemctl-native/1.0-r6/license-destdir/systemd-systemctl-native/LICENSE: [Errno 2] No such file or directory: '/esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/x86_64-linux/systemd-systemctl-native/1.0-r6/license-destdir/systemd-systemctl-native/LICENSE'
WARNING: strace-4.11-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/strace/strace-4.11.tar.xz, attempting MIRRORS if available
WARNING: sysfsutils-2.1.0-r5 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/linux-diag/sysfsutils-2.1.0.tar.gz, attempting MIRRORS if available
WARNING: run-postinsts-1.0-r9 do_populate_lic: Could not copy license file /esw/fspbld-local/obmcbar/Builds/extract/1da3a353e58e6439f5b7c5a0634e1b382bb27bfc/LICENSE to /esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/all-openbmc-linux/run-postinsts/1.0-r9/license-destdir/run-postinsts/LICENSE: [Errno 2] No such file or directory: '/esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/all-openbmc-linux/run-postinsts/1.0-r9/license-destdir/run-postinsts/LICENSE'
WARNING: squashfs-tools-native-4.3+gitrAUTOINC+9c1db6d13a-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/sevenzip/lzma465.tar.bz2;name=lzma, attempting MIRRORS if available
WARNING: systemd-compat-units-1.0-r29 do_populate_lic: Could not copy license file /esw/fspbld-local/obmcbar/Builds/extract/1da3a353e58e6439f5b7c5a0634e1b382bb27bfc/LICENSE to /esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/all-openbmc-linux/systemd-compat-units/1.0-r29/license-destdir/systemd-compat-units/LICENSE: [Errno 2] No such file or directory: '/esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/all-openbmc-linux/systemd-compat-units/1.0-r29/license-destdir/systemd-compat-units/LICENSE'
WARNING: core-image-minimal-initramfs-1.0-r0 do_populate_lic: Could not copy license file /esw/fspbld-local/obmcbar/Builds/extract/1da3a353e58e6439f5b7c5a0634e1b382bb27bfc/LICENSE to /esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/barreleye-openbmc-linux-gnueabi/core-image-minimal-initramfs/1.0-r0/license-destdir/core-image-minimal-initramfs/LICENSE: [Errno 2] No such file or directory: '/esw/fspbld-local/obmcbar/Builds/ob8.1650.20161213a/output/build/tmp/work/barreleye-openbmc-linux-gnueabi/core-image-minimal-initramfs/1.0-r0/license-destdir/core-image-minimal-initramfs/LICENSE'
```

amboar, 2016-12-14:
I assume the MIRRORS alternatives for the failed fetches succeeded?

sdturne, 2016-12-15:
Yes they did.  I can only assume that this was some mysterious internal or environmental problem on the build machine whose symptom was not well reported by the build environment, because a subsequent rebuild of the same level of code succeeded today on a different build machine.  I will close this issue.  Thanks for the investigation.

closed 2016-12-15 by sdturne