openbmc/openbmc #883: Remove dependencies on org.openbmc.control.Host?

GHF, 2016-12-14:
I was testing out https://gerrit.openbmc-project.xyz/#/c/1217/11 (to fix #647) and found that replacing org.openbmc.control.Host with a one-shot script breaks quite a lot of functionality. 

For example, org.openbmc.settings.Host will never start because it waits for Host: b78c6def968a515c5b9f2d44f44c24b19134b6d5

Moreover, org.openbmc.settings.Host (even when run manually) never starts up correctly, probably because of whatever instance_query stuff is encoded here: https://github.com/openbmc/phosphor-settingsd/blob/master/settings.yaml

```
root@zaius:~# busctl tree --no-pager org.openbmc.settings.Host
Only root object discovered.
```

these broken dependencies are bad enough that obmc-chassis-start@0.target can't be reached

Is it feasible to remove the current implementation of + dependencies of org.openbmc.control.Host as Brad suggested (https://gist.github.com/bradbishop/a5150ef32692a099a5dfd1c7c74cd915)? Or do we have to replace it with a different implementation of org.openbmc.control.Host?

I don't know how these applications work well enough to understand the implications.

spinler, 2016-12-14:
We (obviously) hit this same problem on Witherspoon, and have to hack some things up to reach standby.

williamspatrick, 2016-12-14:
I'm not pleased with how that 'object querying' stuff ended up in settingsd.  We had a mess of trying to get it working after it went in and dealing with these systemd dependencies.  It isn't right as it is.

In the long-term, we need that settingsd to be listening for object-created signals if it wants to dynamically create settings objects based on inventory objects.

In the short-term, I think we should hack at it to make a special case for 'host' to fall back to always having 'host0' in the case of an empty query.  Then we can get rid of the dependency on the "real" host0 object existing.

In the medium-term, @geissonator will be having the new xyz.openbmc_project.State.Host objects that we can move settingsd over to.

williamspatrick, 2016-12-14:
I think we need settingsd object to exist for `obmc-chassis-start@0` so that we know if we are in "debug-mode" or not.

GHF, 2016-12-14:
Thanks Patrick. So we need a null/stub daemon that takes the host0 dbus name?

Is debug mode useful without control.Host? It looks to me like it's only used for setting FSI mode between BMC and FSP2 debug.

I'm blocked on this framework working before mailing Zaius VCS control (to be sequenced between workaround and IPL start). Locally, I tried building both obmc-op-control-host and p9-host-start into the image, which is a horrible hack that only works for me because control_host.exe happens to not work for us (thus not conflicting with pdbg).

spinler, 2016-12-15:
I implemented the short term solution Patrick suggested at:
https://gerrit.openbmc-project.xyz/#/c/1519/
https://gerrit.openbmc-project.xyz/#/c/1520/

With those changes, I can reach the obmc-standby.target.

I also removed the host0 dependency from the old style state management:
https://gerrit.openbmc-project.xyz/#/c/1518/

That's in Witherspoon.py, so each system would need that done separately.

gtmills, 2018-10-12:
@GHF Can this be closed?

closed 2018-10-14 by GHF