openbmc/openbmc #886: MRW Common VPD

geissonator, 2016-12-14:


ratagupt, 2016-12-16:
Need data in the format like ..
Fru
     Associated Device
                 Required Attributes(for each device)
Need to make a list of the required attributes for Associated Non Fru device which is required from MRW.
May Need to make the changes in the Inventory DBUS interfaces also depends on the attribute.

ratagupt, 2016-12-23:
- [ ] write a perl script to fetch the data from MRW.

- [ ] Write a function which generate YAML file.

- [ ] Write a bitbake file.

ratagupt, 2016-12-23:
Currently we are interested in following  IPMI FRU types
Proc,Core,Dimm,Membuff,System,NODE(Chassis),MotherBoard.
@williamspatrick :- Do we need to extract something else which I missed?

closed 2017-02-01 by geissonator