openbmc/openbmc #887: P9 DD1 requires AVS bus workaround

spinler, 2016-12-14:
P9 DD1.x requires a workaround for the AVS bus, p9_dd1_avsbus_workaround.  

Here it is in the poweron sequence:
```
 ucd_disable_vcs
 p9_dd1_avsbus_workaround <---this
  avsbus_disable   # <-- for witherspoon only, so 'vrm' can run
  echo Applying voltage overrides... 
  vrm vdna=0.9 vddra=1.35 vdnb=0.9 vddrb=1.35  #for witherspoon only
  avsbus_enable  <-- for witherspoon only, need for 'vrm'
  vcs_off
  echo -n  powering on....
  gpio 345 out 1                # on
```

p9_dd1_avsbus_workaround:
```
#!/bin/sh
# #############################################################
# Needs run once each A/C power cycle. prior to powering on.
# avsbus crc errors
# Work-around to fix AVSBus errors due to P9 DD1 timing glitch" commands.
# Should be fixed in DD2 but commands should still work either way.
# important note, my system did not show VDN on bus until power was cycled up/down.
# Rework application error (this should be a known issue).

echo "Set VDDIO to 1.5V (P9 DD1 errata workaround)"
i2cset -y 4 0x12 0x2E 0x23 b # VDD/VCS 0
i2cset -y 4 0x13 0x2E 0x23 b # VDN 0
i2cset -y 5 0x12 0x2E 0x23 b # VDD/VCS 1
i2cset -y 5 0x13 0x2E 0x23 b # VDN 1
```

Not sure if these commands are common to all systems or not.

GHF, 2016-12-15:
the `#for witherspoon only` part is indeed for Witherspoon only, as we've found Zaius can boot with our VRMs' default voltages. However, Zaius's VRs have AVSBus disabled by default so we need to perform `avsbus_enable` in the sequence so that the host can take control of the VRs.

We also need a variant of the voltage tuning: the host apparently sets voltages that are too low to boot again (not to mention fails the Zaius power sequencer thresholds). So we need to disable AVSBus control (and thus reverting to VRM programmed default setpoints) either

1. prior to each power-on OR
2. after each power-off

so that the _subsequent_ boot will work.

williamspatrick, 2016-12-15:
openbmc/openbmc#891 gives the specifics for Witherspoon.

@spinler - I suspect that we should put the AVS workaround into a separate script and maybe a separate service file.  Can we put a dependency relationship between the VCS workaround service and the AVS workaround service?  Then we can install a system-specific AVS workaround script but still get it called in the right place for all machines.

GHF, 2016-12-16:
I mailed the Zaius implementation at https://gerrit.openbmc-project.xyz/#/c/1528/

I went with "AVSBus disable at power off." Per #891, this relies on VOUT_COMMAND="whatever the VRM HW set in manufacturing" being a valid boot voltage, which it is for Zaius. If you folks implement reading a VOUT_COMMAND from XML, then we can switch to that as a shared common implementation.

spinler, 2016-12-20:
https://gerrit.openbmc-project.xyz/#/c/1568/

mine260309, 2016-12-27:
Romulus change: https://gerrit.openbmc-project.xyz/#/c/1613

closed 2017-01-05 by williamspatrick