openbmc/openbmc #889: Boot host from emulated PNOR (memboot)

GHF, 2016-12-14:
Patrick and Benjamin Herrenschmidt both mentioned memboot: LPC bridge mapped to a region in memory other than the PNOR's SPI flash controller. From what I can see, we need:

- a kernel driver to configure the LPC bridge
- an application that copies/extracts the PNOR to memory and request that the LPC bridge map to that window

Host/MRW folks also want to customize the PNOR and thus need code that understand its layout scheme (TOC) and mediate read/write access, so I'll let them flesh our how that should work. See also https://github.com/openbmc/openbmc/issues/553

amboar, 2016-12-15:
Here are some bits that I'm aware of on this topic: @cyrilbur-ibm has sent a couple of patches for LPC control and mailbox communications:

* LPC: http://patchwork.ozlabs.org/patch/704320/
* MBOX: http://patchwork.ozlabs.org/patch/704318/

There's also the preliminary memboot hack in skiboot that's been around for a while:

https://github.com/open-power/skiboot/tree/master/external/memboot

Clearly the memboot tool needs work, but at least something exists that can be experimented with if desired.

rlippert, 2016-12-16:
Looks like there is also an unsubmitted pull request for P8 hostboot here:

https://github.com/open-power/hostboot/pull/19

This would need to be rebased for the P9 hostboot branch.

gtmills, 2018-05-23:
@rlippert @GHF Can this be closed?

rlippert, 2018-05-23:
I've never personally seen this actually work but will just assume with the new ubifs/virtual PNOR stuff it is working now :)

gtmills, 2018-05-23:
Going to close. Year and half old. Please reopen if still an issue.

amboar, 2018-05-23:
It's still an issue

amboar, 2019-03-28:
This is working; we even have a plain file backend now so touching the flash isn't required for testing host firmware: https://github.com/openbmc/mboxbridge/commit/5320f6e02453261e6c75d3bef101ad1bef6de82e

closed 2019-03-28 by amboar