openbmc/openbmc #890: Enable LPC to AHB bridge disabled on witherspoon

mikey, 2016-12-15:
In the openbmc kernel we set the enabled the LPC to AHB bridge here:
https://github.com/openbmc/linux/blob/731a90c2a97d8909c27cf0ce6b52b03c0068274b/arch/arm/mach-aspeed/aspeed.c#L43
This has bit 8 (ENL2H: Enable LPC to AHB bridge)  set.

But on a real system, I see this.:
```
root@w2:~# devmem 0x1E789080 32
0x00740400
root@w2:~# uname -a
Linux w2 4.7.10-731a90c2a97d8909c27cf0ce6b52b03c0068274b #1 Thu Nov 10 01:05:16 UTC 2016 armv6l GNU/Linux
root@w2:~#  
```
Which has bit 8 (ENL2H: Enable LPC to AHB bridge) disabled. 

So is something in userspace playing with this register after the kernel configures it?

mikey, 2016-12-16:
FWIW, I've confirmed this on a second machine after a clean BMC reboot.

shenki, 2016-12-19:
@geissonator Is there someone who can take a look at our userspace to work out what is going on here?

I confirmed that booting our kernel without the openbmc userspace gives the expected value.

geissonator, 2017-01-18:
We're thinking maybe something in the meta-phosphor/common/recipes-phosphor/flash/obmc-flash-bios/obmc-flash-init.service potentially but weird it's seen on witherspoon but not zaius.  @spinler is going to try a few things on w1.

spinler, 2017-01-18:
Uma tried without obmc-flash-init.service  but that didn't help

spinler, 2017-01-18:
I think the problem is because of this check:
```
if (of_machine_is_compatible("ibm,witherspoon-bmc"))
		do_witherspoon_setup();
```
but the device tree has:
```
	compatible = "ibm,witherspoon", "aspeed,ast2500";
```
And Zaius has the expected ```	compatible = "ingrasys,zaius-bmc", "aspeed,ast2500";```

williamspatrick, 2017-01-18:
```
cc9cc99d (Joel Stanley     2016-07-28 16:31:22 +0930   8) 	compatible = "ibm,witherspoon", "aspeed,ast2500";
```

```
8139ec738 (Joel Stanley                   2016-11-22 14:38:48 +1030 246) 	if (of_machine_is_compatible("ibm,witherspoon-bmc"))
```

```
@@ -200,6 +222,8 @@ static void __init aspeed_init_early(void)
                do_ast2500evb_setup();
        if (of_machine_is_compatible("ingrasys,zaius-bmc"))
                do_zaius_setup();
+       if (of_machine_is_compatible("ibm,witherspoon-bmc"))
+               do_witherspoon_setup();
 }
```

@shenki - This has never worked.

williamspatrick, 2017-01-20:
Resolved now.

closed 2017-01-20 by williamspatrick