openbmc/openbmc #892: Customize services that use pdbg with machine name

GHF, 2016-12-16:
pdbg requires a "-d <device>" argument to select GPIO mappings that work for the board the BMC is on. "<device>" can be "long form": full machine names such as "witherspoon".

The services that use pdbg (SBE start start_host@.service and VCS workaround vcs_workaround@.service) need to be customized at build time with a substitution for machine name into the script invocations (see 927394011a2d63a23ce65dbefd35476a260bb8a3).

Alternatively, they can be customized with a per-machine environment variables file that is loaded at runtime.

shenki, 2016-12-16:
Another alternative is we detect the system from the device tree. This is a change to pdbg that would not depend on adding other parts of the system.

williamspatrick, 2017-01-05:
@GHF - Is this done now?

GHF, 2017-01-06:
Yes, fixed in aac4adf831eebc74d97076e94125ebf5b6d5da48.

closed 2017-01-06 by GHF