openbmc/openbmc #893: ObjectMapper stops responding after error

GHF, 2016-12-16:
At some point the mapper got into a bad state, causing all calls routed through it to fail:

```
Nov 28 13:52:34 zaius systemd[1]: Started Start Power0.
Nov 28 13:52:34 zaius systemd[1]: Reached target Power0 On.
Nov 28 13:52:34 zaius systemd[1]: Stopped target Chassis0 (Stop).
Nov 28 13:52:39 zaius mapper[4675]: Error finding '/org/openbmc/control/power0' service: Device or resource busy
Nov 28 13:52:39 zaius systemd[1]: op-power-start@0.service: Main process exited, code=exited, status=1/FAILURE
Nov 28 13:53:09 zaius systemd[1]: Stopped target Power0 On.
Nov 28 13:53:09 zaius systemd[1]: Stopped Start Power0.
Nov 28 13:53:09 zaius systemd[1]: op-power-start@0.service: Unit entered failed state.
Nov 28 13:53:09 zaius systemd[1]: op-power-start@0.service: Failed with result 'exit-code'.
```

I tried a `/usr/sbin/mapper call /org/openbmc/control/power0 org.openbmc.control.Power setPowerState i 0` with the same results.

The power0 interface was definitely there:

```
root@zaius:~# busctl introspect --no-pager org.openbmc.control.Power /org/openbmc/control/power0
NAME                                TYPE      SIGNATURE RESULT/VALUE FLAGS
org.freedesktop.DBus.Introspectable interface -         -            -
.Introspect                         method    -         s            -
org.freedesktop.DBus.Peer           interface -         -            -
.GetMachineId                       method    -         s            -
.Ping                               method    -         -            -
org.freedesktop.DBus.Properties     interface -         -            -
.Get                                method    ss        v            -
.GetAll                             method    s         a{sv}        -
.Set                                method    ssv       -            -
.PropertiesChanged                  signal    sa{sv}as  -            -
org.openbmc.Control                 interface -         -            -
.init                               method    -         -            -
.heatbeat                           property  i         0            emits-change
.poll_interval                      property  i         3000         emits-change writable
.GotoSystemState                    signal    s         -            -
.Heartbeat                          signal    s         -            -
.Started                            signal    -         -            -
org.openbmc.control.Power           interface -         -            -
.getPowerState                      method    -         i            -
.setPowerState                      method    i         -            -
.pgood                              property  i         1            emits-change
.pgood_timeout                      property  i         10           emits-change writable
.state                              property  i         1            emits-change
.PowerGood                          signal    -         -            -
.PowerLost                          signal    -         -            -
```

And everything worked again after I restarted the mapper daemon (`systemctl restart xyz.openbmc_project.ObjectMapper`). Here are the logs for it:

```
root@zaius:~# journalctl -u xyz.openbmc_project.ObjectMapper --no-pager
-- Logs begin at Sun 2016-11-27 19:46:30 UTC, end at Mon 2016-11-28 14:09:45 UTC. --
Nov 27 19:46:42 zaius systemd[1]: Starting Phosphor DBus Service Discovery Manager...
Nov 27 19:47:07 zaius phosphor-mapper[768]: ObjectMapper startup complete.  Discovery in progress...
Nov 27 19:47:15 zaius phosphor-mapper[768]: ObjectMapper discovery complete
Nov 27 19:47:15 zaius systemd[1]: Started Phosphor DBus Service Discovery Manager.
Nov 28 13:21:28 zaius phosphor-mapper[768]: Traceback (most recent call last):
Nov 28 13:21:28 zaius phosphor-mapper[768]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
Nov 28 13:21:28 zaius phosphor-mapper[768]:     *message.get_args_list()))
Nov 28 13:21:28 zaius phosphor-mapper[768]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/server.py", line 170, in <lambda>
Nov 28 13:21:28 zaius phosphor-mapper[768]:     error_handler=lambda x: error_callback(service, path, x))
Nov 28 13:21:28 zaius phosphor-mapper[768]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/server.py", line 292, in discovery_error
Nov 28 13:21:28 zaius phosphor-mapper[768]:     raise e
Nov 28 13:21:28 zaius phosphor-mapper[768]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Nov 28 14:08:13 zaius systemd[1]: Stopping Phosphor DBus Service Discovery Manager...
Nov 28 14:08:13 zaius systemd[1]: Stopped Phosphor DBus Service Discovery Manager.
Nov 28 14:08:13 zaius systemd[1]: Starting Phosphor DBus Service Discovery Manager...
Nov 28 14:08:16 zaius phosphor-mapper[4710]: ObjectMapper startup complete.  Discovery in progress...
Nov 28 14:08:22 zaius phosphor-mapper[4710]: ObjectMapper discovery complete
Nov 28 14:08:22 zaius systemd[1]: Started Phosphor DBus Service Discovery Manager.
```

rlippert, 2016-12-16:
I also noticed that after the mapper fails this way, IPMI ESEL messages from hostboot start reporting this error about "Failed to reserve_esel":

 26.89092|IPMI|dd: I>read b2h ok 2c:42 seq a8 len 0 cc ce
 26.89104|IPMI|sel: E>Failed to reserve_sel, o_cc ce
 26.89117|IPMI|sel: <<send_esel o_err=00000000, o_cc=xce, sel recID=x00, esel recID=x00
 26.89118|IPMI|sel: <<process_esel
 26.89119|IPMI|sel: >>process_esel
 26.89119|IPMI|sel: >>sel list size 1
 26.89120|IPMI|sel: >>send_esel
 26.89121|IPMI|rp: queuing sync 28:42

adamliyi, 2016-12-19:
@bradbishop , will you look at this?

gtmills, 2018-05-31:
@GHF @rlippert No update in over a year. Can this be closed or still a problem?

GHF, 2018-06-01:
It's for org.openbmc, so I'm guessing that it no longer is a problem. I haven't look at BMC in a while.

rlippert, 2018-06-01:
Probably fixed (or atleast helped) by https://github.com/openbmc/phosphor-objmgr/commit/140f4102169adc30911413352494961867b2b6a3

gtmills, 2018-06-01:
@rlippert Based on comments above, closing. If still an issue please reopen. thanks!

closed 2018-06-01 by gtmills