openbmc/openbmc #894: [Witherspoon:v1.99.0-331-g9224355] Phosphor LED Management Daemon not started

sivassrr, 2016-12-18:
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-331"
VERSION_ID="v1.99.0-331-g9224355-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-331"
BUILD_ID="v1.99.0"
root@witherspoon:~# cat /etc/version
20161216213323

[  OK  ] Stopped Phosphor LED Management Daemon.
[FAILED] Failed to start Phosphor LED Management Daemon.
See 'systemctl status org.openbmc.control.led.service' for details.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 witherspoon ttyS4

witherspoon login: ftgmac100 1e660000.ethernet eth0: NCSI interface up
random: nonblocking pool is initialized

 org.openbmc.control.led.service - Phosphor LED Management Daemon
   Loaded: loaded (/lib/systemd/system/org.openbmc.control.led.service; enabled;
 vendor preset: enabled)
   Active: inactive (dead) since Sun 2016-12-18 00:37:39 UTC; 8h ago
  Process: 735 ExecStart=/usr/sbin/led_controller.exe (code=exited, status=0/SUC
CESS)
 Main PID: 735 (code=exited, status=0/SUCCESS)

Dec 18 00:37:39 witherspoon systemd[1]: org.openbmc.control.led.service: Service
 hold-off time over, scheduling restart.
Dec 18 00:37:39 witherspoon systemd[1]: Stopped Phosphor LED Management Daemon.
Dec 18 00:37:39 witherspoon systemd[1]: [[0;1;39morg.openbmc.control.led.service
: Start request repeated too quickly.[[0m
Dec 18 00:37:39 witherspoon systemd[1]: [[0;1;31mFailed to start Phosphor LED Ma
nagement Daemon.[[0m
[journalctlwitherspoonled.txt](https://github.com/openbmc/openbmc/files/659352/journalctlwitherspoonled.txt)

sivassrr, 2016-12-18:
Had kicked off long run test that is 10 Iteration of our CT bucket. At the end of the run found in the serial console that "LED Management Daemon not started".

amboar, 2016-12-19:
@sivassrr this sounds related to #705

sivassrr, 2016-12-19:
Level what I had used is https://openpower.xyz/job/openbmc-build/417/distro=ubuntu,target=witherspoon/

adamliyi, 2016-12-19:
@vishwabmc , looks you are also looking at #705?

sivassrr, 2016-12-22:
Able to recreate with 422 driver as well. 

[  OK  ] Reached target System Time Synchronized.
[  OK  ] Started Permit User Sessions.
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Started Clear one time boot overrides.
[  OK  ] Started Phosphor Watchdog0.
[  OK  ] Started Phosphor BMC0 Control.
[  OK  ] Started Phosphor Host0 Checkstop Monitor.
[  OK  ] Started Phosphor Event Manager.
[  OK  ] Started Network Service.
[  OK  ] Reached target Network.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
         Starting Phosphor Inband IPMI...
[  OK  ] Started Serial Getty on ttyS4.
[  OK  ] Reached target Login Prompts.
[  OK  ] Reached target Multi-User System.
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Started Phosphor Inband IPMI.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
ftgmac100 1e660000.ethernet eth0: NCSI interface up
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Stopped Phosphor LED Management Daemon.
[FAILED] Failed to start Phosphor LED Management Daemon.
See 'systemctl status org.openbmc.control.led.service' for details.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 witherspoon ttyS4

witherspoon login: random: nonblocking pool is initialized

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 witherspoon ttyS4

witherspoon login: root
Password: 
root@witherspoon:~# cat /etc/version;cat /etc/os-release
20161222025539
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-351"
VERSION_ID="v1.99.0-351-g8c1f920-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-351"
BUILD_ID="v1.99.0"
root@witherspoon:~#

amboar, 2016-12-22:
@sivassrr Please use the triple  back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

amboar, 2016-12-22:
@sivassrr what is this 422 driver you mention?

sivassrr, 2017-01-12:
Even today's driver https://openpower.xyz/view/OpenBMC/job/openbmc-build/437/distro=ubuntu,target=witherspoon/artifact/images/witherspoon/

```
[  OK  ] Stopped Phosphor LED Management Daemon.
[FAILED] Failed to start Phosphor LED Management Daemon.
See 'systemctl status org.openbmc.control.led.service' for details.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 witherspoon ttyS4

witherspoon login: ftgmac100 1e660000.ethernet eth0: NCSI interface up
random: nonblocking pool is initialized

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 witherspoon ttyS4

witherspoon login: root
Password: 
root@witherspoon:~# cat /etc/version
20170111221154
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-377"
VERSION_ID="v1.99.0-377-ge4c152a-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-377"
BUILD_ID="v1.99.0"
```

Output of systemctl status org.openbmc.control.led.service

```
org.openbmc.control.led.service - Phosphor LED Management Daemon
   Loaded: loaded (/lib/systemd/system/org.openbmc.control.led.service; enabled;
 vendor preset: enabled)
   Active: inactive (dead) since Mon 2016-11-21 06:08:23 UTC; 1min 11s ago
  Process: 739 ExecStart=/usr/sbin/led_controller.exe (code=exited, status=0/SUC
CESS)
 Main PID: 739 (code=exited, status=0/SUCCESS)

Nov 21 06:08:23 witherspoon systemd[1]: org.openbmc.control.led.service: Service
 hold-off time over, scheduling restart.
Nov 21 06:08:23 witherspoon systemd[1]: Stopped Phosphor LED Management Daemon.
Nov 21 06:08:23 witherspoon systemd[1]: [[0;1;39morg.openbmc.control.led.service
: Start request repeated too quickly.[[0m
Nov 21 06:08:23 witherspoon systemd[1]: [[0;1;31mFailed to start Phosphor LED Ma
nagement Daemon.[[0m
```

vishwabmc, 2017-01-12:
@sivassrr Till the time we get /sys/class/leds populated with actual LEDs, we will continue to see ledmanager issues.

@spinler and @shenki would be enabling it.

sivassrr, 2017-01-15:
Here is the output with latest driver 443(https://openpower.xyz/job/openbmc-build/443/distro=ubuntu,target=witherspoon/artifact/images/witherspoon/)

```
[  OK  ] Started D-Bus System Message Bus.
[  OK  ] Started Phosphor Watchdog0.
[  OK  ] Started Phosphor BMC0 Control.
[  OK  ] Started Phosphor IPMI BT DBus Bridge.
[  OK  ] Started Phosphor Event Manager.
[  OK  ] Started Avahi mDNS/DNS-SD Stack.
[  OK  ] Started Phosphor Host0 Checkstop Monitor.
         Starting Network Service...
         Starting Phosphor DBus Service Discovery Manager...
[  OK  ] Started Clear one time boot overrides.
[  OK  ] Started Permit User Sessions.
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Started Network Service.
[  OK  ] Reached target Network.
[  OK  ] Started Lightweight SLP Server.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
[  OK  ] Started Serial Getty on ttyS4.
[  OK  ] Reached target Login Prompts.
[  OK  ] Reached target Multi-User System.
         Starting Phosphor Inband IPMI...
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Started Phosphor Inband IPMI.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
[  OK  ] Started Phosphor LED Management Daemon.
[  OK  ] Stopped Phosphor LED Management Daemon.
         Starting Phosphor LED Management Daemon...
[  OK  ] Started Phosphor LED Management Daemon.
ftgmac100 1e660000.ethernet eth0: NCSI interface up
[  OK  ] Stopped Phosphor LED Management Daemon.
[FAILED] Failed to start Phosphor LED Management Daemon.
See 'systemctl status org.openbmc.control.led.service' for details.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 witherspoon ttyS4

witherspoon login: random: nonblocking pool is initialized

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 witherspoon ttyS4

witherspoon login: root
Password: 
root@witherspoon:~# cat /etc/version
20170115005748
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-385"
VERSION_ID="v1.99.0-385-ge9ba289-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-385"
BUILD_ID="v1.99.0"
root@witherspoon:~# journalctl > ledmanagementdaemonfailedjournal.txt
root@witherspoon:~# ls /sys/class/leds
root@witherspoon:~# 
```

sivassrr, 2017-01-15:
[ledmanagementdaemonfailedjournal.txt](https://github.com/openbmc/openbmc/files/706600/ledmanagementdaemonfailedjournal.txt)
Thank you @vishwabmc for your update. Here is the latest details from latest driver 443.

gkeishin, 2017-05-26:
This is no longer a bug.. fixed and verified ..

closed 2017-05-26 by gkeishin