openbmc/openbmc #895: [bitbake] Taskhash mismatch for os-release

ya-mouse, 2016-12-19:
I've faced two issues:
1. DISTRO_VERSION is not overwritten by phosphor's settings (got v1.99.0-githash-...)
2. When moving .git out get a taskhash mismatch.

To fix this just add a line after [this](https://github.com/openbmc/openbmc/blob/master/meta-phosphor/common/recipes-core/os-release/os-release.bbappend#L21):
> do_deploy[vardepsexclude] = "BUILD_ID VERSION VERSION_ID"

ya-mouse, 2016-12-19:
Duplicate to #124

closed 2016-12-19 by ya-mouse

amboar, 2016-12-19:
@ya-mouse do you mind pushing a fix into gerrit? I've run into this as well and it's a little irritating.

ya-mouse, 2016-12-20:
I've found that it fix (while os-release.bb has changed) only for the first time. If I change env, then it came back until I touch the os-release.bb. So, need to dive deeper to find a cause.

ya-mouse, 2016-12-21:
Just to backlog the findings. There is a [build_dependencies](https://github.com/openbmc/openbmc/blob/master/import-layers/yocto-poky/bitbake/lib/bb/data.py#L408) routine to calculate deps depending on flags. We have to use:

> do_compile[vardepsexclude] = "VAR1 VAR2..."

instead of:

> do_compile_remove[vardeps] = "VAR1 VAR2..."

[Here](https://github.com/openbmc/openbmc/blob/master/import-layers/yocto-poky/bitbake/lib/bb/siggen.py#L136-L141) is the hash calculation part including variables values.
In our case we miss to exclude at least one more variable: PRETTY_NAME. Also, we might exclude NAME.

I can push a fix to gerrit. It's just one line fix.

williamspatrick, 2016-12-22:
Yes, please do. It would be appreciated.