openbmc/openbmc #896: reboot hung after code update - umount: can't unmount /oldroot: Device or resource busy

geissonator, 2016-12-20:
I was updating BarPVT with some CI images of my new code.  The code in question is just adding some new systemd unit files so should have no affect in this area.  The code is based off of master (which is d6d3ef30d right now).  Here's what I saw in the serial after the code update:

```
[  OK  ] Stopped Load/Save Random Seed.
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Reached target Shutdown.
[  OK  ] Reached target Final Step.
         Starting Reboot...
watchdog: watchdog0: watchdog did not stop!
systemd-shutdown[1]: Sending SIGTERM to remaining processes...
systemd-journald[415]: Received SIGTERM from PID 1 (systemd-shutdow).
systemd-shutdown[1]: Sending SIGKILL to remaining processes...
systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
systemd-shutdown[1]: Unmounting file systems.
systemd-shutdown[1]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work'.
systemd-shutdown[1]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work'.
systemd-shutdown[1]: All filesystems unmounted.
systemd-shutdown[1]: Deactivating swaps.
systemd-shutdown[1]: All swaps deactivated.
systemd-shutdown[1]: Detaching loop devices.
systemd-shutdown[1]: All loop devices detached.
systemd-shutdown[1]: Detaching DM devices.
systemd-shutdown[1]: All DM devices detached.
systemd-shutdown[1]: Successfully changed into root pivot.
systemd-shutdown[1]: Returning to initrd...
watchdog: watchdog0: watchdog did not stop!
shutdown: reboot --log-level 6 --log-target kmsg --log-color
+ awk /oldroot|mnt/ { print $2 }
+ sort -r
+ umount /oldroot/sys/kernel/debug
+ umount /oldroot/sys/fs/cgroup/systemd
+ umount /oldroot/sys/fs/cgroup
+ umount /oldroot/sys
+ umount /oldroot/proc
+ umount /oldroot/dev/shm
+ umount /oldroot/dev/pts
+ umount /oldroot/dev
+ umount /oldroot
umount: can't unmount /oldroot: Device or resource busy
+ umount /mnt
+ set +x
Remaining mounts:
cow /oldroot overlay ro,relatime,lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work 0 0
tmpfs / tmpfs rw,nosuid,nodev,mode=755 0 0
dev /dev devtmpfs rw,relatime,size=248232k,nr_inodes=62058,mode=755 0 0
proc /proc proc rw,relatime 0 0
sys /sys sysfs rw,relatime 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
reboot: Restarting system

```

I'll try a PDU cycle next.

geissonator, 2016-12-20:
System came back fine after the power cycle.

geissonator, 2017-01-04:
Hit this again tonight when updating to a CI image.  PDU cycle fixed it again.
```
+ umount /oldroot/sys/kernel/debug
+ umount /oldroot/sys/fs/cgroup/systemd
+ umount /oldroot/sys/fs/cgroup
+ umount /oldroot/sys
+ umount /oldroot/proc
+ umount /oldroot/dev/shm
+ umount /oldroot/dev/pts
+ umount /oldroot/dev
+ umount /oldroot
umount: can't unmount /oldroot: Device or resource busy
+ umount /mnt
+ set +x
Remaining mounts:
cow /oldroot overlay ro,relatime,lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work 0 0
tmpfs / tmpfs rw,nosuid,nodev,mode=755 0 0
dev /dev devtmpfs rw,relatime,size=248232k,nr_inodes=62058,mode=755 0 0
proc /proc proc rw,relatime 0 0
sys /sys sysfs rw,relatime 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
reboot: Restarting system
```

geissonator, 2017-01-04:
Here's the output on a working run:
```
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Reached target Shutdown.
watchdog: watchdog0: watchdog did not stop!
systemd-shutdown[1]: Sending SIGTERM to remaining processes...
systemd-journald[408]: Received SIGTERM from PID 1 (systemd-shutdow).
systemd-shutdown[1]: Sending SIGKILL to remaining processes...
systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
systemd-shutdown[1]: Unmounting file systems.
systemd-shutdown[1]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work'.
systemd-shutdown[1]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work'.
systemd-shutdown[1]: All filesystems unmounted.
systemd-shutdown[1]: Deactivating swaps.
systemd-shutdown[1]: All swaps deactivated.
systemd-shutdown[1]: Detaching loop devices.
systemd-shutdown[1]: All loop devices detached.
systemd-shutdown[1]: Detaching DM devices.
systemd-shutdown[1]: All DM devices detached.
systemd-shutdown[1]: Successfully changed into root pivot.
systemd-shutdown[1]: Returning to initrd...
watchdog: watchdog0: watchdog did not stop!
shutdown: reboot --log-level 6 --log-target kmsg --log-color
+ awk /oldroot|mnt/ { print $2 }
+ sort -r
+ umount /oldroot/sys/kernel/debug
+ umount /oldroot/sys/fs/cgroup/systemd
+ umount /oldroot/sys/fs/cgroup
+ umount /oldroot/sys
+ umount /oldroot/proc
+ umount /oldroot/dev/shm
+ umount /oldroot/dev/pts
+ umount /oldroot/dev
+ umount /oldroot
+ umount /mnt
+ set +x
Remaining mounts:
tmpfs / tmpfs rw,nosuid,nodev,mode=755 0 0
dev /dev devtmpfs rw,relatime,size=248232k,nr_inodes=62058,mode=755 0 0
proc /proc proc rw,relatime 0 0
sys /sys sysfs rw,relatime 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
reboot: Restarting system

U-Boot 2016.07 (Dec 19 2016 - 06:14:18 +0000)

DRAM:  496 MiB
WARNING: Caches not enabled
...
```

geissonator, 2017-01-04:
#912 seems to be a recreate of this issue.

geissonator, 2017-01-09:
@amboar could you take a look at this one?  Seems like failing to unmount a filesystem should not cause a reboot to hang?

amboar, 2017-01-09:
@geissonator sure, I'll try reproduce/diagnose it

geissonator, 2017-01-12:
Thanks @amboar.  Yeah, we're using the .all.tar image for the update.  We're using this automated test:
https://github.com/openbmc/openbmc-test-automation/blob/master/extended/code_update/update_bmc.robot#L7

So it's a mix of the REST interface + an scp of the .all.tar image over.  We see this issue maybe ~5-10% of the time.

geissonator, 2017-02-23:
Haven't seen this one in a long time so closing out.

closed 2017-02-23 by geissonator