openbmc/openbmc #897: Host won't boot

cyrilbur-ibm, 2016-12-21:
To reproduce: 

On on the serial console
- cut power to palmetto using pdu
- wait 1 minute
- apply power
- wait for boot
- login
- `obmcutil poweron`

I have `obmc-console-client` running in an ssh connection launched before trying to boot the host.

On palmetto:
```
root@palmetto:~# cat /etc/os-release && uname -a
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-327"
VERSION_ID="v1.99.0-327-ga58823b-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-327"
BUILD_ID="v1.99.0"
Linux palmetto 4.7.10-74268a3a99db82b2af4fb8029d6a014a41404513 #1 Mon Dec 19 05:02:45 UTC 2016 armv5tejl GNU/Linux
```

Host just doesn't ever come up - blank host console and only repeated:
```
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up
```
on the bmc console

journalctl shows:
```
Dec 21 04:53:11 palmetto host_watchdog.exe[690]: Setting watchdog poll interval to: 30000
Dec 21 04:53:11 palmetto host_watchdog.exe[690]: Starting watchdog with poll interval: 30000
Dec 21 04:53:11 palmetto control_host.exe[780]: Using primary side of the bios flash
Dec 21 04:53:12 palmetto systemd[1]: Started Start Watchdog0.
Dec 21 04:53:12 palmetto system_manager.py[746]: Running System State: HOST_BOOTING
Dec 21 04:53:12 palmetto host_watchdog.exe[690]: Setting watchdog poll interval to: 30000
Dec 21 04:53:12 palmetto host_watchdog.exe[690]: Starting watchdog with poll interval: 30000
Dec 21 04:53:12 palmetto systemd[1]: Reached target Chassis0 (Start).
Dec 21 04:53:42 palmetto host_watchdog.exe[690]: Polling watchdog: 0
Dec 21 04:53:42 palmetto chassis_control.py[706]: Watchdog Error, Hard Rebooting
Dec 21 04:53:42 palmetto chassis_control.py[706]: Turn off power
Dec 21 04:53:42 palmetto systemd[1]: Starting Cleanup of Temporary Directories...
Dec 21 04:53:42 palmetto systemd[1]: Starting Wait for Power0 to turn off...
Dec 21 04:53:42 palmetto systemd[1]: Starting Stop Watchdog0...
Dec 21 04:53:42 palmetto systemd[1]: Stopped target Chassis0 (Start).
Dec 21 04:53:42 palmetto systemd[1]: Stopped Start Watchdog0.
Dec 21 04:53:42 palmetto systemd[1]: Stopped target Host0 (Started).
Dec 21 04:53:42 palmetto systemd[1]: Stopped target Start Host0.
Dec 21 04:53:42 palmetto system_manager.py[746]: Running System State: HOST_POWERING_OFF
Dec 21 04:53:42 palmetto systemd[1]: Stopped Start Host0.
Dec 21 04:53:42 palmetto power_control.exe[781]: PowerControl: setting power up POWER_PIN to 1
Dec 21 04:53:42 palmetto host_watchdog.exe[690]: Stopping watchdog
Dec 21 04:53:42 palmetto systemd[1]: Stopped PCIE Slot Presence Detection For Chassis0.
Dec 21 04:53:42 palmetto systemd[1]: Stopped target Power0 (On).
Dec 21 04:53:42 palmetto systemd[1]: Stopped target Power0 On.
Dec 21 04:53:42 palmetto systemd[1]: Stopped Start Power0.
Dec 21 04:53:42 palmetto systemd[1]: Started Stop Power0.
Dec 21 04:53:42 palmetto systemd[1]: Stopped Wait for Power0 to turn on.
Dec 21 04:53:42 palmetto systemd[1]: Stopped target Start Host0 (Pre).
Dec 21 04:53:42 palmetto systemd[1]: Stopped Inventory Upload.
Dec 21 04:53:43 palmetto systemd[1]: Started Cleanup of Temporary Directories.
Dec 21 04:53:43 palmetto systemd[1]: Started Stop Watchdog0.
Dec 21 04:53:45 palmetto power_control.exe[781]: PowerControl: setting reset USB_RESET to 1
Dec 21 04:53:45 palmetto power_control.exe[781]: PowerControl: setting pci reset PCIE_RESET to 1
Dec 21 04:53:45 palmetto system_manager.py[746]: Running System State: HOST_POWERED_OFF
Dec 21 04:53:45 palmetto sensor_manager2.py[696]: Deleting OCC device
Dec 21 04:53:45 palmetto chassis_control.py[706]: Turn on power and boot
Dec 21 04:53:45 palmetto systemd[1]: Started Wait for Power0 to turn off.
Dec 21 04:53:45 palmetto systemd[1]: Starting Inventory Upload...
Dec 21 04:53:45 palmetto systemd[1]: Starting Wait for Power0 to turn on...
Dec 21 04:53:45 palmetto systemd[1]: Stopped target Power0 Off.
Dec 21 04:53:45 palmetto systemd[1]: Stopped target Chassis0 (Stop).
Dec 21 04:53:45 palmetto systemd[1]: Stopped Stop Watchdog0.
Dec 21 04:53:45 palmetto systemd[1]: Stopped Stop Power0.
Dec 21 04:53:45 palmetto systemd[1]: Stopped Wait for Power0 to turn off.
Dec 21 04:53:45 palmetto systemd[1]: Started Start Power0.
Dec 21 04:53:45 palmetto power_control.exe[781]: PowerControl: setting power up POWER_PIN to 0
Dec 21 04:53:45 palmetto system_manager.py[746]: Running System State: HOST_POWERING_ON
Dec 21 04:53:46 palmetto kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Dec 21 04:53:46 palmetto kernel[710]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Dec 21 04:53:48 palmetto kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Dec 21 04:53:48 palmetto kernel[710]: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Dec 21 04:53:48 palmetto power_control.exe[781]: PowerControl: setting reset USB_RESET to 0
Dec 21 04:53:48 palmetto power_control.exe[781]: PowerControl: setting pci reset PCIE_RESET to 0
Dec 21 04:53:48 palmetto system_manager.py[746]: Running System State: HOST_POWERED_ON
Dec 21 04:53:48 palmetto systemd[1]: Started Wait for Power0 to turn on.
Dec 21 04:53:48 palmetto systemd[1]: Reached target Power0 On.
Dec 21 04:53:48 palmetto systemd[1]: Reached target Power0 (On).
Dec 21 04:53:48 palmetto systemd[1]: Starting PCIE Slot Presence Detection For Chassis0...
Dec 21 04:53:49 palmetto systemd[1]: Started PCIE Slot Presence Detection For Chassis0.
Dec 21 04:53:50 palmetto inventory_upload.py[1522]: ERROR (name_properties): "name" property in /bmc/inventory/membuf is incorrect ("TN71-BP012" instead of bas
e node name)
Dec 21 04:53:50 palmetto inventory_upload.py[1522]: ERROR (name_properties): "name" property in /bmc/inventory/dimm1 is incorrect ("0x0b" instead of base node 
name)
Dec 21 04:53:50 palmetto inventory_upload.py[1522]: ERROR (name_properties): "name" property in /bmc/inventory/dimm2 is incorrect ("0x0b" instead of base node 
name)
Dec 21 04:53:50 palmetto inventory_upload.py[1522]: ERROR (name_properties): "name" property in /bmc/inventory/dimm3 is incorrect ("0x0b" instead of base node 
name)
Dec 21 04:53:50 palmetto inventory_upload.py[1522]: ERROR (name_properties): "name" property in /bmc/inventory/cpu is incorrect ("PROCESSOR MODULE" instead of 
base node name)
Dec 21 04:53:50 palmetto inventory_upload.py[1522]: ERROR (name_properties): "name" property in /bmc/inventory/dimm0 is incorrect ("0x0b" instead of base node 
name)
Dec 21 04:53:50 palmetto inventory_upload.py[1522]: ERROR: Input tree has errors, aborting (use -f to force output)
Dec 21 04:53:50 palmetto inventory_upload.py[1522]: Uploading inventory to PNOR in dtb format...
Dec 21 04:53:50 palmetto inventory_upload.py[1522]: About to erase 0x01ebc000..0x01ec5000 !
Dec 21 04:53:50 palmetto inventory_upload.py[1522]: Erasing...
Dec 21 04:53:50 palmetto inventory_upload.py[1522]: [126B blob data]
Dec 21 04:53:50 palmetto inventory_upload.py[1522]: Error 4 erasing 0x01ebc000
Dec 21 04:53:50 palmetto systemd[1]: Started Inventory Upload.
Dec 21 04:53:50 palmetto systemd[1]: Reached target Start Host0 (Pre).
Dec 21 04:53:50 palmetto systemd[1]: Starting Start Host0...
Dec 21 04:53:50 palmetto control_host.exe[780]: Booting host
Dec 21 04:53:50 palmetto systemd[1]: Started Start Host0.
Dec 21 04:53:50 palmetto systemd[1]: Reached target Start Host0.
Dec 21 04:53:50 palmetto systemd[1]: Reached target Host0 (Started).
Dec 21 04:53:50 palmetto systemd[1]: Starting Start Watchdog0...
```

The above seems to repeat

gtmills, 2018-05-23:
@mine260309 Do you know if this is still valid? Can we close?

mine260309, 2018-07-06:
Sure it could be closed. We have Palmetto working fine on master.

closed 2018-07-06 by gtmills