openbmc/openbmc #898: obmc-console-client escape character "~." is same with ssh connection escape character.

willenlee, 2016-12-21:
When we use ssh to connect to BMC and executing obmc-console-client then entering escape character "~." to  escape obmc-console-client. that will cause ssh session terminated.
Our questions are :
1. ssh and obmc-console-client escape character should not be the same by default setting,right?
2. Have any method to set escape character of obmc-console-client dynamically?

mine260309, 2016-12-22:
I don't think this is an issue.
* The command to exit from ssh session is `<Return>~.`
* The command to exit from ssh sesseion's ssh session is `<Return>~~.`

So to exit from obmc-console-client, simply type `<Return>~~.`.

staneyshen, 2016-12-22:
According to the info from below link, user will use ~. to exit obmc-console-client, therfore user will encounter the ssh session terminate problem.

https://github.com/openbmc/docs/blob/master/console.md

Local console connections
If you're already logged into an OpenBMC machine, you can start a console session directly, using:
**$ obmc-console-client**
To exit from a console, type:
**return ~ .**

mine260309, 2016-12-22:
* If you're using serial console and run `obmc-console-client`, you should type `return ~ .` to quit;
* If you're using ssh to login to OpenBMC, and then run `obmc-console-client`, it is actually running ssh on ssh, so you need to type `return ~~.` to quit `obmc-console-client` and stays in ssh session.

staneyshen, 2016-12-22:
I agree with your explanation but user doesn't know obmc-console-client is running ssh, they think it likes SOL function.

May I know that will obmc-console-client support -e parameter(like ssh -e "escape char")? Thanks.

amboar, 2016-12-22:
@mine260309's approach is the normal approach. We just need to document in an obvious place that the double-escape sequence is required. I think that's better than using a non-standard escape sequence.

gtmills, 2018-05-23:
I will document and then close this issue.

amboar, 2019-04-02:
How's that documentation going @gtmills ?

gtmills, 2019-04-24:
Apologies for forgetting about this. 
https://gerrit.openbmc-project.xyz/c/openbmc/docs/+/20943

rfrandse, 2019-04-25:
<https://gerrit.openbmc-project.xyz/c/openbmc/docs/+/20943> Console: Added escaping console while inside ssh
Resolves  #898 obmc-console-client escape character "~." is same with ssh connection escape character.

closed 2019-04-29 by geissonator