openbmc/openbmc #901: Host time is getting sync to NTP time with Split owner and Manual mode

rahulmah, 2016-12-21:
This is how we are hitting this failure, Palmetto system was set with BMC owner and Manual mode. At this point, both HOST and BMC time was 2016-02-29 09:10:08.000. 

Then we tried to set owner to Split and owner to Manual. Later we tried to set BMC time to 2016-02-29 09:10:00.

Now after set BMC time operation, we see that BMC time is correctly set to 2016-02-29 09:10:00.000 but HOST time is set to NTP time - 2016-11-21 06:10:27.000

Expectation is that HOST time should remain un change after setting BMC time with split_manual combination. This is not happening. 

Journal log: 
[Journal logs for Split_manual case.txt](https://github.com/openbmc/openbmc/files/665995/Journal.logs.for.Split_manual.case.txt)

vishwabmc, 2017-06-13:
I don't understand the problem description.

`Then we tried to set owner to Split and owner to Manual.` --> That is not a valid input.

`HOST time is set to NTP time ` --> I don't follow this. Where is NTP in the picture here ?. 

Also, not sure why you say Host_time should stay unchanged. How much was the delta from previous ?. what is really the expectation here ?. When was the host time checked ?

vishwabmc, 2017-06-14:
@rahulmah ^

vishwabmc, 2017-07-03:
I rejected this with this comment : 

This does not look like a valid usecase. Host would set the time based on what they get from NTP servers and it does not make sense to set a time in BMC with something other than what that NTP server would provide.

Unless there are NTP servers that set time with huge difference ( which I dont see how and why ), this is not a valid case.

vishwabmc, 2017-07-26:
@williamspatrick I don't see any counter updates on this. Could you reject this ?.

rahulmah, 2017-07-26:
Vishwa I missed checking on the issue. Let you update you on this by tomorrow.

rahulmah, 2017-07-27:
@vishwabmc : I retested this issue. NTP time is no where in picture of this issue. So please ignore this NTP from it.  Basical this is what is happening. 

1.  System is set to Split owner and Manual mode.
2. BMC and HOST time is coming as 2017-07-25 19:24:48 from REST.
3. Now I tried to set BMC time to 2016-02-29 09:10:00 using REST.
4. After time setting, BMC time is correctly coming as 2016-02-29 09:10:00. But I also see that even the host time is set to 2016-02-29 09:11:28. Host time is not expected to change here because of SPLIT owner.

Let me know if you want a separate issue for this and it sound altogether different.

vishwabmc, 2017-07-27:
@rahulmah SPLIT mode does not mean that GetHostTime() would return whatever Host Set it Last time. SPLIT is : 

```     *  SPLIT   Both BMC and HOST will maintain their individual clocks but then 
     *          the time information is stored in BMC. BMC can have either NTP
     *          or MANUAL as it's source of time and will set the time directly
     *          on the BMC. When IPMI_SET_SEL_TIME is received, then the delta
     *          between that and BMC's time is calculated and is stored.
     *          When BMC reads the time, the current time is returned.
     *          When IPMI_GET_SEL_TIME is received, BMC's time is retrieved and
     *          then the delta offset is factored in prior to returning.
```

vishwabmc, 2017-07-28:
# **STATUS**

rahulmah, 2017-07-31:
@vishwabmc : Thanks for correcting me. I am am seeing that host time as 2016-02-29 09:11:28 at the end of below steps. Can you please, let me know how we are calculating and coming to this host time? From your SPLIT definition, I am unable to come to this host time.

1. System is set to Split owner and Manual mode.
2. BMC and HOST time is coming as 2017-07-25 19:24:48 from REST.
3. Now I tried to set BMC time to 2016-02-29 09:10:00 using REST.
4. After time setting, BMC time is correctly coming as 2016-02-29 09:10:00.  And host time is coming as 2016-02-29 09:11:28.

vishwabmc, 2017-07-31:
@rahulmah what was the time interval between step 3 and 4 ?. I don't see how you are seeing BMC time exactly what you have set without any uptick of a second. Could you revisit the testcase ?

rahulmah, 2017-08-01:
@vishwabmc : This test is run using automation thats why there we see BMC time matching. As far as step 3 and 4 are concerned, there is a delay of 1 second. I have tried to run this test case multiple times but I am continuosly seeing this problem.

vishwabmc, 2017-08-01:
@rahulmah Can you try executing the commands manually ?. I am trying to eliminate any testcase variation here.

rahulmah, 2017-08-02:
@vishwabmc : Yes I have tried the steps manually and seeing the same behaviour.

1. Set to Split owner and Manual mode using REST

```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"SPLIT\"}" -X PUT  https://<BMC_IP>/org/openbmc/settings/host0/attr/tie_owner
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"MANUAL\"}" -X PUT  https://<BMC_IP>/org/openbmc/settings/host0/attr/tme_mode
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
bash-4.1$ curl -c cjar -b cjar -k X GET  https://<BMC_IP>/org/openbmc/settings/host0"
{
  "data": {
    "auto_reboot": "yes",
    "boot_flags": "Default",
    "boot_policy": "ONETIME",
    "network_config": "ipaddress=,prefix=,gateway=,mac=,addr_type=",
    "power_cap": 1000,
    "power_policy": "RESTORE_LAST_STATE",
    "restricted_mode": 0,
    "system_state": "HOST_POWERED_OFF",
    "time_mode": "MANUAL",
    "time_owner": "SPLIT",
    "use_dhcp_ntp": "yes"
  },
  "message": "200 OK",
  "status": "ok"
  }
```

2. Get BMC and HOST time .

```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": [\"BMC\"]}" https://<BMC_IP>/org/openbmc/TimeManager/action/Getime
{
  "data": [
    "Fri Jul 28 18:38:25 2017 GMT",
    1501267105624548
  ],
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": [\"HOST\"]}" https://<BMC_IP>/org/openbmc/TimeManager/action/GeTime
{
  "data": [
    "Fri Jul 28 18:38:26 2017 GMT",
    1501267106214876
  ],
  "message": "200 OK",
  "status": "ok"
}
```
3. Now tried to set BMC time to 2016-02-29 09:10:00 using REST.

```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": [\"BMC\", \"2016-02-29 09:10:00\"]}" https://<BMC_IP>/org/openbc/TimeManager/action/SetTime
{
  "data": 0,
  "message": "200 OK",
  "status": "ok"
}
```

4. After time setting, BMC time is correctly coming as 2016-02-29 09:10:03. And host time is coming as 2016-02-29 09:15:59.

```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": [\"BMC\"]}" https://<BMC_IP>/org/openbmc/TimeManager/action/Getime
{
  "data": [
    "Mon Feb 29 09:10:03 2016 GMT",
    1456737003222488
  ],
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": [\"HOST\"]}" https://<BMC_IP>/org/openbmc/TimeManager/action/GeTime
{
  "data": [
    "Mon Feb 29 09:15:59 2016 GMT",       <----- HOST time is around 5 mins ahead of BMC time. 
    1456737359311627
  ],
  "message": "200 OK",
  "status": "ok"
}bash-4.1$
```

vishwabmc, 2017-08-07:
@rahulmah Thanks for the test. I suggest that we wait on the new time manager and then try ?.

rahulmah, 2017-10-23:
Sure let me test this out.

rahulmah, 2017-10-24:
@mine260309: I have retested this by following the steps which I mention on 2nd Aug above. And with new time xyz changes, I see that BMC is set to the given time and HOST is also having same set timing. This is expected I believe.  So seems like I can close the issue

rahulmah, 2017-10-24:
@mine260309: With latest time xyz changes, I am still facing the same problem which I have highlighted on 2nd August. I have followed the same steps with latest master `v1.99.10-169-g0bb2201` and see the same behavior.  Steps attached. 
[Set BMC time split mode.txt](https://github.com/openbmc/openbmc/files/1410423/Set.BMC.time.split.mode.txt)

mine260309, 2017-10-25:
@rahulmah I did the same test and the result is OK:
```
$ curl -b cjar -k https://rom106/xyz/openbmc_project/time/enumerate
{
  "data": {
    "/xyz/openbmc_project/time/bmc": {
      "Elapsed": 1508868434285572
    },
    "/xyz/openbmc_project/time/host": {
      "Elapsed": 1508868434295933
    },
    "/xyz/openbmc_project/time/owner": {
      "TimeOwner": "xyz.openbmc_project.Time.Owner.Owners.Split"
    },
    "/xyz/openbmc_project/time/sync_method": {
      "TimeSyncMethod": "xyz.openbmc_project.Time.Synchronization.Method.Manual"
    }
  },
  "message": "200 OK",
  "status": "ok"
}

$ curl -b cjar -k -H "Content-Type: application/json" -X PUT -d '{"data": "1456737000000000"}' https://rom106/xyz/openbmc_project/time/bmc/attr/Elapsed
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}

$ curl -b cjar -k https://rom106/xyz/openbmc_project/time/enumerate
{
  "data": {
    "/xyz/openbmc_project/time/bmc": {
      "Elapsed": 1456737006126959
    },
    "/xyz/openbmc_project/time/host": {
      "Elapsed": 1508868699024457
    },
    "/xyz/openbmc_project/time/owner": {
      "TimeOwner": "xyz.openbmc_project.Time.Owner.Owners.Split"
    },
    "/xyz/openbmc_project/time/sync_method": {
      "TimeSyncMethod": "xyz.openbmc_project.Time.Synchronization.Method.Manual"
    }
  },
  "message": "200 OK",
  "status": "ok"
}
```

Could you kindly help to provide the json-pretty format journal log with your test?

vishwabmc, 2017-10-25:
@rahulmah Missed asking this when you said you tested on Aug:6 and re-tested this now.

During that time, the settings were already moved to xyz namespace and I saw test mentioned updating settings in "org.openbmc.settings", which was not used `but` was just not deleted.

So, I am unsure what was really the mode accepted by the Time Manager and the updates you made to org.openbmc.settings were not consumed but ignored. 

I recommend you re-test with using the settings in xyz namespace and while the BMC is in READY state and let us know

https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Time/Owner.interface.yaml

https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Time/Synchronization.interface.yaml

They are in 

```
xyz.openbmc_project.Settings

    /xyz/openbmc_project/time/sync_method
        org.freedesktop.DBus.Peer
        org.freedesktop.DBus.Introspectable
        org.freedesktop.DBus.Properties
        xyz.openbmc_project.Time.Synchronization
            TimeSyncMethod = xyz.openbmc_project.Time.Synchronization.Method.NTP
    /xyz/openbmc_project/time/owner
        org.freedesktop.DBus.Peer
        org.freedesktop.DBus.Introspectable
        org.freedesktop.DBus.Properties```

rahulmah, 2017-11-08:
@mine260309 : As discussed, we are still seeing this issue with new time xyz interface. Below are the steps to reproduce it. 

1. Set BMC and NTP mode
```curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.Time.Owner.Owners.BMC\"}" -X PUT  https://<BMC_IP>/xyz/openbmc_project/time/owner/attr/TimeOwner

curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.Time.Synchronization.Method.NTP\"}" -X PUT  https://<BMC_IP>/xyz/openbmc_project/time/sync_method/attr/TimeSyncMethod

curl -c cjar -b cjar -k -X GET  https://<BMC_IP>/xyz/openbmc_project/time/enumerate
```
2. Now set manual mode and split owner.
```curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.Time.Synchronization.Method.Manual\"}" -X PUT  https://<BMC_IP>/xyz/openbmc_project/time/sync_method/attr/TimeSyncMethod

curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.Time.Owner.Owners.Split\"}" -X PUT  https://<BMC_IP>/xyz/openbmc_project/time/owner/attr/TimeOwner

curl -c cjar -b cjar -k -X GET  https://wsbmc014/xyz/openbmc_project/time/enumerate
{
  "data": {
    "/xyz/openbmc_project/time/bmc": {
      "Elapsed": 1510098046223818
    },
    "/xyz/openbmc_project/time/host": {
      "Elapsed": 1510098046238616
    },
    "/xyz/openbmc_project/time/owner": {
      "TimeOwner": "xyz.openbmc_project.Time.Owner.Owners.Split"
    },
    "/xyz/openbmc_project/time/sync_method": {
      "TimeSyncMethod": "xyz.openbmc_project.Time.Synchronization.Method.Manual"
    }
  },
  "message": "200 OK",
  "status": "ok"
}
```

3. Change BMC time to 1487304700000000 (i.e. GMT: Friday, 17 February 2017 04:11:40)
```curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": 1487304700000000}" -X PUT  https://<BMC_IP>/xyz/openbmc_project/time/bmc/attr/Elapsed```

4. Now check BMC and host time. 
```
bash-4.1$ curl -c cjar -b cjar -k X GET  https://wsbmc014/xyz/openbmc_project/time/enumerate                                                                  {
  "data": {
    "/xyz/openbmc_project/time/bmc": {
      "Elapsed": 1487304709243455                      <----------------GMT: Friday, 17 February 2017 04:11:49.243
    },
    "/xyz/openbmc_project/time/host": {
      "Elapsed": 1487304860030999                      <-----------------GMT: Friday, 17 February 2017 04:14:20.031
    },
    "/xyz/openbmc_project/time/owner": {
      "TimeOwner": "xyz.openbmc_project.Time.Owner.Owners.Split"
    },
    "/xyz/openbmc_project/time/sync_method": {
      "TimeSyncMethod": "xyz.openbmc_project.Time.Synchronization.Method.Manual"
    }
  },
  "message": "200 OK",
  "status": "ok"
}
```

rfrandse, 2017-11-08:
<https://gerrit.openbmc-project.xyz/7830> Fix host time incorrect issue switching to SPLIT
Resolves  #901 Barreleye: Host time is getting sync to NTP time with Split owner and Manual mode

rahulmah, 2017-11-08:
Changing defect headline to reflect failure correctly

closed 2017-12-04 by geissonator