openbmc/openbmc #902: Clock time mode change is not getting applied on poweron system

rahulmah, 2016-12-21:
Time mode change is not getting applied on power on system. I checked and found that, it needs a reboot to get the setting applied. 

Expected results: Time mode related changes should get applied regardless of system state.

See step to reproduce for mode info.

1. Make sure system is in HOST_BOOTED state and Time owner is BMC and Time mode is NTP
```
root@barreleye:~# obmcutil state
 = HOST_BOOTED
root@barreleye:~#
```

and Time owner is BMC and Time mode is NTP
```
rahulmah@rackspace:~$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/org/openbmc/settings/host0
{
  "data": {
    "boot_flags": "default",
    "boot_policy": "ONETIME",
    "network_config": "ipaddress=,prefix=,gateway=,mac=,addr_type=",
    "power_cap": 0,
    "power_policy": "RESTORE_LAST_STATE",
    "restricted_mode": 0,
    "system_state": "HOST_POWERED_ON",
    "time_mode": "NTP",
    "time_owner": "BMC",
    "use_dhcp_ntp": "yes"
  },
  "message": "200 OK",
  "status": "ok"
```
```
}rahulmah@rackspace:~$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"Manual\"}" -X PUT  https://xx.xx.xx.xx/org/openbmc/settings/ost0/attr/time_mode^C
rahulmah@rackspace:~$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/org/openbmc/TimeManager
{
  "data": {
    "curr_time_mode": "NTP",
    "curr_time_owner": "BMC",
    "requested_time_mode": "NTP",
    "requested_time_owner": "BMC"
  },
  "message": "200 OK",
  "status": "ok"
```
2. Now try to change time mode from NTP to Manual. Here you will see that current time mode is not set to Manual. 
```
}rahulmah@rackspace:~$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"Manual\"}" -X PUT  https://xx.xx.xx.xx/org/openbmc/settings/ost0/attr/time_mode
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```
Timemanger requested time is change to Manual but current time mode is still NTP. 
```
rahulmah@rackspace:~$ curl -c cjar -b cjar -khttps://xx.xx.xx.xx/org/openbmc/TimeManager                                                                       {
  "data": {
    "curr_time_mode": "NTP",
    "curr_time_owner": "BMC",
    "requested_time_mode": "MANUAL",
    "requested_time_owner": "BMC"
  },
  "message": "200 OK",
  "status": "ok"
}rahulmah@rackspace:~$
```

Logs from timemange
```
root@barreleye:~# /usr/sbin/timemanager
Registering dbus methods
Empty DhcpNtp string
Reading Power Control key: pgood
Getting System Settings: time_mode
Requested_Mode: NTP Current_Mode: NTP
Mode is already set to : NTP
Getting System Settings: time_owner
Owner is already set to : BMC
Getting System Settings: use_dhcp_ntp
use_dhcp_ntp = yes
Last known host_offset:0
Initial Uptime Usec: 1481569225512243
 User Settings have changed..
Requested_Mode: Manual Current_Mode: NTP
Deferring update until system state allows it
```

BMC fw info: 
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-338"
VERSION_ID="v1.99.0-338-gd6d3ef3-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-338"
BUILD_ID="v1.99.0"
root@barreleye:~#

Journal logs: 
[Journal log for time mode change.txt](https://github.com/openbmc/openbmc/files/666241/Journal.log.for.time.mode.change.txt)

sivassrr, 2017-05-25:
Do we still see this issue?  And it is applicable only to barreleye, right

williamspatrick, 2017-05-30:
This is by design.

closed 2017-05-30 by williamspatrick