openbmc/openbmc #903: how to compile u-boot of openbmc using local file

sdemirhan, 2016-12-21:
I would like change u-boot of openbmc. So i download u-boot from https://github.com/openbmc/u-boot and i made a change in that.  I want to get openbmc image using this local directory. How to reference this local directory and how to compile full of openbmc when i changed the source code associated with u-boot. 
Do you have any suggestion?
Seda

kc8apf, 2016-12-21:
After running '. openbmc-env', a tool called devtool will be in your path.  Run 'devtool modify -n u-boot <path-to-local-source>'.  This first creates a new Yocto layer in your build directory where devtool stores recipe modifications.  It then constructs a .bbappend for the u-boot recipe and uses the externalsource class to replace the download, fetch, and patch steps with no-ops.  The result is that when you build u-boot, it will use the local source directory as is.  Keep in mind that the u-boot recipe does _not_ do a clean nor does the externalsource class know when the source has been changed.  If you change the source, you'll need to run 'bitbake -c cleansstate u-boot' to clear BitBake's caches otherwise BitBake will just used the cached result.  Depending on what you are doing, you may need to run 'make mrproper' in the u-boot source directory to clear any built objects.  When you are all done, run 'devtool reset u-boot' to remove the .bbappend from the devtool Yocto layer.  The source will be untouched but BitBake will go back to using the version specified in the recipe.

williamspatrick, 2016-12-22:
There are three general approaches to modifying any package:

1. Update the SRC_URI variable in the original recipe.bb file to point at a custom git-repo of your own.
2. Modify the SRC_URI variable in a recipe.bbappend file in a machine-specific layer or meta-phosphor layer to point at a custom git-repo of your own.
3. Create a patch file with your changes and update SRC_URI using method 1 or 2 to include the patch.

I know some of our developers will create their own github fork of the project they are modifying so that they can use method 1 and point at their own github fork.  This allows them to more easily share development with their peers who might want to use or test their in-development changes.

@kc8apf gave you details that he has discovered on u-boot specifically as well as the 'devtool' which is a helper utility provided by Yocto to, it sounds like, assist with method 2.

Please let us know if you need additional assistance.  I would recommend checking in on IRC if you need real-time help.

closed 2016-12-22 by williamspatrick