openbmc/openbmc #904: [Witherspoon:v1.99.0-351-g8c1f920] Driver 422 - Sensor related TCs fails because of dbus exception

sivassrr, 2016-12-22:
root@witherspoon:~# cat /etc/version
20161222025539
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-351"
VERSION_ID="v1.99.0-351-g8c1f920-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-351"
BUILD_ID="v1.99.0"
root@witherspoon:~# 

Start / End / Elapsed: 	20161222 13:01:31.812 / 20161222 13:01:31.814 / 00:00:00.002
13:01:31.814 	INFO 	${msg} = Response code:500, Content:{
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException(\"Unknown interface 'org.openbmc.Object.Delete'.\",)", 
    "traceback": [
      "...

Logs the given message with the given level.
Start / End / Elapsed: 	20161222 13:01:31.815 / 20161222 13:01:31.816 / 00:00:00.001
13:01:31.815 	INFO 	Response code:500, Content:{
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException(\"Unknown interface 'org.openbmc.Object.Delete'.\",)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle", 
      "    return route.call(**args)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 522, in __call__", 
      "    return self.callback(*a, **kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 678, in wrap", 
      "    return self.to_jsonp(callback(*a, **kw))", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper", 
      "    rv = callback(*a, **ka)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 613, in wrap", 
      "    resp = {'data': callback(*a, **kw)}", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 87, in __call__", 
      "    return getattr(self, 'do_' + request.method.lower())(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 183, in do_get", 
      "    mapper_data=request.route_data['map']).dataitems()}", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 184, in enumerate_subtree", 
      "    path, owner, interfaces, match))", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 107, in __get_properties_on_bus", 
      "    properties_iface, i))", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 93, in __get_properties_on_iface", 
      "    properties_iface.GetAll, iface)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 79, in __try_properties_interface", 
      "    return f(*a)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
      "    **keywords)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
      "    message, timeout)", 
      "DBusException: org.freedesktop.DBus.Error.UnknownInterface: Unknown interface 'org.openbmc.Object.Delete'."
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"
} 	
13:01:31.816 	INFO 	${resp} = <Response [500]>
[Witherspooncombined10iterationsDriver422.pdf](https://github.com/openbmc/openbmc/files/668646/Witherspooncombined10iterationsDriver422.pdf)
[failurejournallog.txt](https://github.com/openbmc/openbmc/files/668650/failurejournallog.txt)

williamspatrick, 2016-12-22:
Please use backtick escapes on logs. 

What was the rest API call being done when this failed?

sivassrr, 2016-12-22:
Basically Get request using the url /org/openbmc/enumerate:  Details as follows:

{ret} = RequestsLibrary.RequestsKeywords . Get Request openbmc, ${base_uri}, &{kwargs}, timeout=${timeout}
Documentation: 	

Send a GET request on the session object found using the
Start / End / Elapsed: 	20161222 13:01:27.818 / 20161222 13:01:31.803 / 00:00:03.985
13:01:31.802 	INFO 	Get Request using : alias=openbmc, uri=/org/openbmc/enumerate, headers=None 	
13:01:31.803 	INFO 	${ret} = <Response [500]>

sivassrr, 2016-12-22:
More detais related to driver available @ https://github.com/openbmc/openbmc-test-automation/issues/220

sivassrr, 2017-05-29:
With recent drivers, Not seeing this issue. So closing this issue.

closed 2017-05-29 by sivassrr