openbmc/openbmc #908: openBmc uart console

sdemirhan, 2016-12-28:
Hi, 
I would like to use the u-boot console to get data u-boot of openbmc for palmetto. When i power on openbmc, i want to see u-boot flow of openbmc on console so i use serial channel. Serial channel default is uart5(0x1e784000) at ast2400 platform. I want to change this address to 0x1e78f000. I changed the source code to try different ways. Firstly,
1- I changed  CONFIG_DEBUG_UART_PHYS  address to 0x1e78f000 in defconfig file which is placed in openbmc/meta-openbmc-bsp/meta-aspeed/meta-ast2400/recipes-kernel/linux/linux-obmc. I compiled the code of openbmc and i placed it in system. System did not work, by mean i did not access openbmc using ssh . So i set the default value(0x1e784000) to CONFIG_DEBUG_UART_PHYS address. 
2- I changed CONFIG_SYS_NS16550_COM1 value from AST_UART0_BASE(0x1e784000) to  AST_UART4_BASE (it is same the 0x1e78f000)  which is placed in tmp/work/palmetto-openbmc-linux-gnueabi/u-boot-fw-utils/v2016.07+gitAUTOINC+f9b2a49f57-r0/git/include/configs/ast-common.h. So i set the default configure(AST_UART0_BASE) to CONFIG_SYS_NS16550_COM1.
Again i did not acces openbmc using ssh. It did not work.
3- I changed "r0 from 0x1e78400 to 0x1e78f000" which is placed openbmc/build/tmp/work/x86_64-linux/u-boot-mkimage-native/v2016.07+gitAUTOINC+f9b2a49f57-r0/git/arch/arm/mach-aspeed/platform_g4.S.
Again i did not acces openbmc using ssh. It did not work.

Do you have any recommends how can access console ?
When i access linux of openbmc, i want to send data from palmetto to my computer it is connectted to Palmetto with serial line. Palmetto's uart address(0x1e78f000) is different from default address(0x1e74000). I use stty command sending data from palmetto to my computer.
How can i send data to my computer from palmetto system? Do you have any suggestion?

Thanks!!!

Seda

amboar, 2017-01-03:
Hi Seda,

> I would like to use the u-boot console to get data u-boot of openbmc for palmetto. When i power on openbmc, i want to see u-boot flow of openbmc on console so i use serial channel. Serial channel default is uart5(0x1e784000) at ast2400 platform. I want to change this address to 0x1e78f000.

Why do you want to make this change? Do you want to route the console out of a different physical UART port? As an alternative the Aspeed SoCs have a fairly involved switching mechanism which allows almost arbitrary re-routing of UART IOs to Ports. If you have the datasheet, see section `1.7.3
UART Interface Application` and the description of the LPC `HICRA` register.

Regardless, `0x1e78f000` is routed to port 4 by default (unless you reconfigure the route via `HICRA[27:25]`), and I can't find a reference to the port 4 nets in my Palmetto schematic.

> I changed the source code to try different ways. Firstly,
> 1- I changed CONFIG_DEBUG_UART_PHYS address to 0x1e78f000 in defconfig file which is placed in [...]

This seems a little confused. You talk about getting u-boot output on the BMC console in your opening paragraph and then above you talk about using ssh to test.

OpenBMC has two "serial" outputs:

1. The BMC console: This a normal DB-9 RS-232 UART configuration
2. The host console: This is usually accessed via `ssh -p 2200 my-openbmc` and is a SoL replacement

These consoles are distinct concepts. Booting the BMC will not necessarily give you output on the host console (unless you follow through and boot the host e.g. via the REST interface or have previously configured the restart power state). But, it should definitely give you output on the BMC console.

> 
> Do you have any recommends how can access console ?

We typically don't have any issues with the BMC console, and unless you've physically altered your Palmetto I can't immediately see why it shouldn't work as is. If you're talking about the host console, then we do have some issues booting Palmetto hosts correctly, and you may be running into these bugs.

> When i access linux of openbmc, i want to send data from palmetto to my computer it is connectted to Palmetto with serial line. Palmetto's uart address(0x1e78f000) is different from default address(0x1e74000).

Can you expand on what you mean here? The default address for what?

> I use stty command sending data from palmetto to my computer.
> How can i send data to my computer from palmetto system? Do you have any suggestion?

You can also ssh to the BMC (on the regular port 22) assuming it's booted into the OpenBMC userspace.

yiozkok, 2017-01-03:
Dear Amboar, 

Thank you very much for your detailed reply. I'm working with Seda. Let me clarify the issue. We have TYAN palmetto with AMI Bios. We are trying to run palmetto with openbmc firmware. As you know palmetto has 128 MB flash by default. We purchased a compatible 256 Mb. winbond flash for openbmc.  First of all we build openbmc for palmetto with default values and flashed the image on our 256 mb spi flash. We hope to see some output from serial console. As you indicated openbmc console is by default set to uart5 which is physically connected to 0x1E7840000 base address on ast2400. On palmetto there is only one RS 232 serial output. As we do not know which aspeed port is routed to this output we questioned that. To clarify that issue we booted palmetto with ami bios and in ami bios linux we looked into dmesg output and there, we found that amibios is being used 0x1e78F0000 which indicates uart4 for console output. And in amibios linux we can send out same data via serial port. That is because we think that we should update openbmc console setting with uart4 instead of uart5. 
But as you indicated uart routing can be changed in ast2400 so still there is a missing point which can not be clarified without having palmetto schematic. Because maybe in amibios the routing could be changed. 
As far as I understand you have palmetto schematic. Is it possible to share the schematic with us. 

By the way, in openbmc we tried all ast2400 uart port by modifying low level platform_g4.s assembly file to send out something via serial port, but unfortunately we could not succedded yet. 

In summary,
We appreciated any help to figure out why we can not access rs-232 console of our palmetto with openbmc.
and if you can share palmetto schematic, if possible.

Do you have any checklist to figure out what is going wrong?

Regards,
Yusuf.

amboar, 2017-01-04:
> Thank you very much for your detailed reply. I'm working with Seda. Let me clarify the issue. We have TYAN palmetto with AMI Bios. 

Ah, nice. Thanks for trying OpenBMC!

> We are trying to run palmetto with openbmc firmware. As you know palmetto has 128 MB flash by default. We purchased a compatible 256 Mb. winbond flash for openbmc. First of all we build openbmc for [...]

I was incorrect above, it's us who have modified our Palmettos. Essentially you have two options:

1. Manufacture a cable to expose the COM2 header on the motherboard
2. (Untested) Reroute UART5 to COM1 by configuring `HICRA` to `0x10001000` (with an update mask of `0x70007000`) in u-boot and Linux.

That second suggestion I generated simply by eye-balling the Aspeed datasheet, so treat that as just as likely not to work without further verification.

> As far as I understand you have palmetto schematic. Is it possible to share the schematic with us.

I'm not sure of the release state of the schematic, I'd have to ask around.

yiozkok, 2017-01-04:
thank you for clarification,
we manufactured a cable for com2 header. only pins 2, 3 and 5 are connected. We tried both null and cross connection. But unfortunately nothing printed out by default openbmc image (that is uart5 is defined as console).

Now I'm checking if port routing is modified in openbmc source code? And I wonder which uart is connected to COM2 header on palmetto? As I assume it should be uart2 according to the ast2400 userguide. In that case default openbmc source code should  be modified, am I wrong?

best wishes,
Yusuf.

jk-ozlabs, 2017-01-04:
@yiozkok : just to clarify, you're referring to the console of the BMC, right? (not the POWER host)

yiozkok, 2017-01-04:
yes, I mean bmc console. We are trying to communicate BMC with our host PC (not power8) on which we have the development environment.

jk-ozlabs, 2017-01-04:
@yiozkok : ok, thanks.

On our development machines, for BMC console we're using uart5 (ie, at address `0x1E784000`), and connecting to the small pin header on the board (and not the DB9 port), and have "straight-through" serial routing (so UART5 is IO5). I'm not sure where the "COM2" designation comes from, as that's slightly confusing.

If you're changing from AMI firmware, ensure that you've also replaced the u-boot image too, as that may set up serial configuration that isn't compatible with OpenBMC serial port setup. Make sure you're using the correct baud rate too, as I *think* we use a different baud between OpenBMC and AMI.

yiozkok, 2017-01-04:
Yeah, at the end, we got data from console. We figured out that BMC_Debug1 jumper should be set. After we connected the jumper on 2-3 pins we could get data. 

Finally I would like to ask you about a mysterious socket on palmetto board. 
There are two sockets for BMC firmware (BMC SPI FW ROM Socket (BMC1)) and PNOR firmware on the (on the socket next to BMC1 there is PNOR image which is related to Power8 firmware). But there is an other mysterious socket on the palmetto board. I attached the following image to point which socket I mean? I know that is not related to console but I would like to ask if you have any info, which may affect our further work. 

[image]

Thank you very much for your help. 

best wishes,
Yusuf.

yiozkok, 2017-01-04:
By the way, for future record if anybody else need that help;

BMC_DEBUG1 jumper is for TX pin of console (connects TX to outside (COM2))
BMC_DEBUG2 jumper is for RX pin of console (connects RX to outside (COM2))

by setting only BMC_DEBUG1 jumper you can only receive data from BMC but can not send any charecter into BMC :-)

regards,

amboar, 2017-01-04:
@yiozkok Nice work on hunting down the issue.

We should start an AMI to OpenBMC conversion wiki page if we don't already have something along those lines.

yiozkok, 2017-01-04:
That is a good idea,

Besides, having a txt document in openbmc repository in palmetto folder could be very nice and helpfull.

jk-ozlabs, 2017-01-04:
@yiozkok: that looks like SPI flash connected to the SATA controller. I'm not fully across the details what the controller uses it for, but there may be documentation for that Marvell SATA controller (88SE9235) elsewhere.

amboar, 2017-01-05:
@yiozkok @jk-ozlabs I've pushed a change to openbmc/docs into gerrit which documents the process:

https://gerrit.openbmc-project.xyz/#/c/1635/1

Feedback welcome!

gtmills, 2018-08-20:
Closing since 1635 has been merged. Thanks!

closed 2018-08-20 by gtmills