openbmc/openbmc #909: [Witherspoon - v1.99.0-353-ga3b8a0a] Driver 423 - Clear error log REST not working in some system

sivassrr, 2017-01-02:
``` Few System clear error log not happening. (Note: Working system also has same driver 

root@witherspoon:# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.0-353"
VERSION_ID="v1.99.0-353-ga3b8a0a-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.0-353"
BUILD_ID="v1.99.0"
root@witherspoon:# cat /etc/version
20161231134643
root@witherspoon:~# 

Failure System output:
20:24:48.721 	INFO 	Post Request using : alias=openbmc, uri=/org/openbmc/records/events/action/clear, data={"data": []}, headers={u'Content-Type': u'application/json'}, files=None, allow_redirects=True 	
20:24:48.721 	INFO 	${ret} = <Response [200]>

On reading the output, response should be empty: but we got non empty string. 
hould Be Empty ${json['data']}
Documentation: 	

Verifies that the given item is empty.
Start / End / Elapsed: 	20170101 20:24:50.519 / 20170101 20:24:50.537 / 00:00:00.018
20:24:50.520 	INFO 	Length is 114 	
20:24:50.520 	FAIL 	'[u'/org/openbmc/records/events/1160', u'/org/openbmc/records/events/1163', u'/org/openbmc/records/events/1165', u'/org/openbmc/records/events/1166', u'/org/openbmc/records/events/1167', u'/org/openbmc/records/events/1047', u'/org/openbmc/records/events/1044', u'/org/openbmc/records/events/1043', u'/org/openbmc/records/events/1042', u'/org/openbmc/records/events/1041', u'/org/openbmc/records/events/1040', u'/org/openbmc/records/events/1049', u'/org/openbmc/records/events/1078', u'/org/openbmc/records/events/1079', u'/org/openbmc/records/events/1073', u'/org/openbmc/records/events/1071', u'/org/openbmc/records/events/1076', u'/org/openbmc/records/events/1077', u'/org/openbmc/records/events/1074', u'/org/openbmc/records/events/1146', u'/org/openbmc/records/events/1147', u'/org/openbmc/records/events/1144', u'/org/openbmc/records/events/1145', u'/org/openbmc/records/events/1143', u'/org/openbmc/records/events/1141', u'/org/openbmc/records/events/1089', u'/org/openbmc/records/events/1148', u'/org/openbmc/records/events/1149', u'/org/openbmc/records/events/1088', u'/org/openbmc/records/events/1068', u'/org/openbmc/records/events/1065', u'/org/openbmc/records/events/1064', u'/org/openbmc/records/events/1067', u'/org/openbmc/records/events/1066', u'/org/openbmc/records/events/1061', u'/org/openbmc/records/events/1062', u'/org/openbmc/records/events/1018', u'/org/openbmc/records/events/1019', u'/org/openbmc/records/events/1011', u'/org/openbmc/records/events/1013', u'/org/openbmc/records/events/1014', u'/org/openbmc/records/events/1016', u'/org/openbmc/records/events/1017', u'/org/openbmc/records/events/1159', u'/org/openbmc/records/events/1091', u'/org/openbmc/records/events/1092', u'/org/openbmc/records/events/1095', u'/org/openbmc/records/events/1097', u'/org/openbmc/records/events/1098', u'/org/openbmc/records/events/1152', u'/org/openbmc/records/events/1155', u'/org/openbmc/records/events/1154', u'/org/openbmc/records/events/1157', u'/org/openbmc/records/events/1156', u'/org/openbmc/records/events/1090', u'/org/openbmc/records/events/1158', u'/org/openbmc/records/events/1008', u'/org/openbmc/records/events/1003', u'/org/openbmc/records/events/1002', u'/org/openbmc/records/events/1001', u'/org/openbmc/records/events/1007', u'/org/openbmc/records/events/1006', u'/org/openbmc/records/events/1005', u'/org/openbmc/records/events/1004', u'/org/openbmc/records/events/1083', u'/org/openbmc/records/events/1080', u'/org/openbmc/records/events/1086', u'/org/openbmc/records/events/1085', u'/org/openbmc/records/events/1124', u'/org/openbmc/records/events/1125', u'/org/openbmc/records/events/1126', u'/org/openbmc/records/events/1127', u'/org/openbmc/records/events/1121', u'/org/openbmc/records/events/1123', u'/org/openbmc/records/events/1038', u'/org/openbmc/records/events/1037', u'/org/openbmc/records/events/1035', u'/org/openbmc/records/events/1032', u'/org/openbmc/records/events/1030', u'/org/openbmc/records/events/1031', u'/org/openbmc/records/events/1138', u'/org/openbmc/records/events/1137', u'/org/openbmc/records/events/1136', u'/org/openbmc/records/events/1135', u'/org/openbmc/records/events/1134', u'/org/openbmc/records/events/1132', u'/org/openbmc/records/events/1130', u'/org/openbmc/records/events/1102', u'/org/openbmc/records/events/1103', u'/org/openbmc/records/events/1100', u'/org/openbmc/records/events/1101', u'/org/openbmc/records/events/1107', u'/org/openbmc/records/events/1104', u'/org/openbmc/records/events/1109', u'/org/openbmc/records/events/1020', u'/org/openbmc/records/events/1023', u'/org/openbmc/records/events/1025', u'/org/openbmc/records/events/1026', u'/org/openbmc/records/events/1029', u'/org/openbmc/records/events/1028', u'/org/openbmc/records/events/1115', u'/org/openbmc/records/events/1114', u'/org/openbmc/records/events/1116', u'/org/openbmc/records/events/1110', u'/org/openbmc/records/events/1113', u'/org/openbmc/records/events/1112', u'/org/openbmc/records/events/1119', u'/org/openbmc/records/events/1054', u'/org/openbmc/records/events/1055', u'/org/openbmc/records/events/1056', u'/org/openbmc/records/events/1050', u'/org/openbmc/records/events/1052', u'/org/openbmc/records/events/1053', u'/org/openbmc/records/events/1059']' should be empty.

sivassrr, 2017-01-02:
```
Working System Output:
equestsLibrary.RequestsKeywords . Post Request openbmc, ${base_uri}, &{kwargs}, timeout=${timeout}
Documentation: 	

Send a POST request on the session object found using the
Start / End / Elapsed: 	20170102 10:42:50.411 / 20170102 10:42:50.882 / 00:00:00.471
10:42:50.882 	INFO 	Post Request using : alias=openbmc, uri=/org/openbmc/records/events/action/clear, data={"data": []}, headers={u'Content-Type': u'application/json'}, files=None, allow_redirects=True 	
10:42:50.882 	INFO 	${ret} = <Response [200]>

Response:
Should Be Empty ${json['data']}
Documentation: 	

Verifies that the given item is empty.
Start / End / Elapsed: 	20170102 10:42:52.356 / 20170102 10:42:52.357 / 00:00:00.001
10:42:52.357 	INFO 	Length is 0   <======

sivassrr, 2017-01-02:
Updated the logs in https://github.com/openbmc/openbmc-test-automation/issues/220

sivassrr, 2017-01-03:
```
In another instance, When trying to delete particular event (Details given below)
Send a GET request on the session object found using the
Start / End / Elapsed: 	20170101 20:44:05.480 / 20170101 20:44:05.781 / 00:00:00.301
20:44:05.781 	INFO 	Get Request using : alias=openbmc, uri=/org/openbmc/records/events/1183/action/delete, headers=None 	
20:44:05.781 	INFO 	${ret} = <Response [500]>

===============================================
Following dbus-exception happened.
Logs the given message with the given level.
Start / End / Elapsed: 	20170101 20:44:05.791 / 20170101 20:44:05.792 / 00:00:00.001
20:44:05.791 	INFO 	Response code:500, Content:{
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException(\"Unknown object '/org/openbmc/records/events/1183'.\",)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 858, in _handle", 
      "    route, args = self.router.match(environ)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 744, in custom_router_match", 
      "    route.callback._setup(**args)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 82, in _setup", 
      "    self.find(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 199, in find", 
      "    m = self.find_method_on_bus(path, method, *items)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 235, in find_method_on_bus", 
      "    data = iface.Introspect()", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
      "    **keywords)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
      "    message, timeout)", 
      "DBusException: org.freedesktop.DBus.Error.UnknownObject: Unknown object '/org/openbmc/records/events/1183'."
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"
} 	
20:44:05.793 	INFO 	${resp} = <Response [500]>
===================================================================

Note: In good system 10 iterations succeed without any issues like above.

sivassrr, 2017-01-03:
[20170101204709294537_BMC_dmesg.txt](https://github.com/openbmc/openbmc/files/681727/20170101204709294537_BMC_dmesg.txt)
[20170101204709294537_BMC_general.txt](https://github.com/openbmc/openbmc/files/681728/20170101204709294537_BMC_general.txt)
[20170101204709294537_BMC_inventory.txt](https://github.com/openbmc/openbmc/files/681730/20170101204709294537_BMC_inventory.txt)
[20170101204709294537_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/681731/20170101204709294537_BMC_journalctl.txt)
[20170101204709294537_BMC_meminfo.txt](https://github.com/openbmc/openbmc/files/681732/20170101204709294537_BMC_meminfo.txt)
[20170101204709294537_BMC_proc_list.txt](https://github.com/openbmc/openbmc/files/681729/20170101204709294537_BMC_proc_list.txt)
[20170101204709294537_BMC_procinfo.txt](https://github.com/openbmc/openbmc/files/681733/20170101204709294537_BMC_procinfo.txt)
[20170101204709294537_BMC_sensor_list.txt](https://github.com/openbmc/openbmc/files/681735/20170101204709294537_BMC_sensor_list.txt)
[test_history.txt](https://github.com/openbmc/openbmc/files/681734/test_history.txt)

amboar, 2017-01-03:
@sivassrr I applaud your use of the triple-backtick here, but you've now gone too far! Please only put *code* or *log output* between the triple-backticks: If you want to make comments about the code or logs please leave your comments *outside* the triple-backticks. Thanks.

sivassrr, 2017-01-14:
With Driver 442 failed on clear event REST didnt work, but this time it is different errror (503 - services not available).  Error as follows:

```
 	

Send a POST request on the session object found using the
Start / End / Elapsed: 	20170113 10:36:45.184 / 20170113 10:36:45.514 / 00:00:00.330
10:36:45.514 	INFO 	Post Request using : alias=openbmc, uri=/org/openbmc/records/events/action/clear, data={"data": []}, headers={u'Content-Type': u'application/json'}, files=None, allow_redirects=True 	
10:36:45.514 	INFO 	${ret} = <Response [503]> 	
00:00:00.007 KEYWORD BuiltIn . Run Keyword If '${quiet}' == '${0}', Log Response, ${ret}
Documentation: 	

Runs the given keyword with the given arguments, if condition is true.
Start / End / Elapsed: 	20170113 10:36:45.515 / 20170113 10:36:45.522 / 00:00:00.007
00:00:00.004 KEYWORD rest_client . Log Response ${ret}
Start / End / Elapsed: 	20170113 10:36:45.518 / 20170113 10:36:45.522 / 00:00:00.004
00:00:00.001 KEYWORD ${msg} = BuiltIn . Catenate SEPARATOR=, Response code:, ${resp.status_code}, , Content:, ${resp.content}
Documentation: 	

Catenates the given items together and returns the resulted string.
Start / End / Elapsed: 	20170113 10:36:45.519 / 20170113 10:36:45.520 / 00:00:00.001
10:36:45.520 	INFO 	${msg} = Response code:503, Content:{
  "data": {
    "description": "org.freedesktop.DBus.Error.ObjectPathInUse: busy processing bus traffic"
  }, 
  "message": "503 Service Unavailable", 
  "status": "error"...
```

sivassrr, 2017-01-14:
[cleareventDr442.zip](https://github.com/openbmc/openbmc/files/705935/cleareventDr442.zip)

sivassrr, 2017-01-14:
Basically still the clear REST is not working. Earlier it was "500 Internal error" Now it is different (503).  
Let me know if you folks want a new issue for the same.

williamspatrick, 2017-03-10:
This has been fixed in the latest code.

closed 2017-03-10 by williamspatrick