openbmc/openbmc #912: [Barrleye v1.99.0-354-gc7248eb ] watchdog: watchdog0: watchdog did not stop!

gkeishin, 2017-01-04:
Driver : v1.99.0-354-gc7248eb 
Build : https://openpower.xyz/job/openbmc-build/424/

``` During code update, we noticed that the BMC goes for back to back reboots due to this watchdog SIGTERM. and hangs in the flash... However on PDU power cycle back it seems to come back normal```
**Log Snippet :**
watchdog: watchdog0: watchdog did not stop!
systemd-shutdown[1]: Sending SIGTERM to remaining processes...
systemd-journald[416]: Received SIGTERM from PID 1 (systemd-shutdow).
systemd-shutdown[1]: Sending SIGKILL to remaining processes...
systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
systemd-shutdown[1]: Unmounting file systems.
systemd-shutdown[1]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work'.
systemd-shutdown[1]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work'.
systemd-shutdown[1]: All filesystems unmounted.
systemd-shutdown[1]: Deactivating swaps.
systemd-shutdown[1]: All swaps deactivated.
systemd-shutdown[1]: Detaching loop devices.
systemd-shutdown[1]: All loop devices detached.
systemd-shutdown[1]: Detaching DM devices.
systemd-shutdown[1]: All DM devices detached.
systemd-shutdown[1]: Successfully changed into root pivot.
systemd-shutdown[1]: Returning to initrd...
watchdog: watchdog0: watchdog did not stop!
shutdown: reboot --log-level 6 --log-target kmsg --log-color

amboar, 2017-01-05:
"watchdog did not stop!" is expected output during code update. See #522.

gkeishin, 2017-01-05:
so we can close this one out @amboar

gkeishin, 2017-01-05:
@amboar  The difference I see from #522 is that of reported during erase operation.. however in this case, it happened post update and was booting up to prompt and then it triggers the SIGTERM.. the symptoms may be the same .. but just the timing where it crashes is what I think, differentiate from it.. 

Will keep a watch on latter builds before closing this. Thanks

amboar, 2017-01-06:
@gkeishin Please attach the entire log to the issue when reporting bugs. The build behind the link has been deleted and so we can no-longer access the log.

gkeishin, 2017-01-06:
@amboar  The only log I could save off was from the terminal Serial Console, not sure if this will helps 

[Console-log.txt](https://github.com/openbmc/openbmc/files/688957/Console-log.txt)

The other logs we couldn't capture since it crashed as soon it reaches the prompt..

amboar, 2017-01-08:
@gkeishin Thanks. So it looks like:

1. The system is up
2. A reboot is initiated
3. `watchdog: watchdog0: watchdog did not stop!` is printed in rootfs and initrd reboot states
4. The system comes up to a `getty`
5. A second reboot is initiated
6. `watchdog: watchdog0: watchdog did not stop!` is printed in rootfs and initrd reboot states

Is the issue still just that `watchdog: watchdog0: watchdog did not stop!` is printed before `SIGTERM` is sent?

What initiated the reboot in step 5 above?

gkeishin, 2017-01-09:
@amboar Apart from step 2 reboot which is initiated as part of the code update... post that we don't issue reboot.. un expected reboot was seen post getty and we saw the watchdog SIGTERM message onto the console.. 

We were looking for explanation why this SIGTERM being sent and why it hang eventually post the reboot due to this or some other issues.

gkeishin, 2017-01-19:
I don't see this error anymore or not able to recreate in recent code updates.. so closing this.. Will re-open it with new logs if seen.

closed 2017-01-19 by gkeishin